maragelis/ParadoxRs232toMqtt #2: Controlling alarms through the Android application

PascalSI, 2018-03-27:
Hello, wanted to know, this allows you to watch only events with the protection of the Paradox panel?
Or it allows you to control the panel (arm, disarm)?
I did not quite understand how it works ...

maragelis, 2018-03-28:
Hi use the OTA version its the last version with Over the air updates. Yes this uses mqtt to send and receive events from the alarm system 
check out the wiki. 
If you need help just drop a line. Been using it for about 3 months now with out any issues.

PascalSI, 2018-03-28:
Hi, i already ordered ESP8266 (ESP-01), i have to get it in a week.
And which application do you use for the Android? I watched the wiki, but i did not understand everything because i did not know English well, i use translations

maragelis, 2018-03-28:
I use iphone homekit not android, to use android you will need an mqtt application with a server at home that comunicates with the panel.

closed 2018-05-04 by maragelis