maragelis/ParadoxRs232toMqtt #5: Physical Connections to Paradox and ESP8266

augisbud, 2018-06-20:
Hello, I'am trying to intergrate this to my openhab2 setup at home, and having issues with the physical connections including the resistors and pins, could you provide some sourt of schematics?

maragelis, 2018-06-21:
After meny tries , go without resistors it will work just fine.

maragelis, 2018-06-21:
updated wiki

stefanodias, 2018-06-21:
Hello, I'm trying to integrate with the Evo Paradox board, will it work? Cool you upgraded the wiki but did not see anything about the resistors, did not you use it?

maragelis, 2018-06-21:
yes you can do without, if the evo uses the standerd 37byte message it will work

maragelis, 2018-06-21:
you could start by just adding a ftdi adapter and sniffing the serial. if you see standerd 37byte messages then it will work.

stefanodias, 2018-06-21:
Nice, thanks for the tips. Another question is, can I send commands to the card or just read?

maragelis, 2018-06-21:
Yes you can, I have been using it with nodered  getting feedback and sending commands. I have all my zones doors windows as devices on openHAB. It nice arming and knowing lights will be turned off and heat turned down.

stefanodias, 2018-06-21:
What are the paradox board models you've tested?

maragelis, 2018-06-21:
I have tested sp4000 and sp6000

pilievwm, 2018-07-19:
Congrats Maragelis,
What a cool project. I am super frustrated that i bought Ip150 module for $120 instead using $2 wemos :)

Anyway. Now i am trying to connect wemos d1 mini to my Spectra 65 (SP65) but without any luck. 
I can not get the status nor the output through the mqtt. 
Here is the layout of the SP65 board

[image]

Your help and advices will make me so happy to retire the IP150 module and use the wemos instead.

maragelis, 2018-07-20:
Everything you need is in the wiki. I bet the sp65 is the same as the sp6000. As long as the panel users the 37byte message format the project will work. To check you will need to sniff the serial with a serial adapter and look at the serial messages.

maragelis, 2018-07-20:
1.)  Gpio 15 or D8 can be used to debug just connect to RS232 module RX pin
2.) panel password is sent in mqtt message 
3.) sp65 looks the same as sp6000 so serial is the same pin2 from top is ground then pin 3 RX then TX
4.) no you dont need to change the names, if you want its up to you..

are you getting status messages on /home/PARADOX4000/status ?
if not the wemos is not connected to the network. 

try your  RS232 module straight on the panel are you reading anything?

maragelis, 2018-07-20:
I would suggest if you want to try out branch/test its more stable and uses gpio 13/15 (D7/D8) for rx/tx and serial/usb as debug. 

https://github.com/maragelis/ParadoxRs232toMqtt/tree/test

hope i helped

pilievwm, 2018-07-20:
WeMos is not getting any status messages. It is connected to the network and also it is connected to the mqtt server for sure. I can see the IP address of the WeMos at my MQTT broker and it reads that it is connected. The problem is that i am not getting any statuses when i subscribe to /home/PARADOX4000/status

I will try to read something from the RS232 module straight to the computer. Do you know a serial monitoring software for windows 10?

If i understand you correctly... I have to connect the WeMos to the SP65 via RX/TX and than to connect RS232 module to the WeMos 13/15 to debug it? Where should i connect the serial/usb to debug it?

maragelis, 2018-07-20:
on init the wemos sends a hello message "ParadoxController V2.11 " this you should see if you left everything as is in the code the status message is on topic /paradox/status and not /home/PARADOX4000/status

try this on

maragelis, 2018-07-20:
const char *root_topicOut = "/paradox/out";
  const char *root_topicStatus = "/paradox/status";
  const char *root_topicIn = "/paradox/in";

maragelis, 2018-07-20:
changed wiki to the right topics so there is no misunderstanding. Sorry about that.

pilievwm, 2018-07-20:
Oh wow. I've never tried it with /paradox/status topic :)
According to your documentation, the topics are different
[image]

Will test it later and will keep you updated :) Thanks for your support!

pilievwm, 2018-07-20:
Maragelis it works like a charm! Awesome! Now how to control the system? I want to be able to arm it or disarm it? Can you tell me how to do that?

maragelis, 2018-07-20:
The wiki has most of them just look it up. The one you want is event 6 subevent 4 which is arm sleep if I am not mistaken

maragelis, 2018-07-21:
I don't know much about home assistant never used it. If anyone wants to help me I can send arm trigger events on another topic, just for hassio

maragelis, 2018-07-21:
Updated the code to use the same 4 digit code used on the keypad for security reasons. so dont use the 0000 just use the same one you use to arm disarm .

maragelis, 2018-07-24:
Any news on this can I close the issue ?

maragelis, 2018-07-30:
https://www.home-assistant.io/components/alarm_control_panel.manual_mqtt/ this can be used if you have node-red installed as well on the hassio pc. I think i can cook something up. 
Do you use node-red ?

maragelis, 2018-07-30:
try this with the plugin and node-red import this flow into node and the plugin should start working, just change the mqtt server

[{"id":"cdfcb044.6512c","type":"tab","label":"github","disabled":false,"info":""},{"id":"5af303dd.8907ec","type":"mqtt in","z":"cdfcb044.6512c","name":"","topic":"/paradox/out","qos":"2","broker":"240f31f6.8f29ce","x":110,"y":400,"wires":[["2184666b.a93eaa"]]},{"id":"2184666b.a93eaa","type":"switch","z":"cdfcb044.6512c","name":"","property":"payload.event","propertyType":"msg","rules":[{"t":"eq","v":"6","vt":"num"},{"t":"eq","v":"2","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":310,"y":400,"wires":[["7bc20bd.244f6f4"],["172bffae.5565b"]]},{"id":"7bc20bd.244f6f4","type":"switch","z":"cdfcb044.6512c","name":"Arm Codes","property":"payload.sub_event","propertyType":"msg","rules":[{"t":"eq","v":"3","vt":"num"},{"t":"eq","v":"4","vt":"num"}],"checkall":"true","outputs":2,"x":590,"y":360,"wires":[["a437b26b.63e96"],["4b32231f.ac7dbc"]],"outputLabels":["Stay Mode",""]},{"id":"a437b26b.63e96","type":"trigger","z":"cdfcb044.6512c","op1":"armed_home","op2":"","op1type":"str","op2type":"nul","duration":"2","extend":false,"units":"s","reset":"","bytopic":"all","name":"armed_home","x":790,"y":340,"wires":[["347c42d9.2b68ae"]]},{"id":"4b32231f.ac7dbc","type":"trigger","z":"cdfcb044.6512c","op1":"armed_night","op2":"","op1type":"str","op2type":"nul","duration":"2","extend":false,"units":"s","reset":"","bytopic":"all","name":"armed_night","x":790,"y":380,"wires":[["347c42d9.2b68ae"]]},{"id":"347c42d9.2b68ae","type":"change","z":"cdfcb044.6512c","name":"Set Sleep","rules":[{"t":"set","p":"Sleep","pt":"flow","to":"true","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":980,"y":360,"wires":[["d4469f05.ae736"]]},{"id":"5a8f90e1.5966f","type":"switch","z":"cdfcb044.6512c","name":"","property":"Sleep","propertyType":"flow","rules":[{"t":"false"}],"checkall":"true","outputs":1,"x":1070,"y":440,"wires":[["d4469f05.ae736"]]},{"id":"961cd4ae.bf9458","type":"trigger","z":"cdfcb044.6512c","op1":"armed_away","op2":"","op1type":"str","op2type":"nul","duration":"2","extend":false,"units":"s","reset":"","bytopic":"all","name":"armed_away","x":890,"y":440,"wires":[["5a8f90e1.5966f"]]},{"id":"dae4293f.8665e8","type":"delay","z":"cdfcb044.6512c","name":"","pauseType":"delay","timeout":"3","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":720,"y":440,"wires":[["961cd4ae.bf9458"]]},{"id":"172bffae.5565b","type":"switch","z":"cdfcb044.6512c","name":"Arm Codes","property":"payload.sub_event","propertyType":"msg","rules":[{"t":"eq","v":"12","vt":"num"},{"t":"eq","v":"11","vt":"num"},{"t":"eq","v":"4","vt":"num"}],"checkall":"true","outputs":3,"x":530,"y":460,"wires":[["dae4293f.8665e8"],["e1341b0e.2d7c88"],["23db5828.0a3e98"]],"outputLabels":["Full Arm","Disarm",""]},{"id":"e1341b0e.2d7c88","type":"trigger","z":"cdfcb044.6512c","op1":"disarmed","op2":"","op1type":"str","op2type":"nul","duration":"5","extend":false,"units":"s","reset":"","bytopic":"all","name":"Disarmed","x":760,"y":500,"wires":[["d6b05c08.4b539"]]},{"id":"d6b05c08.4b539","type":"change","z":"cdfcb044.6512c","name":"Set Sleep","rules":[{"t":"set","p":"Sleep","pt":"flow","to":"false","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":920,"y":500,"wires":[["d4469f05.ae736"]]},{"id":"23db5828.0a3e98","type":"trigger","z":"cdfcb044.6512c","op1":"triggered","op2":"","op1type":"str","op2type":"nul","duration":"2","extend":true,"units":"s","reset":"","bytopic":"all","name":"triggered","x":780,"y":560,"wires":[["d4469f05.ae736"]]},{"id":"d4469f05.ae736","type":"mqtt out","z":"cdfcb044.6512c","name":"","topic":"home/alarm","qos":"","retain":"","broker":"240f31f6.8f29ce","x":1350,"y":480,"wires":[]},{"id":"240f31f6.8f29ce","type":"mqtt-broker","z":"","name":"","broker":"192.168.2.230","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","willTopic":"","willQos":"0","willPayload":""}]

pilievwm, 2018-07-30:
Unfortunately, I don't have node-red. Can you explain what you did here? Is it a flow depending on what is the message?

maragelis, 2018-07-30:
no have not had that issue

JevonW, 2018-08-04:
Great project!

I managed to get this working with a MG5050, Home Assistant and Node-Red, this is the node Red Flow I've strung together.

[{"id":"4b73d18c.d53fa8","type":"tab","label":"Paradox","disabled":false,"info":""},{"id":"a2853419.412578","type":"mqtt in","z":"4b73d18c.d53fa8","name":"","topic":"/paradox/out","qos":"2","broker":"b02cb90a.238498","x":110,"y":160,"wires":[["32b63bda.38cdb4"]]},{"id":"94fb2be6.e30308","type":"switch","z":"4b73d18c.d53fa8","name":"event","property":"payload.event","propertyType":"msg","rules":[{"t":"eq","v":"2","vt":"num"},{"t":"eq","v":"6","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":410,"y":160,"wires":[["89a8d178.c86aa8"],["9cdbc3fd.2e141"]]},{"id":"af3d2a82.4a60d","type":"mqtt out","z":"4b73d18c.d53fa8","name":"","topic":"home/alarm/set","qos":"","retain":"","broker":"b02cb90a.238498","x":1360,"y":140,"wires":[]},{"id":"c367e9e8.2cd6a8","type":"trigger","z":"4b73d18c.d53fa8","op1":"ARM_HOME","op2":"","op1type":"str","op2type":"nul","duration":"2","extend":false,"units":"s","reset":"","bytopic":"all","name":"armed_home","x":1150,"y":60,"wires":[["af3d2a82.4a60d"]]},{"id":"32b63bda.38cdb4","type":"json","z":"4b73d18c.d53fa8","name":"","property":"payload","action":"","pretty":false,"x":270,"y":160,"wires":[["94fb2be6.e30308"]]},{"id":"89a8d178.c86aa8","type":"switch","z":"4b73d18c.d53fa8","name":"sub_event arm/disarm","property":"payload.sub_event","propertyType":"msg","rules":[{"t":"eq","v":"12","vt":"str"},{"t":"eq","v":"11","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":600,"y":120,"wires":[["19e1a548.633aa3"],["6991acf6.88a504"]]},{"id":"6991acf6.88a504","type":"trigger","z":"4b73d18c.d53fa8","op1":"DISARM","op2":"","op1type":"str","op2type":"nul","duration":"2","extend":false,"units":"s","reset":"","bytopic":"all","name":"disarm","x":1130,"y":140,"wires":[["af3d2a82.4a60d"]]},{"id":"e824c55a.56e9","type":"mqtt in","z":"4b73d18c.d53fa8","name":"","topic":"home/alarm","qos":"2","broker":"b02cb90a.238498","x":110,"y":400,"wires":[["41a7328f.e810b4"]]},{"id":"41a7328f.e810b4","type":"switch","z":"4b73d18c.d53fa8","name":"","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"disarmed","vt":"str"},{"t":"eq","v":"armed_home","vt":"str"},{"t":"eq","v":"armed_away","vt":"str"}],"checkall":"true","repair":false,"outputs":3,"x":350,"y":400,"wires":[["e43022bc.412b5","1d1e4590.735a4a"],["e9627864.609c3","167593dc.f02c4c"],["f56a4eaf.7f25a","6e2eb15a.d9098"]]},{"id":"e43022bc.412b5","type":"trigger","z":"4b73d18c.d53fa8","op1":"{\"password\":\"1234\",\"Command\":\"disarm\",\"Subcommand\":\"0\"}","op2":"0","op1type":"str","op2type":"str","duration":"2","extend":false,"units":"s","reset":"","bytopic":"all","name":"disarm inside","x":570,"y":240,"wires":[["dd568678.70ad38"]]},{"id":"dd568678.70ad38","type":"mqtt out","z":"4b73d18c.d53fa8","name":"","topic":"/paradox/in","qos":"","retain":"","broker":"b02cb90a.238498","x":1170,"y":360,"wires":[]},{"id":"1d1e4590.735a4a","type":"trigger","z":"4b73d18c.d53fa8","op1":"{\"password\":\"1234\",\"Command\":\"disarm\",\"Subcommand\":\"1\"}","op2":"0","op1type":"str","op2type":"str","duration":"5","extend":false,"units":"s","reset":"","bytopic":"all","name":"disarm outside","x":580,"y":300,"wires":[["20e39d7b.084b32"]]},{"id":"e9627864.609c3","type":"trigger","z":"4b73d18c.d53fa8","op1":"{\"password\":\"1234\",\"Command\":\"sleep\",\"Subcommand\":\"0\"}","op2":"0","op1type":"str","op2type":"str","duration":"3","extend":false,"units":"s","reset":"","bytopic":"all","name":"arm home / sleep","x":590,"y":360,"wires":[["dd568678.70ad38"]]},{"id":"167593dc.f02c4c","type":"trigger","z":"4b73d18c.d53fa8","op1":"{\"password\":\"1234\",\"Command\":\"arm\",\"Subcommand\":\"1\"}","op2":"0","op1type":"str","op2type":"str","duration":"5","extend":false,"units":"s","reset":"","bytopic":"all","name":"arm outside","x":570,"y":420,"wires":[["c00a3956.6417a"]]},{"id":"f56a4eaf.7f25a","type":"trigger","z":"4b73d18c.d53fa8","op1":"{\"password\":\"1234\",\"Command\":\"arm\",\"Subcommand\":\"0\"}","op2":"0","op1type":"str","op2type":"str","duration":"2","extend":false,"units":"s","reset":"","bytopic":"all","name":"arm inside","x":570,"y":480,"wires":[["dd568678.70ad38"]]},{"id":"6e2eb15a.d9098","type":"trigger","z":"4b73d18c.d53fa8","op1":"{\"password\":\"1234\",\"Command\":\"arm\",\"Subcommand\":\"1\"}","op2":"0","op1type":"str","op2type":"str","duration":"3","extend":false,"units":"s","reset":"","bytopic":"all","name":"arm outside","x":570,"y":540,"wires":[["91d13299.83a13"]]},{"id":"19e1a548.633aa3","type":"switch","z":"4b73d18c.d53fa8","name":"arm partition","property":"payload.dummy","propertyType":"msg","rules":[{"t":"eq","v":"Inside          ","vt":"str"},{"t":"eq","v":"Outside         ","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":810,"y":100,"wires":[["c367e9e8.2cd6a8"],["e310de2.e8c592"]]},{"id":"9cdbc3fd.2e141","type":"switch","z":"4b73d18c.d53fa8","name":"sub_event sleep","property":"payload.sub_event","propertyType":"msg","rules":[{"t":"eq","v":"4","vt":"str"}],"checkall":"true","repair":false,"outputs":1,"x":590,"y":180,"wires":[["28b32953.a15156"]]},{"id":"28b32953.a15156","type":"delay","z":"4b73d18c.d53fa8","name":"","pauseType":"delay","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":800,"y":180,"wires":[["2a241be9.dfa66c"]]},{"id":"2a241be9.dfa66c","type":" [... cut ...] Units":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":740,"y":540,"wires":[["dd568678.70ad38"]]},{"id":"b02cb90a.238498","type":"mqtt-broker","z":"","name":"","broker":"192.168.2.11","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]

JevonW, 2018-08-04:
And this is what I added to the Home Assistant Configuration.yaml

alarm_control_panel:
  - platform: manual_mqtt
    state_topic: home/alarm
    command_topic: home/alarm/set
    pending_time: 20
    delay_time: 10
    trigger_time: 4
    disarmed:
      trigger_time: 0
    armed_home:
      pending_time: 0
      delay_time: 0

maragelis, 2018-08-04:
Do you mind if I add this to the wiki

JevonW, 2018-08-04:
Not at all, please do.

maragelis, 2018-08-04:
@JevonW there seems to be a problem with the flow I cannot get it to import into node-red can you check it please

rjduraocosta, 2018-09-16:
Hello @Maragelis,

I ran across your project and find it very cool, interesting and useful project to control my MG500 panel and integrate it with openHAB. I have a few questions if you don't mind:
- How are you powering the Wemos mini board?
- I saw that you mentioned that it is possible to connect TX/RD from the Wemos directly to the Panel, is this still the case or it is better to go with the resistors approach?
- I don'r have the PC Password to connect to the Panel, it is different from the default 0000. The alarm company that installed the panel changed it. I already asked for it but I don't know if they are going to send it to me. From what I read here, to send command you do not need it either because you use the user code. For the status of the panel is the PC Password necessary?
- Which branch should I use? Master, dev or test?
- From where did you buy the Wemos? I saw some with the esp8622EX, is this the same?

Thank for your time.

maragelis, 2018-09-16:
Hi @rjduraocosta 

1.) Power i use the other two pins 12V(+) and (-) through a lm7805 straight to the wemos 5v pin.
2.)  Go direct no resistors it works great.
3.) The latest update uses the user password the same pin used to arm from the panel.
4.) Use Master, dev and Test have more functions but changes the pins around so RX/TX are used for debug .
5.) i get mine from https://www.banggood.com/Wemos-D1-Mini-V3_0_0-WIFI-Internet-Of-Things-Development-Board-Based-ESP8266-4MB-p-1264245.html?rmmds=search

rjduraocosta, 2018-09-16:
You sure helped. Thanks a lot for your prompt reply. I will proceed as you mentioned and give it a go.

Thanks again.

rjduraocosta, 2018-09-16:
Hello again,

I already order the suggested material and I can't to start with this. I am installing the Arduino IDE and added the board. I tried to compile the code but I am having the following errors:

C:\Wemos d1 mini\ParadoxRs232toMqtt-master\ParadoxAlarmSystem\ParadoxAlarmSystemOTA\ParadoxAlarmSystemOTA.ino: In function 'inPayload Decodejson(char*)':

ParadoxAlarmSystemOTA:538:3: error: 'DynamicJsonBuffer' was not declared in this scope

ParadoxAlarmSystemOTA:538:21: error: expected ';' before 'jsonBuffer'

ParadoxAlarmSystemOTA:539:22: error: 'jsonBuffer' was not declared in this scope

ParadoxAlarmSystemOTA:540:13: error: 'class ArduinoJson::JsonObject' has no member named 'success'

C:\Wemos d1 mini\ParadoxRs232toMqtt-master\ParadoxAlarmSystem\ParadoxAlarmSystemOTA\ParadoxAlarmSystemOTA.ino: In function 'void setup_wifi()':

ParadoxAlarmSystemOTA:667:7: error: 'DynamicJsonBuffer' was not declared in this scope

ParadoxAlarmSystemOTA:667:25: error: expected ';' before 'jsonBuffer'

ParadoxAlarmSystemOTA:668:26: error: 'jsonBuffer' was not declared in this scope

ParadoxAlarmSystemOTA:677:12: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'

ParadoxAlarmSystemOTA:678:12: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'

C:\Wemos d1 mini\ParadoxRs232toMqtt-master\ParadoxAlarmSystem\ParadoxAlarmSystemOTA\ParadoxAlarmSystemOTA.ino: In function 'void mountfs()':

ParadoxAlarmSystemOTA:767:9: error: 'DynamicJsonBuffer' was not declared in this scope

ParadoxAlarmSystemOTA:767:27: error: expected ';' before 'jsonBuffer'

ParadoxAlarmSystemOTA:768:28: error: 'jsonBuffer' was not declared in this scope

ParadoxAlarmSystemOTA:769:14: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'

ParadoxAlarmSystemOTA:770:18: error: 'class ArduinoJson::JsonObject' has no member named 'success'

exit status 1
'DynamicJsonBuffer' was not declared in this scope

Could you please give me so help?

Another thing please, how do you configure the wifi settings (ssid, password)?

Thanks in advanced for your help.

rjduraocosta, 2018-09-16:
Sorry to step in again, but I sorted out the erros part by installing the v5.13.2 the ArduinoJson library.

I still have the configuration of the wifi settings (ssid, password) of my home network doubt.

Where do I set them?

Could you please give me so help on this?

Thanks in advanced for all your help.

maragelis, 2018-09-16:
as soon as you turn on the chip you will find a new wifi network, connect to that and enter 192.168.4.1 into you browser. you will get a config page to set your home wifi

rjduraocosta, 2018-09-16:
Perfect!

Thanks a lot for your help once again.

rjduraocosta, 2018-09-17:
Hello again @maragelis,

I am wondering the following:

- With the buck converter or LM7805 that you suggested above the voltage drop is going to be of 7V so it is going to heat things up a bit. Did you notice any heating up of the Wemos board or your LM7805? 
- What cable section are you using between the Wemos and the LM7805/buck converter?
- Would a USB phone charger also work (something around 1A output) directtly in the Wemos USB port?

Sorry to bother you with these questions but I really don't whant to mess anything up.

Thanks again.

maragelis, 2018-09-17:
1) the lm7805 gets hot yes about 60-70 degrees centigrade, I use a buck so no heat. 

2) normal alarm cable does the job. 

3) I use the panel power so that I can get events even when power is out. Panel battery powers the wemos.

maragelis, 2018-09-17:
Check this out https://photos.app.goo.gl/okS8ujDfoXX48CQa6

rjduraocosta, 2018-09-17:
Hello again,

Perfect!

Thanks a lot for sharing. Nice assembly that you have there.

I can't wait to start.

Thanks again.

maragelis, 2018-09-29:
How are you powering the wemos, make sure it has a good power supply, at least 1amp. Most problems with esp8266 are power related

rjduraocosta, 2018-09-29:
Hello, 

Thank you for your prompt reply.

I am powering the wemos just like you recommended me, direct from the panel with this buck converter:

https://www.banggood.com/LM317-DC-DC-1_5A-1_2-37V-Adjustable-Power-Supply-Board-DC-Converter-Buck-Step-Down-Module-p-1171756.html?rmmds=buy

I regulated it to output 5,1V.

I  find it strange that the TX pin on the wemos has a voltage of 3,8V.

I think it is pretty close to work but there is a small piece missing.

Sorry, but do have anything else in mind?

Thanks again.

maragelis, 2018-09-29:
Remove the tx pin just leave the RX and see what happens.  Let's try get the input working.

rjduraocosta, 2018-09-29:
That's an excellent idea. I will give it a go and keep you posted with the result.

Thanks again for all your help.

maragelis, 2018-09-29:
Add you serial cable to debug pins set Trace to 1 and check debug messages

rjduraocosta, 2018-09-29:
Well here are the logs from the trace and i connected only the wemos RX connected:

[29/09/2018 09:06:26] Read data (COM5) 
    73 65 72 69 61 6c 20 6d 6f 6e 69 74 6f 72 20 69   serial monitor i 
    73 20 75 70 0d 0a 50 61 72 61 64 6f 78 20 73 65   s up..Paradox se 
    72 69 61 6c 20 6d 6f 6e 69 74 6f 72 20 69 73 20   rial monitor is  
    75 70 0d 0a                                       up..             
[29/09/2018 09:06:28] Read data (COM5) 
    73 65 72 69 61 6c 20 6d 6f 6e 69 74 6f 72 20 69   serial monitor i 
    73 20 75 70 0d 0a 50 61 72 61 64 6f 78 20 73 65   s up..Paradox se 
    72 69 61 6c 20 6d 6f 6e 69 74 6f 72 20 69 73 20   rial monitor is  
    75 70 0d 0a                                       up..             
[29/09/2018 09:06:30] Read data (COM5) 
    52 75 6e 6e 69 6e 67 20 4d 6f 75 6e 74 46 73 0d   Running MountFs. 
    0a                                                .                
[29/09/2018 09:06:30] Read data (COM5) 
    6d 6f 75 6e 74 65 64 20 66 69 6c 65 20 73 79 73   mounted file sys 
    74 65 6d 0d 0a 72 65 61 64 69 6e 67 20 63 6f 6e   tem..reading con 
    66 69 67 20 66 69 6c 65 0d 0a 6f 70 65 6e 65 64   fig file..opened 
    20 63 6f 6e 66 69 67 20 66 69 6c 65 0d 0a 0a 70    config file...p 
    61 72 73 65 64 20 6a 73 6f 6e 0d 0a 76 61 6c 75   arsed json..valu 
    65 73 20 61 72 20 6e 6f 20 6e 75 6c 6c 20 0d 0a   es ar no null .. 
[29/09/2018 09:06:35] Read data (COM5) 
    63 6f 6e 6e 65 63 74 65 64 2e 2e 2e 79 65 65 79   connected...yeey 
    20 3a 29 0d 0a 6c 6f 63 61 6c 20 69 70 20 3a 20    :)..local ip :  
    0d 0a 31 39 32 2e 31 36 38 2e 31 2e 31 31 31 0d   ..192.168.1.111. 
    0a 53 65 74 74 69 6e 67 20 4d 71 74 74 20 53 65   .Setting Mqtt Se 
    72 76 65 72 20 76 61 6c 75 65 73 0d 0a 6d 71 74   rver values..mqt 
    74 5f 73 65 72 76 65 72 20 3a 20 0d 0a 31 39 32   t_server : ..192 
    2e 31 36 38 2e 31 2e 31 36 30 0d 0a 6d 71 74 74   .168.1.160..mqtt 
    5f 73 65 72 76 65 72 5f 70 6f 72 74 20 3a 20 0d   _server_port : . 
    0a 31 38 38 33 0d 0a 53 65 74 74 69 6e 67 20 4d   .1883..Setting M 
    71 74 74 20 53 65 72 76 65 72 20 63 6f 6e 6e 65   qtt Server conne 
    63 74 69 6f 6e 0d 0a 41 74 74 65 6d 70 74 69 6e   ction..Attemptin 
    67 20 4d 51 54 54 20 63 6f 6e 6e 65 63 74 69 6f   g MQTT connectio 
    6e 2e 2e 2e 0d 0a 63 6f 6e 6e 65 63 74 65 64 0d   n.....connected. 
    0a 73 65 6e 64 69 6e 67 20 0d 0a 50 61 72 61 64   .sending ..Parad 
    6f 78 20 43 6f 6e 6e 65 63 74 65 64 0d 0a 74 6f   ox Connected..to 
    20 0d 0a 2f 70 61 72 61 64 6f 78 2f 73 74 61 74    ../paradox/stat 
    75 73 0d 0a 73 75 62 73 63 72 69 70 74 69 6f 6e   us..subscription 
    20 4f 4b 20 74 6f 0d 0a 2f 70 61 72 61 64 6f 78    OK to../paradox 
    2f 69 6e 0d 0a 0d 0a 57 69 46 69 20 63 6f 6e 6e   /in....WiFi conn 
    65 63 74 65 64 0d 0a 49 50 20 61 64 64 72 65 73   ected..IP addres 
    73 3a 20 0d 0a 31 38 36 32 33 37 39 37 31 32 0d   s: ..1862379712. 
    0a 46 69 6e 6e 69 73 68 65 64 20 77 69 66 69 20   .Finnished wifi  
    73 65 74 75 70 0d 0a                              setup..          
[29/09/2018 09:06:36] Read data (COM5) 
    73 65 6e 64 69 6e 67 20 0d 0a 50 61 72 61 64 6f   sending ..Parado 
    78 43 6f 6e 74 72 6f 6c 6c 65 72 20 56 32 2e 31   xController V2.1 
    31 0d 0a 74 6f 20 0d 0a 2f 70 61 72 61 64 6f 78   1..to ../paradox 
    2f 73 74 61 74 75 73 0d 0a                        /status..        

Everything seems to be ok, but no reception on the RX pin.

rjduraocosta, 2018-09-29:
Update:

I passed in front of a panel sensor and I got something:

{ "armstatus":224, "event":1, "sub_event":3, "dummy":"Entrada         "}
{ "armstatus":224, "event":0, "sub_event":3, "dummy":"Entrada         "}

Should i get also status of the panel also? Like if it is armed or something?

Sorry about all these question but I am on the beginning of this journey.

Thanks again.

maragelis, 2018-09-29:
So it's working fine. You got zone 3 open close status. If you arm you will get many events that show up check the wiki for their meaning

rjduraocosta, 2018-09-29:
Yes finally! I don't know what I was doing yesterday evening wrong...it was a long night diging into this...sorry to bother you.

I managed to send successfully an arm, disarm and sleep command, but the stay only seems to work when I send a 0 instead of the stay word. It does not matter at all, the important thing is that it works like a charm.

Thanks for all your help, support and for giving us very usefull project.

maragelis, 2018-09-29:
Ok glad you got it working. just dont forget to turn off trace in the code if you enabled it. as it uses up quite a bit of resource, and you might loose events coming in through the serial.

rjduraocosta, 2018-09-29:
Nice tip thanks I will do as you say:

#define TRACE 0

Now it is time to integrate this with openHAB.

Thanks again for everything.

closed 2018-09-29 by maragelis

SauRoNZA, 2018-10-22:
> After many tries , go without resistors it will work just fine. I use a wemos d1 , 13,15 is only used for debugging I use the rx/tx

Out of interest do you power them from the alarm system or externally?

milimarius, 2021-05-29:
Hello boys, hello rjduraocosta. I have exactly the same message:
paradox / status: Paradox Connected;
unemployed / status: ParadoxController V2.11
....and nothing else.
  I mention that I still have the MG5000 and everything was installed on the ESP-01S. I used a buck converter from 13V to 3.3V and on Rx and Tx I used optocouplers. Please give me some advice on what to do as I have been struggling for 10 hours without result. Thank you

milimarius, 2021-05-29:
> 
> 
> Add you serial cable to debug pins set Trace to 1 and check debug messages

Hello maragelis
"Add you serial cable to debug pins set Trace to 1 and check debug messages". Can you be more explicit please?

maragelis, 2021-05-30:
You can’t use the esp-01 to debug the system. I would suggest you remove the optocouplers and connect straight to the rx/tx terminals. If that doesn’t work swap the terminals around as you might have them backwards. Rx connects to Tx and Tx to Rx.

milimarius, 2021-05-30:
I come with an update:
Thanks maragelis, I did exactly as you said ... I took out the optocouplers and reversed Rx with Tx. Now it works very well and fast, I added all the sensors in HA. Now I have another problem, I don't think Tx works, it doesn't work to send any commands. I did it like this:
-I added the following code in configuration.yaml (I put the real code):
   alarm_control_panel:
  - platform: manual_mqtt
      state_topic: "paradoxdCTL / hassio / Arm"
      command_topic: "paradoxdCTL / in"
      code: 0000
      pending_time: 30
      delay_time: 20
      trigger_time: 4
    payload_disarm: '{"password": "0000", "Command": "disarm", "Subcommand": "0"}'
    payload_arm_night: '{"password": "0000", "Command": "sleep", "Subcommand": "0"}'
    payload_arm_home: '{"password": "0000", "Command": "stay", "Subcommand": "0"}'
    payload_arm_away: '{"password": "0000", "Command": "arm", "Subcommand": "0"}'
-I added the alarm panel in the dashboard
-after I enter the code and press arm.... pending .... and write armed .... but the alarm is not real army. Do I have to do anything else? Is the code wrong? I use home assistant on raspberry pi with node red.

maragelis, 2021-05-30:
Sorry don't know home assistant. Maybe someone else can help. Try mqtt explorer to see if it arms.

[16 more replies]