maragelis/ParadoxRs232toMqtt #7: API Documentation

ipodmusicman, 2018-07-20:
Hi there, thanks so much for this.  I managed to receive the various events from my SP6000, but I am unable to arm / disarm, etc and trying to figure out why as I am not getting the log in response from the panel.  I want to see what the response messages are that are returned from the panel within the "doLogon" method so that I can determine if I am specifying an incorrect password or not.

It would seem that you have the API documentation handy due to the info that you've posed to your wiki.  Is it possible that you can make it available to me please?

maragelis, 2018-07-21:
I will try get the document uploaded asap. If you cannot arm most of the time it's the panel password. 911 register on the panel. Try default 0000. If it still doesn't arm you will need to change this password. I you need help just let me know.

maragelis, 2018-07-21:
https://drive.google.com/file/d/1MHjSpjnsCSfMq87ypP0WdChkDyCvSmhB/view?usp=drivesdk

maragelis, 2018-07-21:
https://drive.google.com/open?id=1MHjSpjnsCSfMq87ypP0WdChkDyCvSmhB

ipodmusicman, 2018-07-21:
Thanks so much for the document!  Cannot wait to get stuck in.

ipodmusicman, 2018-07-21:
Can you also please share the programming guide as well that contains all the events and sub events?  You listed most of them on the wiki, but I'd like to see the rest as I see the code refers to event 48.

FYI, I suspect that the passcode I'm using is incorrect.  I'll need to contact my armed response company to reset it.

maragelis, 2018-07-21:
Like I said I just updated so it uses the user code. So no need for panel programming

maragelis, 2018-07-21:
Fyi. The programming guide can be downloaded from the paradox website.

ipodmusicman, 2018-07-21:
I use a remote to arm/disarm so I might have forgotten the code. :)  Also, what I think could also cause the error is that the armed response company said that due to the fact that I don't have a delay when arming the alarm, you cannot arm it from the panel (which of course makes sense. lol), so not sure if that has an effect or not.  Great, found the programming guide.  Thanks.

closed 2018-07-21 by maragelis