maragelis/ParadoxRs232toMqtt #9: Wired connectivity possible?

marcgarciamarti, 2018-08-06:
Hello,

I would rather prefer not to use a wireless connection between the SP8266 and the Alarm panel. Does anybody know whether that is somehow possible?

many thanks everyone and hats off to this project!!! :)

cybertza, 2018-08-07:
you could try to compile onto an Arduino Mega with Ether Shield, think the code may be a bit heavy for an arduino ethernet.

You would have to change a few minor things,

There is also some Eth addons for ESP that you can try, there will also be a few code changes needed for that.

maragelis, 2018-08-07:
A arduino mega with shield could work. Ihave not tried it.

cybertza, 2018-12-01:
https://www.aliexpress.com/item/TOP-Mini-W5100-LAN-Ethernet-Shield-Network-Module-board-Best/32457200047.html

something like that will work for you, just update everything, but, you will have to do some work, as the wifi manager wont work anymore, so there will be quite a bit of code to update that that was doing.

in all honesty wifi is rather reliable, and if you use a voltage reg from the panel, you will have better lightning immunity.

mydakota, 2019-01-05:
Except for the slim chance someone sends deauth packets to the router comms would be blocked I think a hard wired unit or duel WiFi Ethernet system would rock

maragelis, 2019-01-05:
The project and all similar projects are done to integrate the alarm system into smart homes. Don't try to make them into first level security systems. Paradox has high level security gsm modules and ip150 for that job.

closed 2019-01-31 by maragelis