maragelis/ParadoxRs232toMqtt #10: Panel commands

RowanZee, 2018-08-13:
Hi
I've tried arming the panel through the commands outlined in the wiki however nothing happens. 
I am receiving all status alerts as expected. 
Please assist

Thanks

cybertza, 2018-08-13:
Here is the basis of it:

The 0000 is your alarm user pin that u want to allocate to this.
also it is possible that your tx line is not connected properly. 
##configuration.yalm
`alarm_control_panel: !include alarmpanel.yaml`
## alarmpannel.yalm
```
- platform: mqtt
   name: House Paradox
   state_topic: "paradox/status/"
   command_topic: "paradox1/in/"
   payload_disarm: '{"password":"0000","Command":"disarm","Subcommand":"0"}'
# # payload_arm_home: "STAY"
   payload_arm_night: '{"password":"0000","Command":"sleep","Subcommand":"0"}'
   payload_arm_home: '{"password":"0000","Command":"stay","Subcommand":"0"}'
   payload_arm_away: '{"password":"0000","Command":"arm","Subcommand":"0"}'
```

RowanZee, 2018-08-14:
Thanks, does one need to use the 0000 code. I attempted this with the programmed user password.
Just for clarity do you publish the command to paradox1/in/ 
Why do you have a "1" in the paradox?

cybertza, 2018-08-14:
I have a 1 there because that’s my topic I publish and listen to on Mqtt, I have something else that used the paradox topic, all topics are char sensitive, aka A and a is different / and \ for the path is different, my next recommendation would be to do a set time, to see if your time randomly changes to the hardcoded time, I would assume that you are either posting to the wrong topic or that you have the tx line disconnected. 

Other option is to follow the guide and connect a serial bus, I believe you can just connect an ttl cable to the logpin defined in the code.

As mentioned in the readme the communication was changed from pc password to user code, I still run an older build because there is no new feature that I want yet, will prob upgrade when the web interface comes online or the set date is parsed properly, usb debug is an awesome feature, but as far as I understand the tx/rx pin is hardware serial so it’s more reliable.

SauRoNZA, 2018-11-07:
> Here is the basis of it:
>
> The 0000 is your alarm user pin that u want to allocate to this.
> also it is possible that your tx line is not connected properly.
>
> ## configuration.yalm [...]

Is the above not making use of NodeRed and just using HA directly?

maragelis, 2019-01-06:
https://github.com/maragelis/ParadoxRs232toMqtt/wiki/Home-Assistant-in-V2

closed 2019-02-05 by maragelis