maragelis/ParadoxRs232toMqtt #12: Issue with commands. Help needed.

bigstation17, 2018-09-26:
Thank you very much for this awesome project.
I successfully connect to SP4000, can read alarm status, sensors event, but i cannot send commands.
I double checked wires, change pins, i can read status, but cannot control the panel :(
I tried with user code, pc code "0000", all kinds of combinations.

What i'm doing wrong ?

Here is my config:

[image]

[image]

[image]

[image]

maragelis, 2018-09-29:
are you getting any messages back on the status mqtt sub

maragelis, 2018-09-29:
I also hope you are not missing a quote, in your image its missing
{
 "password":### **_"_**1234",
 "Command":"arm",
 "Subcommand":"0"
}

bigstation17, 2018-09-29:
Noup. Is not missing. It was hide in image. 
When i send an mqtt command i dont get any response back, i get only sensors event, events when i arm manually from keypad...

bigstation17, 2018-09-29:
I checked, even i replaced the d1 mini. 
It is posible that sp4000 board was programed to not send data trough tx?
How can i debug? I have ttl usb serial adapter.

maragelis, 2018-09-29:
you are getting tx readings it is not receiving RX (paradox side) 
I suggest you set trace to 1 and connect seial ttl to debug pin 13 and see what comes up

maragelis, 2018-09-29:
No i have a sp4000 working great i used that (sp4000) in the lab

bigstation17, 2018-09-30:
I connect TX from d1mini > RX ftdi adapter i have this
[image]

maragelis, 2018-09-30:
that looks like binary, try connecting to gpio 13

bigstation17, 2018-09-30:
I don't have any response on D7 pin

maragelis, 2018-09-30:
you have to set trace to 1 in the code to get output debug messages on D7 . Is your user code 4 digits or more?

maragelis, 2018-09-30:
I will ask again not to be rude, are you not getting any messages on paradox/status

maragelis, 2018-09-30:
sorry about that but Debug messages can be read through D8 TXD2 Pin on the wemos or if you prefer IO15.

maragelis, 2018-09-30:
Output is working ok. Try connecting d8 for debug send a arm payload and send messages

bigstation17, 2018-09-30:
On D8 Pin i have:
"⸮⸮⸮⸮⸮⸮}?}oo}⸮]⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮~o[⸮⸮⸮⸮⸮⸮⸮⸮{˹⸮⸮⸮⸮⸮⸮⸮⸮⸮׋⸮}⸮{}⸮⸮⸮⸮⸮⸮zuvuz⸮⸮⸮⸮⸮⸮⸮⸮⸮뽷⸮⸮⸮o⸮k⸮⸮⸮⸮⸮⸮{~⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮{⸮⸮⸮⸮⸮⸮⸮}⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮"

maragelis, 2018-09-30:
Bring it down to 9600

bigstation17, 2018-09-30:
It is at 9600 :(
I connected another mini direct with usb cable on laptop to debug, and when i send messages i have:
[image]

maragelis, 2018-09-30:
U are using esp core 2.4.2 I haven't tested it on that and there are many problems with that. Bring it down to 2.3.0

maragelis, 2018-09-30:
I don't how else to help you.

bigstation17, 2018-09-30:
Ok. Thank you very much.

maragelis, 2019-01-06:
Try release files bin 
https://github.com/maragelis/ParadoxRs232toMqtt/releases

bigstation17, 2019-01-13:
Thank you for your work.
I tried new version and unfortunately i have the same problem. I think that i am a special case :)
I receive all sensors correct, but no with the commands.

Now i have discover a strange behavior. When i send any message to topic paradoxdCTL/in ESP is rebooting. Even if i send a blank message.

[image]

bigstation17, 2019-01-13:
Yes. I tried with ParadoxAlarmSystemOTA.HASSIO.bin, ParadoxAlarmSystemOTA.ino.TRACE.bin and with arduino IDE, erase all data.

Esp is rebooting when i send message to paradoxdCTL/in even is not connected to panel.

serial monitor is up
Running MountFs
mounted file system
reading config file
opened config file
{"mqtt_server":"192.168.1.200","mqtt_port":"1883"}
parsed json
values ar no null 
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.233
connected...yeey :)
Setting Mqtt Server values
mqtt_server : 
192.168.1.200
mqtt_server_port : 
1883
Setting Mqtt Server connection
Attempting MQTT connection...
connected
sending 
{"status":"Paradox connected"}
to 
paradoxdCTL/status
subscription OK to
paradoxdCTL/in

WiFi connected
IP address: 
3909200064
*WM: freeing allocated params!
Starting HTTP...
Starting SSDP...
mDNS responder started
Ready!

Finnished wifi setup
Hey I got a callback 
JSON Returned! ====
returned command = 4
Start login
Address-0=5F
Address-1=20
Address-2=0
Address-3=0
Address-4=0

maragelis, 2019-01-13:
Sp4000 that’s what I am using without any problems. The only thing I can think of is old firmware on your panel.

maragelis, 2019-01-13:
I have a new realese with better debug and better panel support. Will try finish it by tomorrow night.

maragelis, 2019-01-13:
Did you try the 20190111 realese?

bigstation17, 2019-01-13:
Thank you very much, you are very patient with me :D
Yes. I tried every version :)))  20190111 - 20190103, bin files (ESPEasy), compile with arduino..etc. 
I think is an esp problem because every time (even without connected to sp4000) i send message to topic paradoxdCTL/in esp is rebooting. I tried with 2 wemos, same behavior.

maragelis, 2019-01-13:
Did you try sending the payload in the wiki. 
{
 "password":"1234",
 "Command":"arm",
 "Subcommand":"0"
}

mydakota, 2019-01-13:
I'm also having Sim issue with latest v2
Unfortunately I haven't had time to play around but same big station output .
I did also have rebooting issue on commands but think I fixed that by either reflash or I had comms reversed.

bigstation17, 2019-01-14:
> Did you try sending the payload in the wiki.
> {
> "password":"1234",
> "Command":"arm",
> "Subcommand":"0"
> }

Yes. i have rebooting problem with any payload that i send.

bigstation17, 2019-01-15:
> https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/20190114a

Noup. Same issue.
[image]

As soon i send any json message to esp is rebooting. {"xxx":"xxx"} > Reboot. 
Set Trace i s working.

I'm using Wemos D1 mini (https://wiki.wemos.cc/products:d1:d1_mini) regular version with 4mb not PRO version. Can this be the problem ?

maragelis, 2019-01-15:
Please do me a favor as soon as it boots set TRACE=1 
then send this payload as is (please dont change anything)
{"Command":"stay","Subcommand":"0","password":"0000"}
Post serial output

maragelis, 2019-01-15:
I will try a wemos d1 mini tonight and get back to you..

bigstation17, 2019-01-15:
> Please do me a favor as soon as it boots set TRACE=1 send mqtt message
> then send this payload as is (please dont change anything)
> {"Command":"stay","Subcommand":"0","password":"0000"}
> Post serial output

[image]

maragelis, 2019-01-15:
you are crashing on the panel logon session. Looks like the SoftwareSerial lib is crashing.could you paste the full stack trace 
Thanks for your help

bigstation17, 2019-01-15:
Trace is ON
Hey I got a callback 
JSON Returned! ====
parsing Recievied Json Data
returned command = 1
Json Data is ok 
Start login

Soft WDT reset

ctx: cont 
sp: 3ffffc40 end: 3fffffd0 offset: 01b0

>>>stack>>>
3ffffdf0:  00000009 00000001 0000000f 4010683d  
3ffffe00:  3ffe99e4 00000000 3ffefce4 4020b040  
3ffffe10:  00000001 000002f3 000002f3 4010020c  
3ffffe20:  00000000 00000000 3ffffeac 3ffffeaa  
3ffffe30:  3ffefce4 00000024 3ffffea9 40218dd0  
3ffffe40:  00000000 00000001 00000000 00000000  
3ffffe50:  3ffefcdf 00000001 3ffefce4 4020aa1a  
3ffffe60:  00000000 00000000 00000000 00000000  
3ffffe70:  00000000 00000000 00000000 00000000  
3ffffe80:  00000000 00205f00 00000000 00000000  
3ffffe90:  00000000 00000000 00000000 00000000  
3ffffea0:  00000000 00050000 3fff8400 40215c8d  
3ffffeb0:  3fff03dc 000002f3 000002f3 4010020c  
3ffffec0:  00000000 00000001 3fffff00 401006dc  
3ffffed0:  00000000 00000001 3fffff00 402168cc  
3ffffee0:  3ffe9460 00000001 00000000 00000000  
3ffffef0:  3ffefcdf 00000001 00000000 4020ad68  
3fffff00:  00000000 00000000 00000000 00000002  
3fffff10:  00000042 3ffefd24 3fff2c8c 0000003f  
3fffff20:  00000035 00010000 00000064 402175b2  
3fffff30:  3fffff60 00000001 0000000e 00000001  
3fffff40:  0000000e 00005c86 3ffefd24 402184e0  
3fffff50:  3fffdad0 00005c86 3ffefd24 40213530  
3fffff60:  3ffef501 00000000 3ffefbcc 40210acd  
3fffff70:  00000004 3ffefd27 00000047 3fffefb0  
3fffff80:  00000000 00000000 00000000 3fff04dc  
3fffff90:  3fffdad0 00000000 00000024 4020a000  
3fffffa0:  00000000 00000000 3fff04d5 4020a22c  
3fffffb0:  feefeffe 00000000 3fff04d5 40217584  
3fffffc0:  feefeffe feefeffe 3ffe8640 40100739  
<<<stack<<<
e⸮⸮⸮⸮⸮⸮~{"mqtt_server":"192.168.1.200","mqtt_port":"1883"}*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.233
*WM: freeing allocated params!
Starting HTTP...
Starting SSDP...

maragelis, 2019-01-15:
I need to try replicate you error on my rig.

maragelis, 2019-01-15:
OK software serial is doing the job its running out of memory. I will try a wemos d1 mini and see what the problem is.

0x4010683d: micros64 at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring.c line 201
0x4020b040: SoftwareSerial::write(unsigned char) at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/SoftwareSerial/SoftwareSerial.cpp line 188
0x4010020c: _umm_free at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1295
0x40218dd0: fs::FS::begin() at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/FS.cpp line 168
0x4020aa1a: doLogin(unsigned char, unsigned char) at /Users/panagiotis/Documents/Github/ParadoxRs232toMqtt/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino line 814
0x40215c8d: MD5Builder::getChars(char*) at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/MD5Builder.cpp line 84
0x4010020c: _umm_free at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1295
0x401006dc: free at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1755
0x402168cc: String::reserve(unsigned int) at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 148
0x4020ad68: callback(char*, unsigned char*, unsigned int) at /Users/panagiotis/Documents/Github/ParadoxRs232toMqtt/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino line 376
0x402175b2: user_init() at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 230
0x402184e0: run_scheduled_functions() at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Schedule.cpp line 45
0x40213530: PubSubClient::loop() at /Users/panagiotis/Documents/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 331
0x40210acd: ESP8266WebServer::~ESP8266WebServer() at /Users/panagiotis/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h line 185
0x4020a000: readSerialData() at /Users/panagiotis/Documents/Github/ParadoxRs232toMqtt/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino line 327
0x4020a22c: setup() at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.h line 75
0x40217584: user_init() at /Users/panagiotis/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 218

bigstation17, 2019-01-15:
👍 and i will order an wemos d1 pro :))) to test it.

maragelis, 2019-01-15:
Just flashed a new wemos d1 mini works fine.

bigstation17, 2019-01-15:
Well....i used latest bin file 20190114a with EspEasy.
It is normal that esp is rebooting on any json payload ?

[image]

maragelis, 2019-01-15:
{"Command":"stay","Subcommand":"0","password":"0000"}

bigstation17, 2019-01-15:
I get reboot with any command
{"Command":"stay","Subcommand":"0","password":"0000"}
{"Command":"stay","Subcommand":"0","password":"master pass"}
{"Command":"stay","Subcommand":"0","password":"user pass"}

maragelis, 2019-01-15:
Try download ino file and compile yourself chech libs are the same. Update arduino core and libs. Next thing left is for me to send you a working wemos

bigstation17, 2019-01-15:
:)))))) You are very patient.
Thanks anyway..i'm using as it is..reporting works very well.
I ordered wemos pro and i will check with that.
Have a nice evening.

maragelis, 2019-01-15:
The thing is I have two panels an sp6000 and a sp4000 both on my bench for test. They both work ok with the last update.

bigstation17, 2019-01-15:
It's possible to have a different firmware in panel and cannot login through serial port...
But still..i think i have an esp problem due to wdt reset on payloads.

maragelis, 2019-01-15:
It’s a soft wdt and when debugging the crash data it shows memory problem on softwareserial

mydakota, 2019-01-15:
What's the recommendation tested firmware versions for panel please

maragelis, 2019-01-16:
https://github.com/maragelis/ParadoxRs232toMqtt/wiki/Tested-Paradox-Alarm-Systems

tasosstr, 2019-01-24:
> 
> 
> Well....i used latest bin file 20190114a with EspEasy.
> It is normal that esp is rebooting on any json payload ?
> 
> [image]

Αγαπητέ Πάνο, Well Done, you have done a Perfect job ! ! !

I had the same issue, after 2  days i was searching, i got message from panel but when i was try to arm ... or any other command i had a message Problem login ... 

The solution, i move back the pin to Serial Hardware  (Rx - Tx) and change the lines of code from SoftwareSerial to Serial and i use Serial1 for the debug (i din't check if working this) and everything are back to normal.

I have see somewhere in the past how  to use and the 3 Hardware serial at the same time, something change in a file and then you have access to 3 at the same time, i will try to find and i will update here so to use it, because it is better from software, not loosing data :-)

Με Φιλικούς Χαιρετισμούς,
Τάσος

maragelis, 2019-01-25:
Καλημερα Τασο. 
Ευχαριστω για τα καλα σου λογια. 
I had no time lately to work on the project. The last time i checked it was a software error int the softwareserial lib, the truth is I changed it to software so I could get debug messages from the usb connection. I have this project running 1,5 years now on my Paradox with no problems, but using the hardware serial on the chip. 

I will swap back the hardware serial (If you have done the work I would appreciate a pull request), and will use the second TX on the chip for debug. (or send debug through the mqtt channel).

The link you sent is good but its for ESP32 which by default has three harware serial, the esp8266 only has one. 

Will get back as soon as possible...
Πανος

tasosstr, 2019-01-25:
Καλημέρα φίλε Πάνο,

I use a NodeMCU V3 ( https://www.teachmemicro.com/nodemcu-pinout/ ) and now i check the the Serial1 as debug on D4 pin and i am happy because it sends data after Trace=1 

I try to do Pull Request but somewhere stuck so i will paste here the changes on code i did it 

#include <FS.h>   
#include <SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266SSDP.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <DNSServer.h>
#include <PubSubClient.h>
#include <WiFiManager.h>
#include <ArduinoJson.h>

#define mqtt_server       "192.168.1.230"
#define mqtt_port         "1883"
#define Hostname          "paradoxdCTL" //not more than 15 

#define paradoxRX  13
#define paradoxTX  15

#define Stay_Arm  0x01
#define Stay_Arm2 0x02
#define Sleep_Arm 0x03
#define Full_Arm 0x04
#define Disarm  0x05
#define Bypass 0x10

#define MessageLength 37
#define LED LED_BUILTIN

#define Hassio 1
bool TRACE = 0;

const char *root_topicOut =        "paradoxdCTL/out";
const char *root_topicStatus =     "paradoxdCTL/status";
const char *root_topicIn =         "paradoxdCTL/in";
const char *root_topicArmStatus =  "paradoxdCTL/status/Arm";
const char *root_topicZoneStatus = "paradoxdCTL/status/Zone";

WiFiClient espClient;
PubSubClient client(espClient);
//SoftwareSerial paradoxSerial(paradoxRX, paradoxTX, false ,255);

bool shouldSaveConfig = false;
bool ResetConfig = false;
bool PannelConnected =false;
bool PanelError = false;
bool RunningCommand=false;
bool JsonParseError=false;

unsigned long lastReconnectAttempt = 0UL;
unsigned long ul_Interval = 5000UL;

char inData[38]; // Allocate some space for the string
byte pindex = 0; // Index into array; where to store the character
 
ESP8266WebServer HTTP(80);

struct inPayload
{
  byte PcPasswordFirst2Digits;
  byte PcPasswordSecond2Digits;
  byte Command;
  byte Subcommand;
[... 1045 lines cut ...]
        }
      }
    }
    else
    {
      trc("File /config.json doesnt exist");
      //SPIFFS.format();
      trc("Formatted Spiffs");    
    }
  } else {
    trc("failed to mount FS");
  }
}

void trc(String msg){
  if (TRACE) {
 Serial1.println(msg);
 // sendMQTT(root_topicStatus,msg);
  }
}

Will get back as soon as possible...

Με Φιλικούς Χαιρετισμούς,
Τάσος

tasosstr, 2019-01-25:
Ξεχασα η δοκιμή είναι σε SP7000 οπότε βάλτο στην λίστα του ΟΚ

maragelis, 2019-01-25:
https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/20190125

tasosstr, 2019-01-29:
Hello Dear, 

I do some cleanup :-) 

#include <FS.h>   
#include <SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266SSDP.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <DNSServer.h>
#include <PubSubClient.h>
#include <WiFiManager.h>
#include <ArduinoJson.h>

#define mqtt_server       "192.168.1.230"
#define mqtt_port         "1883"
#define Hostname          "paradoxdCTL" //not more than 15 

#define Stay_Arm   0x01
#define Stay_Arm2  0x02
#define Sleep_Arm  0x03
#define Full_Arm   0x04
#define Disarm     0x05
#define Bypass     0x10

#define MessageLength 37
#define LED           LED_BUILTIN

#define Hassio 1
bool TRACE = 0;

const char *root_topicOut        = "paradoxdCTL/out";
const char *root_topicStatus     = "paradoxdCTL/status";
const char *root_topicIn         = "paradoxdCTL/in";
const char *root_topicArmStatus  = "paradoxdCTL/status/Arm";
const char *root_topicZoneStatus = "paradoxdCTL/status/Zone";

WiFiClient espClient;
PubSubClient client(espClient);

bool shouldSaveConfig = false;
bool ResetConfig      = false;
bool PannelConnected  = false;
bool PanelError       = false;
bool RunningCommand   = false;
bool JsonParseError   = false;

unsigned long lastReconnectAttempt = 0UL;
unsigned long ul_Interval          = 5000UL;

char inData[38]; // Allocate some space for the string
byte pindex = 0; // Index into array; where to store the character

ESP8266WebServer HTTP(80);

struct inPayload {
	byte PcPasswordFirst2Digits;
	byte PcPasswordSecond2Digits;
	byte Command;
	byte Subcommand;
};

typedef struct {
	byte armstatus;
	byte event;
	byte sub_event;
	String dummy;
[... 963 lines cut ...]
			trc("Formatted Spiffs");
		}
	}
	else {
		trc("failed to mount FS");
	}
}

void trc(String msg) {
	if (TRACE) {
		Serial1.println(msg);
		// sendMQTT(root_topicStatus,msg);
	}
}

Because you use the checksum in many sub routines i do a function.

Tested and all ok 👍 

Thank you dear Panos 👍 

It is possible to check - control the PGM functions ?

Best Regard's ,
Tasos

maragelis, 2019-01-29:
Pgm is easy just have to pass a new action 0x32 pgm on and 0x33 pgm off

tasosstr, 2019-01-30:
Hello Dear Panos,

You are PERFECT ! ! ! 

Thank you ! ! !

Best Regards,
Tasos