maragelis/ParadoxRs232toMqtt #13: Spectra (Paradox) 1738

SauRoNZA, 2018-10-01:
Hi @maragelis any chance this will work on a Paradox 1738 or is that simply too old?

Also you stated you have added a Home Assistant config to the Wiki but I just don't see it. I assume the Wiki refers to one hosted here on Github and not somewhere else.

Very interested to play around with this, but would avoid the pain and trauma if I know beforehand it won't work.

maragelis, 2018-10-01:
The best way to test if it will work is just connect serial ftdi to the RX pin of the panel and sniff what you get there. if you get 37 byte binary data that starts with E0 you might be in luck.

https://github.com/maragelis/ParadoxRs232toMqtt/wiki/Hassio-Home-Assistant
this uses node-red to do the translation needed for Hassio. If i knew what payloads Hassio needed i would of coded it into the code. (havent gotten around to reverse the above which was submitted by a user)

SauRoNZA, 2018-10-01:
Perfect will give that a try.

Assuming I'm a non-dev idiot I would imagine I can plug it into Atom and use the serial monitoring interface to get that data out?

*** I see now the Home Assistant info was split into a "page" on the right side so I never saw it before.

closed 2018-10-19 by SauRoNZA

LeivoSepp, 2020-12-30:
Paradox 1738 output binary data is just 4 bytes. Just basic messages like zone open/close, arming/disarming, clock and some other messages. I am going to integrate it with my home automation.