maragelis/ParadoxRs232toMqtt #14: WeMos R3

SauRoNZA, 2018-11-01:
Hello @maragelis is there any reason this shouldn't also work on the R3 board?

Specifically this one...

https://robotdyn.com/uno-wifi-r3-atmega328p-esp8266-32mb-flash-usb-ttl-ch340g-micro-usb.html

I'm a complete retard with this and have never flashed one of these boards before so more than likely I'm just being an idiot but I get all sorts of complaints in Arduino IDE.

`/home/shaun/Desktop/ParadoxRs232toMqtt-dev/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino: In function 'inPayload Decodejson(char*)':
ParadoxAlarmSystemOTA:852: error: 'DynamicJsonBuffer' was not declared in this scope
   DynamicJsonBuffer jsonBuffer;
   ^
ParadoxAlarmSystemOTA:852: error: expected ';' before 'jsonBuffer'
   DynamicJsonBuffer jsonBuffer;
                     ^
ParadoxAlarmSystemOTA:853: error: 'jsonBuffer' was not declared in this scope
   JsonObject &root = jsonBuffer.parseObject(Payload);
                      ^
ParadoxAlarmSystemOTA:854: error: 'class ArduinoJson650_0_0::JsonObject' has no member named 'success'
   if (!root.success())
             ^
/home/shaun/Desktop/ParadoxRs232toMqtt-dev/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino: In function 'void setup_wifi()':
ParadoxAlarmSystemOTA:981: error: 'DynamicJsonBuffer' was not declared in this scope
       DynamicJsonBuffer jsonBuffer;
       ^
ParadoxAlarmSystemOTA:981: error: expected ';' before 'jsonBuffer'
       DynamicJsonBuffer jsonBuffer;
                         ^
ParadoxAlarmSystemOTA:982: error: 'jsonBuffer' was not declared in this scope
       JsonObject& json = jsonBuffer.createObject();
                          ^
ParadoxAlarmSystemOTA:991: error: 'class ArduinoJson650_0_0::JsonObject' has no member named 'printTo'
       json.printTo(Serial);
            ^
ParadoxAlarmSystemOTA:992: error: 'class ArduinoJson650_0_0::JsonObject' has no member named 'printTo'
       json.printTo(configFile);
            ^
/home/shaun/Desktop/ParadoxRs232toMqtt-dev/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino: In function 'void mountfs()':
ParadoxAlarmSystemOTA:1081: error: 'DynamicJsonBuffer' was not declared in this scope
         DynamicJsonBuffer jsonBuffer;
         ^
ParadoxAlarmSystemOTA:1081: error: expected ';' before 'jsonBuffer'
         DynamicJsonBuffer jsonBuffer;
                           ^
ParadoxAlarmSystemOTA:1082: error: 'jsonBuffer' was not declared in this scope
         JsonObject& json = jsonBuffer.parseObject(buf.get());
                            ^
ParadoxAlarmSystemOTA:1083: error: 'class ArduinoJson650_0_0::JsonObject' has no member named 'printTo'
         json.printTo(Serial);
              ^
ParadoxAlarmSystemOTA:1084: error: 'class ArduinoJson650_0_0::JsonObject' has no member named 'success'
         if (json.success()) {
                  ^
exit status 1
'DynamicJsonBuffer' was not declared in this scope
`

Maybe I should be using a different tool to upload it that I'm not aware of?

maragelis, 2018-11-01:
There is no reason it should not work, as long as the board is set for esp8266 usage. 
You are getting the above error because you are missing a library, go to arduino IDE and install the pubsubclient library.

SauRoNZA, 2018-11-01:
Yeah I did install that one and the subsequent JSON one and then I get that error out.

Probably me doing something stupid still.

maragelis, 2018-11-01:
are you using esp8266 in the board specs in the arduino ide?

SauRoNZA, 2018-11-01:
Figured it out.

You need to use ArduinoJson 5.* library instead of 6.* which is still in Beta even though it shows up at the top of the list.

Problem was seemingly all in the compile stages and nothing to do with the board.

So we learn.

But now a new issue I'll tackle tomorrow.

`collect2.exe: fatal error: cannot find 'ld'

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.`

mydakota, 2018-11-01:
Hey has anyone used the new IR Inception alarm
Hoping this may also work or closely compatible
As it uses a lot of paradox equipment across the comms. https://www.innerrange.com/files/ProductFiles/Inception%20Controller/Data%20Sheets/5644-Inception_Controller_DataSheet_EU.pdf
Note the mullagan/paradox remote receiver
Talks on bus.
I'm buying a few of these as they look amazing
And hopefully can get this working via rs232 hack

SauRoNZA, 2018-11-01:
> Figured it out.
>
> You need to use ArduinoJson 5.* library instead of 6.* which is still in Beta even though it shows up at the top of the list.
>
> Problem was seemingly all in the compile stages and [...]

Weird between reloading Board Manager and updating WiFiManager to new version it's working now.

SauRoNZA, 2018-11-02:
Uploaded just fine but the Wireless AP just doesn't show up.

maragelis, 2018-11-02:
Is it starting up or is it boot looping. What core are you using v2.3.0 is supported not v2.4.0

SauRoNZA, 2018-11-02:
Downgrading to 2.3 then it doesn't compile again.

`Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, QIO, 115200, 512K (no SPIFFS), ck, Disabled, None"

C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::setupConfigPortal()':

C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:146:71: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(), WiFiManager* const>::type)'

   server->on(String(F("/")), std::bind(&WiFiManager::handleRoot, this));

                                                                       ^

C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:146:71: note: candidates are:

In file included from C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.h:17:0,

                 from C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:13:

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, THandlerFunction handler);

        ^

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn);

        ^

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);

        ^

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided

C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:147:81: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(unsigned char), WiFiManager* const, bool>::type)'

   server->on(String(F("/wifi")), std::bind(&WiFiManager::handleWifi, this, true));

                                                                                 ^

C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:147:81: note: candidates are:

In file included from C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.h:17:0,

                 from C:\Users\user\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:13:

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, THandlerFunction handler);

        ^

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn);

        ^

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided

[... 172 lines cut ...]
C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn);

        ^

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);

        ^

C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Maybe you can advise exactly which version of each package/library should be used?

maragelis, 2018-11-05:
just download wifimanager lib in th e arduino ide

maragelis, 2018-11-06:
get the bin file (precompiled) and just flash it to the ESP
https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/V1

SauRoNZA, 2018-11-06:
> get the bin file (precompiled) and just flash it to the ESP using esptool
> https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/V1

Aha! How did I miss that even when I was looking for it.

Presumably still using Arduino IDE or should I use another tool?

maragelis, 2018-11-06:
just give me a few hours its not uploading (I am on mobile network) will upload tonight

SauRoNZA, 2018-11-06:
> just give me a few hours its not uploading (I am on mobile network) will upload tonight

No super rush, I'm in your debt for making the effort so will check again tonight.

SauRoNZA, 2018-11-07:
Finally had success loading it onto the R3 after a bit of up and down.

Tell me your MQTT configuration doesn't seem to support authentication with a username/password? Or did I miss it somehow?

maragelis, 2018-11-07:
No i dont use auth.

SauRoNZA, 2018-11-24:
I give up, I'm convinced the serial output on my Spectra is either broken or disabled.

Got a WeMoS D1 R2 and flashed to it without an issue, connected it to network see the initial status that it comes online from MQTT messages so that's working.

And then nothing else ever happens.

maragelis, 2018-11-24:
are you sure you dont have the rx tx the connected wrong. what spectra model do you have

SauRoNZA, 2018-11-25:
> are you sure you dont have the rx tx the connected wrong. what spectra model do you have

Yup thought the same and swopped them around with no change in behaviour.

It's the Spectra 1738.

istvanszab, 2018-12-02:
I use auth for MQTT.
I dont't understand your code about auth.
"// If you  want to use a username and password, uncomment next line and comment the line if (client.connect("433toMQTTto433")) {
    //if (client.connect("433toMQTTto433", mqtt_user, mqtt_password)) {
    // and set username and password at the program beginning"

If I uncomment "if (client.connect("433toMQTTto433", mqtt_user, mqtt_password)) {" where I close the if statement?

maragelis, 2018-12-03:
Change this line from 
    if (client.connect(charBuf,root_topicStatus,0,false,"Paradox Disconnected")) {
to
    if (client.connect(charBuf, username, password, root_topicStatus,0,false,"Paradox Disconnected")) {

closed 2019-01-06 by maragelis