maragelis/ParadoxRs232toMqtt #16: IP150 & D1 mini

istvanszab, 2018-12-05:
I have SP6000 with IP150 module.
I connect parallel to the serial port the IP150 and the D1 mini. with your code.
Now the D1 mini works, but the IP150 doesn't.
The leds seems to be OK (Internet, Link, RX/TX) on the IP150 but the panel says connection error.
Do you have an idea of the mistake?

cybertza, 2018-12-05:
Hi,

The 2 should work ok together in my opinion, I know people have done it before.

Albeit you may have better luck using the code for the IP150 module to Mqtt ?

istvanszab, 2018-12-05:
Hi,
no I can't use it. My IP150 has 4.1 firmaware.

istvanszab, 2018-12-06:
When I power the D1 mini, mqtt sends:
/paradox/status: Paradox Connected
/paradox/status: ParadoxController V2.11
/paradox/out: { "armstatus":225, "event":44, "sub_event":19, "dummy":""}
/paradox/out: { "armstatus":225, "event":44, "sub_event":18, "dummy":""}
/paradox/out: { "armstatus":225, "event":2, "sub_event":10, "dummy":"House"}
/paradox/out: { "armstatus":224, "event":48, "sub_event":3, "dummy":""}
/paradox/status: panel logout
/paradox/out: { "armstatus":224, "event":2, "sub_event":16, "dummy":"House"}

After this messages I lost the IP150 module.

cybertza, 2018-12-06:
Ok well I think there is a panel log void, maybe see if it gets called, and also an panel log in, if you somehow skip those and purely listen it should work?

Regards

istvanszab, 2018-12-06:
I'm sorry but I don't understand it.
The panel means the SP6000 or the TM50 keypad?
From the keypad I can log in, arm, disarm, etc.
I don't have BabyWare, NeWare or else software. Can I do what you wrote without those softwaress?

maragelis, 2018-12-10:
There is no way for two devices to read and write at the same time on one serial port.
You will have to create a function in the code to write the serial data to another port and have the IP150 read it from that serial output. In the event of input the code will have to send the binary data back to the panel serial port.
You could disable the write event in my code and just get the event to mqtt. 
If the wemos is working for you (arm and events) why do you want the IP150????

maragelis, 2018-12-10:
/paradox/out: { "armstatus":225, "event":44, "sub_event":19, "dummy":""} IP module supervision loss
/paradox/out: { "armstatus":225, "event":44, "sub_event":18, "dummy":""} IP module no service
/paradox/out: { "armstatus":225, "event":2, "sub_event":10, "dummy":"House"} Ground start (partition 1 only)
/paradox/out: { "armstatus":224, "event":48, "sub_event":3, "dummy":""} Software log off 
/paradox/status: panel logout
/paradox/out: { "armstatus":224, "event":2, "sub_event":16, "dummy":"House"} Report confirmation

istvanszab, 2018-12-11:
Thank you. I'll give it a try after components purchasing.
I want the IP150 because my wife :) (she loves Insite Gold) and because the patrol service.
And I need only to observe the motion sensors status.

closed 2019-01-31 by maragelis