maragelis/ParadoxRs232toMqtt #17: Wemos d1 mini pro losing connection to MQTT

rjduraocosta, 2018-12-12:
Hello @maragelis,

I have been runnig this wonderfull piece of software for a while now but I have been having sometimes the following issue:
I noticed that the wemos get's disconnected from the MQTT brooker. I try to send a command and the alarm panel does not respond. The last status on the MQTT brooker is Paradox Disconnected. Sometimes when it is in this state, Paradox Disconnected, it does not change to anything else and I cannot issue any command. If I do something like passing in front of a sensor or arm/disarm the panel in the local keypad, it seems that the wemos connection to the MQTT brooker "wakes up" and everithyng is up and running again.

Could you please try and give me a hand on this subject.

Thanks in advanced for all your help and support.

Best regards.

maragelis, 2018-12-12:
Had this problem but was my wifi disconnecting check you rssi signal strength mine was border line because of the metal box the alarm system was in. Got myself a wemos pro and used an external antenna and all was good.

rjduraocosta, 2018-12-13:
Hello again @maragelis ,

It makes perfect sence. My d1 mini pro is already on it's way.

Thanks again for all your help and support.

Best regards.

closed 2018-12-13 by rjduraocosta

cybertza, 2018-12-13:
Hey I know this topic is closed allready, but I wonder if there would be any point in creating an /tele/rssi and /tele/status topics

Status taking advantage of the last Will msg part of the MQTT Broker aka connected|disconnected for use in home assistant for showing if the resource is available

rjduraocosta, 2018-12-13:
Hello @cybertza,

That is actually a great idea. 

What do you think @maragelis? Could this be something easy to do?

maragelis, 2018-12-13:
You can also try the test and dev branches that have status messages implemented. But will have to use different pins for rx/tx as there I have used software serial so to free up the serial port for debug.

rjduraocosta, 2018-12-14:
> Last will message is implemented already.” paradox disconnected “ is the last will message, and paradox connected, is the connection message. Due to the rate at which the data is coming in through the serial port there is no time to send other messages. There is no interrupt on serial, so you have to listen continuously for messages. I have tried sending other status messages but at the expense of missing important messages coming in through the serial. I have started a project trying to port this code to python so to use a raspberry or esp32. I myself like the idea of a raspberry zero with Wi-Fi , with a cheap gsm modem for sms fall back in the event you loose Wi-Fi

This is a great idea and it will turn out to be a very nice project I am sure. I will be glad to "take it for a spin" if you need to test it in another panel.

> You can also try the test and dev branches that have status messages implemented. But will have to use different pins for rx/tx as there I have used software serial so to free up the serial port for debug.

Thanks for the suggestion, I will give it a go also.