maragelis/ParadoxRs232toMqtt #18: 2 dead wemos

peterkexas, 2018-12-18:
Hi I am trying to connect wemos d1 mini to Paradox sp 5500 alarm panel to serial and they burn.
What I am doing wrong?

maragelis, 2018-12-19:
can you upload a picture with the connections

peterkexas, 2018-12-19:
I am out right now but will see it tonight again with a multimeter.

peterkexas, 2019-01-02:
Hello again!
I connect my wemos to my sp5500.
I have a mqtt server running on my rpi setup with node red on home assistant but when I try to arm , disarm nothing happened.
Eny help?
Wemos flashed with bin file that I download from here.
Thank you and sorry for my bad English!

maragelis, 2019-01-02:
First check the mqtt topic that you are getting messages this can be done by either using node-red debug node or using an mqtt client. 
1) after flashing did you change the mqtt server ?
2) are you getting connect message in mqtt status topic.

peterkexas, 2019-01-02:
I have change the mqtt server and my wifi credentials. I am trying to debug with node red but nothing happened. I was trying with paradox/# topic and nothing again. Sorry I am very new to this.

maragelis, 2019-01-02:
Try #  or /#

peterkexas, 2019-01-02:
I am trying to send you pm and i can not find it anyway.
With  /#  in Nodered I get massages from paradox/in but nothing from paradox/out.

maragelis, 2019-01-02:
Try changing serial lines rxtx you might have them the wrong way

mydakota, 2019-01-02:
Thanks I have tried that doesn't seem to make much difference. I am using a evo40 main board so when I get home I'll try my MG5050 see if it's the same I suspect it will be as ive herd others have used the Evo panels ok

mydakota, 2019-01-02:
Omg sorry jumping on a Sim problem sorry thought you were talking to me lol

peterkexas, 2019-01-03:
I double check them panel rx to d1 tx , panel tx to d1 rx and ground. On serial monitor I can see that connects to wifi but nothing for mqtt server, is that right?

peterkexas, 2019-01-04:
I check debug from nodered i see only /paradox/in and when restart my wemos i have this : 

1/4/2019, 6:56:42 PMnode: 7cb71166.730df
/paradox/status : msg.payload : string[17]
"Paradox Connected"

1/4/2019, 6:56:44 PMnode: 7cb71166.730df
/paradox/status : msg.payload : string[23]
"ParadoxController V2.11"
I download mqtt.fx and i have the same messages
Have you any idea to help me?
thank you!

maragelis, 2019-01-04:
thats correct you are working fine, are you not getting events on paradox/out ?

peterkexas, 2019-01-04:
No!! nothing.

peterkexas, 2019-01-05:
I have to flash blank.bin file first or flash it directly?

maragelis, 2019-01-05:
yes flash blank.bin

peterkexas, 2019-01-05:
nothing again flash v2 bin connects to my mqtt broker and wifi network but i dont have any message to /paradox/out

maragelis, 2019-01-05:
my friend, v2 is at paradoxCTL/#

peterkexas, 2019-01-05:
Sorry I am stupid.
But trying with paradoxCTL/# and pins 13 15 again nothing.
I am trying to copy your code for nodered and import to flow and has error so I can't importing it.

peterkexas, 2019-01-05:
Thank you!!  thank you !!!
I have done it , managed to take messages from my panel on paradoxdCTL/out now another hard part to connect it with my home assistant via Nodered.
Thanks again for your help!!

mydakota, 2019-01-05:
Good job ,
I'm interested on your node red useage to intergrate home assistant and how that works.
I've just changed some out put messages in code to remove all the " " symbols that home assistant has a hard time with example below

paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01

This now works for arm disarm zone open closes restors and all the other out messages 
I just use a simple motion binary_sensor
In home assistant works a treat . But I'm still testing and need to verify this will not cause other issues

peterkexas, 2019-01-05:
Thank you, if you can send me your code to test it on my setup.
You have to install nodered to machine runs home assistant and then install home assistant pallet and then connect them.
An example for this on how it works we have here paradox232tomqtt on wiki-pages hassio home assistant section

mydakota, 2019-01-06:
Cool the main prob that home assistant had with these json message Hass didn't like more than two " " symbols anymore you have problems . Ive tryed templating to get things to work but ends up extramly complicated and unreliable. This way works very well and very simple solution .

maragelis, 2019-01-06:
reading the hassio binary sensor https://www.home-assistant.io/components/binary_sensor.mqtt/ you could use:
 value_template: '{{ value.x }}'  

in the case of my project you would use 
value_template: '{{value_json.event}}'
payload_on: "1"
payload_off: "0"

mydakota, 2019-01-06:
I personally had a hard time getting the template or value template to work with json but I'm  no HASS expert either. It will definitely be a good addition for this project to support Hass as it's got a large community. 

Slightly changing subject but I've been switching my comms between my MG5050 
and Evo 48 panel unfortunately I get no message from the Evo board I'm assuming it's got a different bit rate or something different I do know the Evo uses 24 digit headers dose that sound like it's not compatible have you talked to anyone with success with the EVO systems thanks.

maragelis, 2019-01-06:
Evo has different message dynamic from 20 to (i dont know)

mydakota, 2019-01-06:
Ah ok ;( my fav Paradox Panels are the Evos but hay with some exspanders and a couple of TM70 Keypads I'll still be happy with the MG5050 I guess. Thanks for the info

maragelis, 2019-01-06:
I have created my own tm70 keypads using this project and openhab/habpanel looks good and works great, and also intergrated lights and heating . 
Just used cheap arduino tablets (60euros)

peterkexas, 2019-01-06:
thanks you saved me!!!!

maragelis, 2019-01-06:
Ok new release is up Hassio and updated test branch

maragelis, 2019-01-06:
just to get the blood running 

[image]

peterkexas, 2019-01-07:
Hi!!  Thanks very much ! V2 works like a charm.
Trying to figure out what is the command_topic to change the state !. The sensors work's fine for me but I can't arm or disarm from hass

maragelis, 2019-01-07:
Check the wiki hassio v2 you will get your answer

peterkexas, 2019-01-08:
:(  ?

maragelis, 2019-01-09:
larm_control_panel:

platform: mqtt
state_topic: “paradoxdCTL/out/state”
payload_disarm: ‘{"password":“0000”,“Command”:“disarm”,“Subcommand”:"0"}’
payload_arm_night: ‘{"password":“0000”,“Command”:“sleep”,“Subcommand”:"0"}’
payload_arm_home: ‘{"password":“0000”,“Command”:“stay”,“Subcommand”:"0"}’
payload_arm_away: ‘{"password":“0000”,“Command”:“arm”,“Subcommand”:"0"}’

peterkexas, 2019-01-09:
I saw that and add it to my configuration and I have state of my alarm but i can't arm or disarm the system.I must add command_topic to configuration to control the alarm.
Sorry for all that but I am trying to learn I have no idea about code and how works.
Thanks again!!

mydakota, 2019-01-11:
Doesn't seem to matter what I do or what I try 
I can't arm disarm or any commands to panel via the paradoxdCTL/in .
I'm not sure what else to try mqtt messages definitely being sent just now response from alarm . Just brought the TM70 keypad and if I arm disarm via the keypad I get all armed states back via mqtt so frustrated lol

mydakota, 2019-01-11:
Also don't you mean the command topic is
paradoxdCTL/in
Not paradoxCTL/in

mydakota, 2019-01-11:
Yip receiving very well I can connect with IP150 via same port so it's got to be something related to the wemos. maybe I'll try change some user codes but I don't feel it's likely the issue as I can arm disarm via keypad ip150

maragelis, 2019-01-11:
are you disconnecting the ip150 .
could you send all events recieved by mqtt when arming the system using the keypad . 
I want to see if you are recieving the event 48 sub 02 thats the software login event

mydakota, 2019-01-11:
I definitely have ip150 disconnect when wemos on I'm fairly sure I'm getting those codes but give me a min I'll go arm trigger entry delay siren and disarm past you the output soon

mydakota, 2019-01-11:
hasbian@hasbian-desktop:~$ mosquitto_sub -v -t '#'
paradoxdCTL/out { armstatus:224, event:2, sub_event:14, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:3, sub_event:2, dummy:}
paradoxdCTL/out { armstatus:224, event:2, sub_event:8, dummy:     Area 1     }
paradoxdCTL/status {"status":"Problem connecting to panel"}
paradoxdCTL/out { armstatus:224, event:2, sub_event:9, dummy:     Area 1     }
paradoxdCTL/out/state armed_away
paradoxdCTL/out { armstatus:224, event:2, sub_event:12, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:29, sub_event:2, dummy:     Area 1     }
paradoxdCTL/out/zone1 ON
paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:13, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:2, sub_event:6, dummy:     Area 1     }
paradoxdCTL/out/state triggered
paradoxdCTL/out { armstatus:224, event:2, sub_event:4, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:36, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out/state disarmed
paradoxdCTL/out { armstatus:224, event:2, sub_event:11, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:2, sub_event:7, dummy:     Area 1     }
paradoxdCTL/status {"status":"Problem connecting to panel"}
paradoxdCTL/out { armstatus:224, event:3, sub_event:3, dummy:}
paradoxdCTL/out { armstatus:224, event:38, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:33, sub_event:2, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:32, sub_event:2, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:8, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:2, sub_event:9, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:2, sub_event:8, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:2, sub_event:9, dummy:     Area 1     }

mydakota, 2019-01-11:
paradoxCTL/in {"password":"1980","Command":"stay","Subcommand":"0"}
paradoxCTL/in {"password":"1980","Command":"arm","Subcommand":"0"}
paradoxdCTL/in {password:1234,Command:arm,Subcommand:0}
paradoxdCTL/in {password:1234,Command:arm,Subcommand:0}
paradoxCTL/in {"password":"1980","Command":"stay","Subcommand":"0"}
paradoxdCTL/in  {password:1980,Command:stay,Subcommand:0}

*WM: server
*WM: Adding parameter
*WM: port
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.107
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.107
*WM: freeing allocated params!
Starting HTTP...
Starting SSDP...
returned command = 4
returned command = 4
returned command = 4
returned command = 4
returned command = 1

maragelis, 2019-01-11:
ok try this topic 
paradoxdCTL/in 
 {"password":"1980","Command":"stay","Subcommand":"0"}

maragelis, 2019-01-11:
paradoxCTL/in {"password":"1980","Command":"stay","Subcommand":"0"}
paradoxCTL/in {"password":"1980","Command":"arm","Subcommand":"0"}
paradoxdCTL/in {password:1234,Command:arm,Subcommand:0}  <-- Wrong Topic 
paradoxdCTL/in {password:1234,Command:arm,Subcommand:0} <-- not strings
paradoxCTL/in {"password":"1980","Command":"stay","Subcommand":"0"} <-- Wrong Topic 
paradoxdCTL/in {password:1980,Command:stay,Subcommand:0} <-- Not strings 
paradoxdCTL/in {"password":"1980","Command":"stay","Subcommand":"0"} <-- CORRECT

mydakota, 2019-01-11:
pub=
`hasbian@hasbian-desktop:~$ mosquitto_pub -h 192.168.1.100 -t paradoxdCTL/in -m "{"password":"1980","Command":"arm","Subcommand":"0"}"`

sub =
`paradoxdCTL/in {password:1980,Command:arm,Subcommand:0}
`
serial mon=
`returned command = 4
`

maragelis, 2019-01-11:
the only thing we can do is to set trace to 1 and get more debug messages

maragelis, 2019-01-11:
i see the problem just wait a min will get back soon

mydakota, 2019-01-11:
Cool I have to go sleep lol 12pm here now got cameras to install in morning need some rest lol. Thanks for your help much appreciated and if you find issue post I'll see in morning and have a go thanks again. Also keen to try trace set to 1 I'm picking that's in Arduino code b4 uploading or can I command from serial interface .....

maragelis, 2019-01-11:
try new  realese 20190111

maragelis, 2019-01-14:
https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/20190114a

maragelis, 2019-01-25:
https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/20190125

peterkexas, 2019-02-02:
Καλησπέρα Πάνο!!
[image]
I have no idea what to do with that!
Can you help me?

tasosstr, 2019-02-02:
Καλησπέρα, έχεις συνδέσει τον πίνακα με το  D7 , D8 ?

peterkexas, 2019-02-02:
Ναι

tasosstr, 2019-02-02:
[image]

Σύνδεση σε SP7000 το ίδιο είναι και για SP6000, SP5500

tasosstr, 2019-02-02:
Είχα το ίδιο πρόβλημα και τελικά ήταν αλλαγή στην Serial port φαντάζομαι χρησιμοποιής την τελευταία έκδοση https://github.com/maragelis/ParadoxRs232toMqtt/

Έχεις βάλει την IP και την πόρτα για το MQTT Server ?

tasosstr, 2019-02-02:
[image]

Αυτή είναι η τελική σύνδεση πάνω στον πίνακα, με το converter τάσης απο >12V σε 5V για τον controller

peterkexas, 2019-02-02:
Έχω συνδέσει σωστά περνώ σήματα από τον πίνακα στον mqtt. Όταν προσπαθώ να του στείλω arm_away η απάντηση που έχω είναι αυτό τις φωτογραφίας παραπάνω.

tasosstr, 2019-02-02:
Νομίζω στέλνεις λάθος command:

paradoxdCTL/out/state gives values:

disarmed
armed_home
armed_away
triggered

--------------------

Τα command που στέλνουμε είναι 

json Command payload template
{ "password":"1234", "Command":"arm", "Subcommand":"0" }

password is user 4 digit password

Command can be one of the following

arm,
disarm,
sleep,
stay,
bypass,
armstate,
zonestate,
panelstatus

subcomand depends on command ,

if Command is arm,sleep,disarm subcomand is partition

if Command is bypass subcomand is zone number from 0 to 31

if Command is panelstatus subcomand 0 = panel data
or subcomand 1 = panel voltage and battery data 

'Αρα δοκίμασε να στείλεις 
{ "password":"1234", "Command":"arm", "Subcommand":"0" }

όπου 1234 θα βάλεις τον κωδικό σου που οπλίζεις - αφοπλίζεις τον συναγερμό σου.

[48 more replies]

closed 2019-02-05 by maragelis