maragelis/ParadoxRs232toMqtt #19: OTA

mydakota, 2018-12-23:
Hi guys,
I’ve uploaded my code and changed setup wifi and MQTT address I can see the device on my network ip 192.168.1.9 now if I try browsing to that address it fails to connect.
Is this normal as I was thinking it should log me into the same setup page after first flashing it . Maybe my arduino library’s are out of date not sure .

mydakota, 2018-12-23:
{"mqtt_server":"192.168.2.230","mqtt_port":"1883"}*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.98
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.98

mydakota, 2018-12-23:
Also I noticed the matter is hasn't changed from 192.168.2.230 to my 192.168.197 server.
I've tried changing it in code then uploading it but it won't change I can't access web page to change it either seems very weird please help

mydakota, 2018-12-24:
```

⸮Dp⸮tAhlE⸮⸮CXH⸮⸮{"mqtt_server":"192.168.1.197","mqtt_port":"1883"}*WM: Adding parameter
*WM: server
*WM: Adding parameter
```
*WM: port
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.98
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.98
```

blanked the flash of d1min and reloaded and config properly but this is my output via serial mon.
d1 mini is connected to paradox serial port as well rx and tx are correctly reversed.

maragelis, 2018-12-24:
Use arduino core 2.3.0 and you should be ok. Had those issues with 2.4

mydakota, 2018-12-24:
Hi thanks for reply ,
I'm a bit new to Arduino so to change the Arduino core to 2.3.0 do I do that I the library update section . Sorry to ask such a noobish question.

SauRoNZA, 2018-12-27:
> Use arduino core 2.3.0 and you should be ok. Had those issues with 2.4

Does that fix the problem with not having access to the web interface once it's connected to Wifi? Or is that the standard behaviour?

maragelis, 2018-12-27:
There is no web interface

mydakota, 2018-12-28:
Ok I've change core files still the same but sounds like from what maragelis says theres no web interface . And the only work around to fix the incorrect mqtt Ip address was to reflash the esp with a blank.bin otherwise it would just rember the old changes . So far I've had no indication of comms between my paradox board and esp but ATM I'm running with out a keypad so maybe a issue . I'll have to wait untill I can pick up a keypad for testing . Thanks for all the help and getting me on the correct core files big help ....

maragelis, 2018-12-28:
just turn off you wifi router the AP will come up and you can change the mqtt server

mydakota, 2018-12-28:
Omg brilliant that thanks lol

closed 2019-01-01 by mydakota