maragelis/ParadoxRs232toMqtt #20: Homeassistant help

mydakota, 2019-01-02:
Hi I have not tested as yet but i am not 100 percent sure how to add JSON commands into MQTT switches in  home-assistant if that's what some are using for arm disarm i would also like some help in config of binary sensors to get zone open close state changes ill post the code i have written please help me fix any mistakes i am sure once i get the first ones working ill be off . thanks in advance...
maybe someone could share there home-assistant config files switch and binary any other please....

switch:
  - platform: mqtt
    name: "Arm/DisarmTest"
    state_topic: "/paradox/status"
    command_topic: "/paradox/in"
    payload_on: {"password":"1980","Command":"arm","Subcommand":"0"}
    payload_off: {"password":"1980","Command":"Disarm","Subcommand":"0"}
    state_on: "arm"
    state_off: "disarm"
    optimistic: false
    qos: 0
    retain: true

maragelis, 2019-01-02:
i suggest you head over to the homeAssistant forum for help as I myself dont use homeAssistant and dont know anything on how it works.

mydakota, 2019-01-02:
Ok cool thanks I'll have a look, if I don't get anyware I can always fall back to the python script for ip150 :(
But I will give it good go getting it working in Hass and if I do make progress I'll share my configs for others . Thanks for this project really cool work.

closed 2019-01-02 by mydakota