maragelis/ParadoxRs232toMqtt #21: comms help

mydakota, 2019-01-02:
no info coming from panel even after sending JSON commands .

serial output from console Arduino below  does this look correct as far as iot put

`{"mqtt_server":"192.168.1.100","mqtt_port":"1883"}*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.112
*WM: freeing allocated params!
_ctx: sys 
sp: 3fffeff0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3ffff1a0:  00000000 00000000 00000000 3fff045c  
3ffff1b0:  00000001 00000019 00000001 00000030  
3ffff1c0:  00000000 4bc6a7f0 9cac0831 00000000  
3ffff1d0:  00000000 00000000 4bc6a7f0 00000000  
3ffff1e0:  3ffefa08 000003bd 000003bd 4010020c  
3ffff1f0:  31996bb3 00418937 31996bb3 00000000  
3ffff200:  00000000 00000000 00418937 00000000  
3ffff210:  00000002 00000000 401067d9 4bc6a7f0  
3ffff220:  00000000 00000000 3fff0934 4020bced  
3ffff230:  0eae6fc0 00000001 3fff0934 4020bd65  
3ffff240:  7fffffff 3ffea52c 3ffef59c 00000001  
3ffff250:  00000019 00000024 3ffef59c 4020f625  
3ffff260:  0000003c 00000001 00002710 40215a72  
3ffff270:  40201d21 00000000 3ffef55c 40215ab5  
3ffff280:  00000001 00000024 3ffef804 00000001  
3ffff290:  00000019 00000024 3ffef804 40207cc9  
3ffff2a0:  3ffef55c 00000001 3ffff30f 4021737c  
3ffff2b0:  00000010 00000001 0000000f 3ffef55c  
3ffff2c0:  00000038 00000001 0000000f 40106889  
3ffff2d0:  3ffff30e 00000000 3ffef55c 4020a1d0  
3ffff2e0:  00000001 00000000 3ffff30d 402145ad  
3ffff2f0:  00000001 00000001 3ffe94ed 3ffe94ee  
3ffff300:  3ffef55c 00000001 3ffe94ed 4021737c  
3ffff310:  3ffef500 00000001 3ffe8ed9 00000000  
3ffff320:  3ffefa08 000003bd 000003bd 4010020c  
3ffff330:  3ffefa08 0000031e 0000031e 4010020c  
3ffff340:  00000019 00000000 3ffff370 401006dc  
3ffff350:  00000019 00000080 3ffef55c 402083c1  
3ffff360:  3fff1818 00000080 3ffef55c 40209f15  
3ffff370:  40201f6d 00000001 00000000 00000100  
3ffff380:  00048019 00000000 00000000 00000001  
3ffff390:  00000019 00000080 3ffef804 00000001  
3ffff3a0:  00000019 00000080 3ffef804 4020820b  
3ffff3b0:  00000000 00000000 00000000 00000000  
3ffff3c0:  00000000 00000000 00000000 00000000  
3ffff3d0:  00000000 00205f00 00000000 00000000  
3ffff3e0:  00000000 00000000 00000000 00000000  
3ffff3f0:  00000000 00010000 3ffe8000 00000001  
3ffff400:  0000000b 00000000 0000000b 40209fbd  
3ffff410:  00000000 00000000 00000000 4020f18b  
3ffff420:  007d0043 00000000 3fff17c4 0000003f  
3ffff430:  00000034 00048019 0000000b 40215ab5  
3ffff440:  7fffffff 000c97a2 3ffef59c 40216abc  
3ffff450:  00000019 000c97a2 3ffef59c 4020f770  
3ffff460:  3ffe9c01 00000001 00002710 40215a72  
3ffff470:  00000004 3ffef59f 00000043 40215ab5  
3ffff480:  00000001 00000024 00000000 00000001  
3ffff490:  00000019 00000024 3ffef804 40207cc9  
3ffff4a0:  3ffef55c 00000001 3ffff50f 4021737c  
3ffff4b0:  00000010 00000001 0000000f 3ffef55c  
3ffff4c0:  00000038 00000001 0000000f 40106889  
3ffff4d0:  3ffff50e 00000000 3ffef55c 4020a1d0  
3ffff4e0:  00000001 00000000 3ffff50d 402145ad  
3ffff4f0:  00000001 00000001 3ffe94ed 3ffe94ee  
3ffff500:  3ffef55c 00000001 3ffe94ed 4021737c  
3ffff510:  3ffef500 00000001 3ffe8ed9 00000000  
3ffff520:  3ffefa08 000003bd 000003bd 4010020c  
3ffff530:  3ffefa08 00000315 00000315 4010020c  
3ffff540:  00000019 00000000 3ffff570 401006dc  
3ffff550:  00000019 00000080 3ffef55c 402083c1  
3ffff560:  3fff17d0 00000080 3ffef55c 40209f15  
3ffff570:  40201f6d 00000001 00000000 00000100  
3ffff580:  00048019 00000000 00000000 00000001  
3ffff590:  00000019 00000080 3ffef804 00000001  
3ffff5a0:  00000019 00000080 3ffef804 4020820b  
3ffff5b0:  00000000 00000000 00000000 00000000  
3ffff5c0:  00000000 00000000 00000000 00000000  
3ffff5d0:  00000000 00205f00 00000000 00000000  
3ffff5e0:  00000000 00000000 00000000 00000000  
3ffff5f0:  00000000 00010000 3ffe8000 00000001  
3ffff600:  0000000b 00000000 0000000b 40209fbd  
3ffff610:  00000000 00000000 00000000 4020f18b  
3ffff620:  007d0043 00000000 3fff177c 0000003f  
3ffff630:  00000034 00048019 0000000b 40215ab5  
3ffff640:  7fffffff 000c8bde 3ffef59c 40216abc  
3ffff650:  00000019 000c8bde 3ffef59c 4020f770  
3ffff660:  3ffe9c01 00000001 00002710 40215a72  
3ffff670:  00000004 3ffef59f 00000043 40215ab5  
3ffff680:  00000001 00000024 00000000 00000001  
3ffff690:  00000019 00000024 3ffef804 40207cc9  
3ffff6a0:  3ffef55c 00000001 3ffff70f 4021737c  
3ffff6b0:  00000010 00000001 0000000f 3ffef55c  
3ffff6c0:  00000038 00000001 0000000f 40106889  
3ffff6d0:  3ffff70e 00000000 3ffef55c 4020a1d0  
3ffff6e0:  00000001 00000000 3ffff70d 402145ad  
3ffff6f0:  00000001 00000001 3ffe94ed 3ffe94ee  
3ffff700:  3ffef55c 00000001 3ffe94ed 4021737c  
3ffff710:  3ffef500 00000001 3ffe8ed9 00000000  
3ffff720:  3ffefa08 000003bd 000003bd 4010020c  
3ffff730:  3ffefa08 0000030c 0000030c 4010020c  
3ffff740:  00000019 00000000 3ffff770 401006dc  
3ffff750:  00000019 00000080 3ffef55c 402083c1  
3ffff760:  3fff1788 00000080 3ffef55c 40209f15  
3ffff770:  40201f6d 00000001 00000000 00000100  
3ffff780:  00048019 00000000 00000000 00000001  
3ffff790:  00000019 00000080 3ffef804 00000001  
3ffff7a0:  00000019 00000080 3ffef804 4020820b  
3ffff7b0:  00000000 00000000 00000000 000000⸮⸮iC2⸮

mydakota, 2019-01-02:
Also note I'm running esp8266 core files 2.3.0
And using WiFi manager file from download zip I also get /paradox/status paradox disconnected , connected, paradox controller v2.11

maragelis, 2019-01-03:
try disconnecting the usb serial connection as this is tired to the rx/tx.
Are you getting data from the panel to mqtt

maragelis, 2019-01-03:
Have uploaded v2 try that one its more stable and has many more options go for the debug release,
Only change to hadware is the serial moved to io 13 and 15

mydakota, 2019-01-03:
Ok thanks for all your help I'll give that a go today.

mydakota, 2019-01-05:
ok thanks for the version2,
in testing with the .bin file i get inputs and output data correctly ready from alarm board so now i know it works. i still have probs with the ParadoxAlarmSystemOTA.ino file it uploads without errors but responds differently in serial Mon  and doesn't read data or send data to the panel .
Ive reinstall Arduino on my Ubuntu system and had some library missing errors that i updated from the library manager eventually uploads without errors and same thing no comms and dosnt load web page that displays paradoxdCTL text..

my guess is i am stuffing up my library files when updating them can i ask if there are any special files i should be using in m Arduino library would it be possible to ask for a zip copy of your arduino libary files so i can be sure i am running correct verions.
as altho the bin file works I need to make some custom changes to json outputs for my homeassistant setup and cant do that with .bin .

thanks for any help .......

mydakota, 2019-01-05:
Ok update seems my Librarys are fine what I did was use the nodmcu flashing utility and flash 4x 1M bin files to blank the unit . After blanking it and re-uploaded from the ParadoxAlarmSystemOTA.ino using Arduino software I now get events from alarm panel I haven't managed to get the system to arm via mqtt commands but big progress for me.

I would suggest any others having issues reading data from alarm would be flash blank bin files to wipe your unit.

mydakota, 2019-01-05:
Sorry I'm a bit of a Arduino noob how do I use the trace bin do I add into command or via serial console in Arduino lol I've looked over the wiki to avoid asking stupid question I also have no idea mqfx lol.
Please don't laugh 😂

maragelis, 2019-01-05:
Ok flash the paradoxalarmsystemota.Trace.bin when done open arduino ide and top right hand corner there is an icon that says open serial monitor.  There you will see debug messages

maragelis, 2019-01-05:
Use https://mqttfx.jensd.de to send and receive mqtt messages

mydakota, 2019-01-05:
Ok cool thanks 
I haven't seen the paradoxalarmsystemota.Trace.bin 
I'll have a look when I get home must be a whole section I've missed on GitHub . Thanks for the info much appreciated.

mydakota, 2019-01-05:
OK cool I haven't as yet used the mqtt fx but i normally just us the terminal and sub to topic with "#"
 below is some of the read outs from my sub looks like im getting some arms and activation from me opening and closing zone 1 programed as entry delay . i still get drop offs and disconnection or unresponsive but i am still testing .

```
paradoxdCTL/out { armstatus:224, event:0, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:7, dummy:     Area 1     }
paradoxdCTL/in {"password":"1980","Command":"disarm","Subcommand":"0"}
paradoxdCTL/out { armstatus:224, event:0, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:38, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:13, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:0, sub_event:1, dummy:Zone 01         }
paradoxdCTL/in {"password":"1980","Command":"arm","Subcommand":"0"}
paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:6, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:2, sub_event:4, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:36, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:0, sub_event:1, dummy:Zone 01         }
paradoxdCTL/in {"password":"1980","Command":"disarm","Subcommand":"0"}
paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:0, sub_event:1, dummy:Zone 01         }
paradoxdCTL/in {"password":"1980","Command":"arm","Subcommand":"0"}
paradoxdCTL/status {"status":"Paradox connected"}
paradoxdCTL/in {"password":"1980","Command":"disarm","Subcommand":"0"}
paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01         }
paradoxdCTL/in {"password":"1980","Command":"arm","Subcommand":"0"}
paradoxdCTL/out { armstatus:224, event:0, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:7, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:38, sub_event:1, dummy:Zone 01         }
paradoxdCTL/in {"password":"1980","Command":"arm","Subcommand":"0"}
paradoxdCTL/out { armstatus:224, event:1, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:13, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:0, sub_event:1, dummy:Zone 01         }
paradoxdCTL/out { armstatus:224, event:2, sub_event:6, dummy:     Area 1     }
paradoxdCTL/out { armstatus:224, event:2, sub_event:4, dummy:     Area 1     }
paradoxdCTL/status {"status":"Problem connecting to panel"}

```
if i reboot wemos status reconnects and pulls through some more data im yet to work out what all the outs mean and what sub evens equal too but i am happy so far its semi working .

```
after wemos reboot
paradoxdCTL/status {"status":"Problem connecting to panel"}
paradoxdCTL/status {"status":"Paradox Disconnected"}
paradoxdCTL/status {"status":"Paradox connected"}
paradoxdCTL/out { armstatus:224, event:44, sub_event:19, dummy:}
paradoxdCTL/out { armstatus:224, event:44, sub_event:18, dummy:}
```

maragelis, 2019-01-06:
check out events.txt file

mydakota, 2019-01-11:
where do i find the events.text file thanks.
in serial mon i get returned command = 4 and 5 with arming or disarming via mqtt command/JSON as per wiki using paradoxdCTL/in ...
when i read through the Arduino file code threes no reference to command 4 or 5 

```
}

byte getPanelCommand(String data){
  byte retval=0x00;
  data.toLowerCase();
  if (data == "stay" || data=="0")
  {
    
    retval = Stay_Arm;
    
  }
  else if (data == "arm" || data=="1")
  {    
    retval= Full_Arm;
  }
  else if (data == "sleep" || data=="2")
  {
    
    retval= Sleep_Arm;
    
  }
  else if (data == "disarm" || data == "3")
  {
    
    retval=Disarm;
    
  }

  else if (data == "bypass" || data == "10")
  {
    
    retval=Bypass;
    
  }

  else if (data == "panelstatus" )
  {
    retval=0x90;
    trc("PAnelStatus command ");
    
  }

  else if (data == "setdate")
  {
    retval=0x89;
    
  }
  else if (data == "armstate")
  {
    retval=0x91;
    
  }
   else if (data == "zonestate")
  {
    retval=0x92;
    
  }
  else if (data == "disconnect" || data == "99")
  {
    retval=0x00;
    PanelDisconnect();
  }
  Serial.print("returned command = ");
  Serial.println(retval , HEX);
  return retval;
}

```

maragelis, 2019-01-11:
#define Stay_Arm  0x01
#define Stay_Arm2 0x02
#define Sleep_Arm 0x03
#define Full_Arm 0x04
#define Disarm  0x05
#define Bypass 0x10

maragelis, 2019-01-11:
https://github.com/maragelis/ParadoxRs232toMqtt/blob/master/Events.txt

rjduraocosta, 2019-01-13:
Helllo @maragelis good afternoon,

To start, very nice job in this new version. Once again thank you for your good work.

Sorry to bump in this issue with another question but I am using a Wemos d1 mini pro, like the one you are using, and I am trying to compile and download the project with the Arduino IDE v1.8.8 and after uploading the board simply does not repond to anything. Could you please tell me how are yoour configurations for the Arduino IDE and wich esp8266 core are you using? I tried to blank the board first an then upload the sketch with the Ide but it stays the same.

I had to blank the board and flash your bin file, and the it worked. But I really wanted to know what I am doing wrong in the IDE...

Thanks again for your support and wonderfull project.

Best regards.

maragelis, 2019-01-13:
I will try get instructions up ASAP. I use arduino core 2.4.2 updated libs. Just changed the pusubclient file max message size to 512.

rjduraocosta, 2019-01-13:
> I will try get instructions up ASAP. I use arduino core 2.4.2 updated libs. Just changed the pusubclient file max message size to 512.

Perfect! I will wait on your instructions.

Thanks again for your prompt reply and always precious help.

Best regards.

maragelis, 2019-01-14:
https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/20190114a

rjduraocosta, 2019-01-15:
> https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/20190114a

Thanks again @maragelis for pointing out the new version. Since I have updated to v2 it happens alot more frequently that I get the events for the zone having precense but not the status that it has no more precense, beeing that at V2 ON but not OFF. I am monitoring the brooker with a client but no messages. after this happens I cannot interact with the system anymore. I have to reset the board. Also I see that sometimes when tinteracting with the system I do not get al the events from event, subevent.
I flashed always a blank 4MB bin file before flashing your bin file.
Could this be due to something I am not seeing.

Thanks again.

maragelis, 2019-01-25:
https://github.com/maragelis/ParadoxRs232toMqtt/releases/tag/20190125

maragelis, 2019-02-05:
@rjduraocosta try new realese fixed a serious bug.

rjduraocosta, 2019-02-05:
> @rjduraocosta try new realese fixed a serious bug.

Thanks @maragelis for the new release. I will give it a go and get back to you on the results.

maragelis, 2019-02-11:
Same as #27

closed 2019-02-11 by maragelis