maragelis/ParadoxRs232toMqtt #27: Is there any Timeout settings?

ciscozine, 2019-02-04:
Hi, 
I have a Magellan 5000 and I use a WemosD1 mini to connect to the alarm system.

At night I arm the alarm, but in the morning "ParadoxRs232toMqtt" seems to be down.
I can ping the wemos but I cannot receive MQTT Message.

What can I do? How can I do to debug it?

I use hassio on a rasperry and "Mosquitto broker" as a MQTT Server.

thanks
Fabio

maragelis, 2019-02-04:
have you tried the new release ? There where many bug fixes done.

ciscozine, 2019-02-04:
Yes, I have tried the last release without resolve the problem

Thanks

maragelis, 2019-02-04:
Is there any chance the raspberry is going to sleep?
Are you also getting "paradox disconnected" on the status topic ?

ciscozine, 2019-02-04:
No It doesn't go to sleep.
Do you know where can I find MQTT broker log on hassio?

thanks

maragelis, 2019-02-04:
No sorry. Use mqtt.fx and leave it connected to status topic

ciscozine, 2019-02-04:
Ok. I'll update you. Thanks

maragelis, 2019-02-04:
No I dont use home assistant. I use openhab and homekit

maragelis, 2019-02-04:
If you want you can try this https://github.com/maragelis/ParadoxRs232toMqtt/tree/Dev/ParadoxAlarmSystem/ParadoxAlarmSystemOTA

It has some mqtt connection updates, and get back to me Thanks

ciscozine, 2019-02-04:
What are the differences between the 2ino files?
thanks

maragelis, 2019-02-04:
Yes it will reconnect

ciscozine, 2019-02-04:
Hi, I'll load the new sketch. I'll update you tomorrow.
A question. I have trace =1 but using console (9600 or 115200) I do not see nothing. Why?

thanks

ciscozine, 2019-02-05:
Again, the same problem. The interface to the paradox seems be down.

I have sniffed the IP flow.

The wemos is live (icmp ok), the mqtt flow still persist. I notice only a SSDP port unreachable  ssdp:

each five minute the wemos send to the broker a ssdp message, but at 3:06am the wemos send 2 ssdp message and 1 of that it's a port unreachable (problem on hassio server?)
[image]
But I don't know if it is the problem.

2 questions:
1. My circuit has no resistor. is this the cause the problem?
2. Can you tell me for each lib, which version have you used? I updated some lib and now, if i try to compile the sketch, I see "DynamicJsonBuffer' was not declared in this scope"

Thanks again

ciscozine, 2019-02-05:
I'll test and I'll update you! Thansk.
My personal suggestion for .bin file. Why don't you give use the possibility to enable authentication and SSL encryption? It would be perfect :)

Thanks

maragelis, 2019-02-05:
I will look into it

ciscozine, 2019-02-05:
I update you.
also with the .bin the problem still persist.
What I notice is that after the problem the log of my mqtt broker on hassio (raspberry) is:
1549393009: Client 80:7D:3A:3E:B5:FF has exceeded timeout, disconnecting.
1549393009: Socket error on client 80:7D:3A:3E:B5:FF, disconnecting.
but I can ping the wemos d1..

Moreover if I unplug the power and plug the power, the wemos d1 led blinks only 1 time and the ping stop to work.
If I detach the serial port to the alarm system and unplug/plug the power, the wemos d1 led blink correctly (3 or 4 times); then if I attach the serial port the broker receive the message correctly.

Thanks

ciscozine, 2019-02-05:
One time I receive also a "{"status":"Problem connecting to panel"}".
I change the cable without any changes. This is why I ask you if it is better use some resistor .

thanks

maragelis, 2019-02-05:
So this happens only when armed ? please help so I can help.

ciscozine, 2019-02-05:
No I noticed that only when disarmed. I cannot test if armed because there is a PIR sensor that I cannot disable

maragelis, 2019-02-05:
So you are disarmed and the wemos disconnects from mqtt ?

ciscozine, 2019-02-05:
I'm sorry. I understand now. Generally when I try do disarm the system (it is armed), the wemos stops to work.

maragelis, 2019-02-05:
Are you maybe having low wifi signal

maragelis, 2019-02-05:
If the wifi signal cut out for too long the times out and cannot reconnect. Please check if you are getting disconnect status message.

maragelis, 2019-02-05:
{"status":"Paradox Disconnected"} this is the last will message if wifi is disconnected, can you check if you are getting this message.

ciscozine, 2019-02-05:
Well, I have an access point at 20cm .. It isn't a wireless problem; in fact i can ping the wemos without any problem.
No I don't see "{"status":"Paradox Disconnected"}" .

ciscozine, 2019-02-05:
Paradox Magellan 5000. I don't know the firmware. Can I find it with your system?

maragelis, 2019-02-05:
IF that is the one with the 433mhz transmitter make sure your cables and wemos are well grounded and shielded. I will look into it as I only have the SP series api ans get back to you. @[rjduraocosta](/rjduraocosta) are you using the same panel ?

maragelis, 2019-02-05:
Could you if you have the means to read the voltage over your tx and rx lines. if they are above 3.3v I suggest a level converter between wemos and panel .

maragelis, 2019-02-05:
Are you powering the wemos from the panel ?

ciscozine, 2019-02-05:
I used power from the panel and from a external power supply. Each one have the problem.
tomorrow I check the cable and the welding.

thanks again

rjduraocosta, 2019-02-06:
Hello @maragelis and @ciscozine ,,

I measured the voltage on my Wemos Pin's and they are the following:

- D8(TX) - 3,26 V
- D7(RX) - 4,85 V
The above makes sence because the Wemos transmits at 3.3V logic at the D8 pin and receives from the panel at D7 pin in 5V logic.

Some ideas:
Move the RX/TX to the original ones? 
Level Shifter? 
Resistors? 
RF Interference?

@ciscozine any news about your setup?

Thanks for the help.

maragelis, 2019-02-06:
I am betting on interference. I will try tackle it in code but it could yield unwanted results.

ciscozine, 2019-02-06:
Hi @maragelis , good news.

I have tested all components with the tester and it seems ok, but I have identified the cause of my problem (I hope). A solder. 

Thanks for all.

For the last but not the least, my suggestion is to make a bin file with the possibility to enable authentication and encryption (ssl).

rjduraocosta, 2019-02-06:
Hello again,

I could try and move the Wemos further away for testing, but I cloud only do this at the weekend. It is all in a box that I have to move further and also do another longer cable.

Thanks for the suggestion.

rjduraocosta, 2019-02-06:
Hello @maragelis good evening,

I have been testing the system and I found one thing, that it is the following:
- I have connected MQTT.fx monitoring the paradoxdCTL/# topic. After some time I see that I stop receiving the zone events while I walk around the house. After this I send a command (stay/disarm/etc.) to the system that the system takes well and I start receiving back the zone events. I have been repeting this for 2 hours and the system does not stay completly down if I keep "waiking" it up from now and then.

If you have some other idea other than mving the wemos away until this weekend, or something I could test I appreciate it.

I do not have a level converter I just have resistors right now. I am not sure if using resistors at the Wemos RX line is fast enough for the serial communication with this configuration: 
[image]

Thanks again for your work.

maragelis, 2019-02-06:
ok so I am starting to understand what is going on. Can you please tell me the last message you receive before it goes down. Is it a mqtt disconnect ? 

The only way we will see whats going on is for you to use a ftdi adapter connect to d4 / RX and get Trace messages

rjduraocosta, 2019-02-06:
I have never seen an mqtt disconnect message before the system goes down. In all the tests the last message in the mqtt broker was a zone event, sometimes ON event sometimes OFF event.

Tomorrow I will use an ftdi adapter and connect it to D4 to see the messages. Before that I have to enable TRACE=1through mqtt, right? Another thing, can I use the Arduino IDE Serial monitor for that or do you recommend another program?

Thanks for the support.

maragelis, 2019-02-06:
Trace=1 yes,  not TRACE=1 yes you can use Arduino IDE just choose right port

maragelis, 2019-02-06:
If you have patience we will fix it I just need your help.

rjduraocosta, 2019-02-06:
Perfect! You can count on me.

Tomorrow I will share the Trace output's with you.

Thanks again for all for the support.

rjduraocosta, 2019-02-07:
Hello @maragelis,

I have the data for you. I restarted the system this morning just to start from scratch and then enabled the Trace and monitored the MQTT messages. For about 2 hours it was all stable with the reporting the zone events to the MQTT broker. After that period I stopped receiving the zone events and hads to send a "wake up" to the system by a command through MQTT, just like I described yesterday.
Attached you will find the Trace file. If you open it with and editor that you can see line numbers look at line 1138 that is when I "woke" the system up.

Thanks again for all your effort and help.
[MG5000_Trace_20190207.txt](https://github.com/maragelis/ParadoxRs232toMqtt/files/2840509/MG5000_Trace_20190207.txt)

maragelis, 2019-02-07:
Ok thanks @rjduraocosta , I saw the problem. The serial buffer is getting corrupt. I do have code which tries to fix this, but looks like its getting stuck somewhere. I will try send you an update to see if this solves the problem. It should be ready by tonight be patient as I have day job too.

rjduraocosta, 2019-02-07:
@maragelis, thank you for your work. Please take your time and when you are ready I will test it for you, no problem. I also have a day job and I know how thinks are no problem at all. I just had todays day off, remaining vacation day from last year, that is why I could test this today.

Thanks again and please take your time.

maragelis, 2019-02-07:
I have a new cleaned up version 2.2 for testing and I thing I have squashed this bug. I also had this long time ago. It happens when to-many events overflow the serial buffer.  In the login i clear the buffer thats why it starts working again. Looks like its taking to long to process the incoming messages.

maragelis, 2019-02-07:
Can you please try this version

https://github.com/maragelis/ParadoxRs232toMqtt/tree/Dev/ParadoxAlarmSystem/ParadoxAlarmSystemOTA

rjduraocosta, 2019-02-07:
Yes of course, thank you.

Will give it a test and keep you posted on the results.

Thanks again.

ciscozine, 2019-02-08:
Hi @maragelis, today after 3 days I have the same problem.
My cable is about 40cm. Could it be the cause of the disconnection?

thanks
Fabio

maragelis, 2019-02-08:
No its a buffer issue

ciscozine, 2019-02-08:
Ok, so it is possible fix via software?
thanks again 👍

rjduraocosta, 2019-02-08:
Hello @maragelis,

I have been testing the last version last night in the evening. I noticed that the system stopped sending the zone events and I tried to send a command to "wake" the system up but this time I had the following  in the MQTT brooker:

[image]

Any ideas that I can test?

Thanks again

maragelis, 2019-02-08:
give me 10 min

maragelis, 2019-02-08:
I have just updated the dev project try that one https://github.com/maragelis/ParadoxRs232toMqtt/tree/Dev/ParadoxAlarmSystem/ParadoxAlarmSystemOTA

ciscozine, 2019-02-08:
@maragelis Does the new release fix the buffer issue?

Thanks

maragelis, 2019-02-08:
I myself HOPE SO

ciscozine, 2019-02-08:
I'll load the new bin and I'll let you know  :)

rjduraocosta, 2019-02-08:
> I have just updated the dev project try that one https://github.com/maragelis/ParadoxRs232toMqtt/tree/Dev/ParadoxAlarmSystem/ParadoxAlarmSystemOTA

Perfect will do so. I can only do it at the end of this afternoon. I will keep you posted on the results.

Thanks again.

maragelis, 2019-02-08:
Great that gives me time to do some house keeping

ciscozine, 2019-02-08:
It doesn't work.
I see:
paradoxdCTL/status - {"status":"Paradox connected"}
paradoxdCTL/status - SYSTEM READY PARADOX_2.2.0 
paradoxdCTL/status/Arm - 

The client is reachable, but it doesn't send any mqtt message to the broker.

maragelis, 2019-02-08:
You will have to wait untill I get home. So I can check.

maragelis, 2019-02-08:
I dont see a error on my side everything works fine

ciscozine, 2019-02-08:
Do you see mqtt message to the broker?

[117 more replies]

PR #33 (2019-02-12): Dev
Fixed #27 
Disabled setdate, will look into it later.
closed 2019-02-12 by maragelis