maragelis/ParadoxRs232toMqtt #29: 'StaticJsonBuffer' was not declared in this scope

ciscozine, 2019-02-08:
Hi, if I try to compile the last version of .ino file I receive this message from the compiler

'StaticJsonBuffer' was not declared in this scope

I need to compile the ino files because I would use authentication and encryption (SSL).

All lib installed are the last stable version.
thanks

maragelis, 2019-02-08:
As you can see here https://arduinojson.org/v5/doc/encoding/ documentation for arduinojson lib version 5.13.4 states clearly on page 13/33 states  StaticJsonBuffer<capacity> jb;

So looks like you are using a version other from this .

ciscozine, 2019-02-08:
Nothing to do
Now I have this error:

`'getSubEvent' was not declared in this scope`

I have installed a clean arduino 1.8.8 with:
esp8266 2.5.0

LIB:
pubsubclient 2.7.0
wifimanager 0.14
arduino json 5.13.4
NTPtimeESP-master I find only that one https://github.com/SensorsIot/NTPtimeESP/ (is it correct?)

thanks

maragelis, 2019-02-08:
yes my friend, NTPtimeESP-master is correct. 

error 'getSubEvent' was not declared in this scope
you have to go to sketch->add file and add the other ino files in the project

ciscozine, 2019-02-08:
Nothing to do:

```
C:\Users\fabio\Documents\Arduino\paradox\ParadoxAlarmSystemOTA\ParadoxAlarmSystemOTA.ino: In function 'void setup_wifi()':

ParadoxAlarmSystemOTA:1004:30: error: call of overloaded 'String(IPAddress)' is ambiguous

     trc((String)WiFi.localIP());
```

maragelis, 2019-02-08:
please use Arduino core 2.4.2

closed 2019-02-08 by maragelis