maragelis/ParadoxRs232toMqtt #38: Voltage regulator schema for MG5050

colcm, 2019-02-15:
Hello,

I have one ESP8266 and one Voltage regulator (3V and 5V from 12V).
Can you help me with a schema how to connect all these components to a Paradox MG5050 panel, please?

Thank you!

maragelis, 2019-02-15:
the YwRobot 545043 is a bread board regulator it cannot be used easily. Get yourself a good buck converter something like this. 
https://www.ebay.com/itm/Adjustable-DC-DC-Buck-Step-down-Converter-5V-24V-To-3-3V-9V-12V-3A-Power-Supply-/122078008388

colcm, 2019-02-15:
Hello,

and how can I use the 12V and GND from MG5050 in this combination (ESP8266 and convertor)? Can I use the GND from MG5050 to be common for ESP8266 and convertor?

Thank you!

maragelis, 2019-02-15:
[image]

closed 2019-03-09 by maragelis