maragelis/ParadoxRs232toMqtt #43: Version 2.2.3 Timeout

pilievwm, 2019-02-24:
Hi @maragelis I just updated the new version and it is getting timeout again.
I think the problem is with the library NTPtimeESP.h
Please tell me how to fix it or remove it?

maragelis, 2019-02-24:
Where are you getting timeout

pilievwm, 2019-02-24:
The wemos device is unreachable. Just like the dev version before 2 weeks :(
It is working good for a while and than it gets unreachable

maragelis, 2019-02-24:
updated master

pilievwm, 2019-02-24:
It is a "Stalin fix" ;) 
No NTP, no problem :)
Is there a special way to update my Wemos via OTA? When i select via my Arduino it reads error:
Device is not found

maragelis, 2019-02-25:
You have to send a OTA=1 message that enables OTA

pilievwm, 2019-02-25:
Ok but where should I send it?

pilievwm, 2019-02-25:
Unfortunately removing NTP doesn't fix the problem. Something is causing the device to get disconnected. 
Something new which is missing into the old version which is still working stable.

tasosstr, 2019-02-25:
Use MQTT .fx and Publish in "paradoxdCTL/in" the command "OTAUpdate=1" (with new version 2.3.0

maragelis, 2019-02-26:
I will need a trace file to see where the problem is, if the previous version is working for you keep that one. Unfortunately I have new projects (paid job) I have to work on and I have no time to maintain this one.

ciscozine, 2019-02-26:
Hi @maragelis .
I tested the last version on master branch.
Initially the wemos is connected to the mqtt server ({"status":"Paradox Connected"}), but nothing is received on the "paradoxdCTL/#" and the commands, for instance to arm the alarm, does not work.

After some minutes I receive the

{"status":"Paradox Disconnected"}

but the wemos is still reachable via ICMP.

When I rollback to the previous release ( I use a dev release of 3 weeks more or less - #define firmware "PARADOX_2.2.1"), all works fine.

Thanks
Fabio

pilievwm, 2019-02-26:
I managed to update the software but it is keep disconnecting 
What could be the reason?

pilievwm, 2019-02-26:
@ciscozine what is your version of the arduino or  other software?
There should be differences because when i ping my device it is working good for 30-40 packages and after that, it is dropping the connection.

ciscozine, 2019-02-26:
I use arduino 1.8.8 with:
esp8266 2.5.0

LIB:
pubsubclient 2.7.0
wifimanager 0.14
arduino json 5.13.4

rjduraocosta, 2019-02-28:
Hello @ciscozine and @pilievwm,

I have been testing the last master version since it's last release, so 4 days now, and I have not had any problems with it. Did you solve your issues yet? What wemos boards are you using?

Best regards.

pilievwm, 2019-03-04:
@maragelis i keep trying... without result. 
Arduino is 1.8.5
esp8266 2.5.0

LIB:
pubsubclient 2.7.0
wifimanager 0.14
arduino json 5.13.4

PubSub 256

But after compile it is working for 1 minute and then it get disconnected from the network. 
Only the old master is working. Can you please get version 2.1.0 and tell me how to add retain message and mqtt auth? 
Please...

maragelis, 2019-03-04:
I will try do this give me time.

maragelis, 2019-03-04:
Just a quick question, have you configured the serial correctly? what I am trying to say is if you are connected to d13 d15 enable the serial swap to 1 by default this is 0 for wemos rx/tx

Egyras, 2019-03-06:
> Just a quick question, have you configured the serial correctly? what I am trying to say is if you are connected to d13 d15 enable the serial swap to 1 by default this is 0 for wemos rx/tx. I set this to 0 in the last build so new users don’t get confused.

Hi, could you provide used library links in github. Write in readme.

maragelis, 2019-03-06:
It is in the readme 
wifimanager
pubsubclient
arduinojson
NTPtimeESP

Egyras, 2019-03-06:
> It is in the readme
> wifimanager
> pubsubclient
> arduinojson
> NTPtimeESP

Just would be better if you did like this
arduinojson (ver.? ) -  https://github.com/bblanchon/ArduinoJson  
.....
.....
Its better doing testing and much easier to get all required libraries with proven versions. Or upload your libraries here in github ;)

maragelis, 2019-03-06:
When I get the time I will try do something better.

pilievwm, 2019-03-07:
@maragelis thanks for that! I really appreciate your work!!!
Hope that you will find time to make it

pilievwm, 2019-03-07:
> Just a quick question, have you configured the serial correctly? what I am trying to say is if you are connected to d13 d15 enable the serial swap to 1 by default this is 0 for wemos rx/tx. I set this to 0 in the last build so new users don’t get confused.

Forgot to answer this question. I did not connect the wemos to the paradox board. It is freshly flushed with the new code and it connects to the wifi and then to the MQTT broker. After a while it gets disconnected

Egyras, 2019-03-07:
For Paradox SP7000 I assume that serial connector is the same as for Babyware or Infield firmware upgrade?

maragelis, 2019-03-07:
@Egyras yes that’s it. 
@Pilievwm I need a trace my friend, or tell me what you did. Did you compile or did you use the bin file? I have many chips running this version with no problems with disconnecting.

Egyras, 2019-03-08:
@maragelis which exact wemos you are using? Original or "copy"?  I have tried with copy and seems that cant get any messages from my sp7000 paradox to wemos.... using original RX and TX...

maragelis, 2019-03-09:
Try swapping rx tx around. It doesn’t matter which wemos you use. Check your cables.

pilievwm, 2019-03-09:
@maragelis when I start to ping 192.168.1.123 -t and in 30-40 packages it gets disconnected. 
The wemos is compiled with the latest version. Same device if i compile version 2.1.1 is working perfectly.

maragelis, 2019-03-09:
@pilievwm I just tested your claim and it does not do that. I suggest you erase the wemos.
Or there is something wrong with your network.

[image]

rjduraocosta, 2019-03-09:
@pilievwm use a blank 4Mb file and flash the Wemos with it. I always do this before flashing a new version and up until now it has not failed me.

Download the esp easy tool from here: 

https://github.com/letscontrolit/ESPEasy/releases

Inside the bin folder there a file named blank_4MB.bin.

Copy that file to the same place where you run the software. Like this:

[image]

Run the FlashESP8266.exe file and chose your COM port and blank file from the dropdow menus and wait until finished:

[image]

Afterwards flash tour wemos with Arduino IDE. First choose another board in boards manager, the first board will do it is Arduino Yun, to force an error. After the error choose the correct board and be sure to select Erase Flash: All Flash Contents:

[image]

I had a few problems when exganging between version that if i did not flas a blank file, despite the Erase all flash content on Arduino IDE, the Wi-Fi settings where not erased on the board. Using the above procedure the wemos gets everytime well initialised.

Just as a comment, I am using v.2.2.3, last version and have not had any issues with it.

I hope this helps you.

Best regards.

maragelis, 2019-03-09:
@rjduraocosta Thanks for the help

rjduraocosta, 2019-03-09:
@maragelis no problem at all. I hope it helps.

Thank you for your work.

pilievwm, 2019-03-10:
Guys nothing works!!! :(
All i have to do is to try the bin file which is readily compiled. The question now is how to setup the mqtt server?

maragelis, 2019-03-10:
connect to its wifi, and open browser (192.168.4.1)

pilievwm, 2019-03-10:
Alright! :) Now it is trying to connect to the mqtt server but:

`1552248337: New connection from 192.168.1.137 on port 1883.
1552248337: Socket error on client <unknown>, disconnecting.`

pilievwm, 2019-03-10:
Can somebody compile one bin with mqtt server address 192.168.1.10 and username "paradox" and password "securitysys"

maragelis, 2019-03-10:
[ParadoxAlarmSystemOTA.ino.d1_mini.bin.zip](https://github.com/maragelis/ParadoxRs232toMqtt/files/2949988/ParadoxAlarmSystemOTA.ino.d1_mini.bin.zip)

pilievwm, 2019-03-10:
Thanks a lot Margelis! Unfortunately, it is the same. Something into that version makes the device unstable.
If i deploy ver. 2.1.1 it is working without any issues...

rjduraocosta, 2019-03-10:
@pilievwm let me see if I can help. What Wemos board are you using d1 mini? d1 mini pro?
I will compile a bin file for you so that you can try mine, just for the test if you want.

pilievwm, 2019-03-10:
Thanks @rjduraocosta @maragelis ! I really appreciate your help!
Wemos D1 mini.
Serial Swap 0
Hassio 1
Homekit 0

rjduraocosta, 2019-03-10:
@pilievwm I compiled 4 version with different SPIFFS sizes for you to test, just in case.

I hope it helps.

[ParadoxAlarmSystemOTA.ino.d1_mini.zip](https://github.com/maragelis/ParadoxRs232toMqtt/files/2950056/ParadoxAlarmSystemOTA.ino.d1_mini.zip)

pilievwm, 2019-03-11:
Thanks @rjduraocosta but the problem is the same.
I have two devices - one is connected to my Paradox. The second is that i test the new version.
Is there a clue that they somehow get mess each other?

rjduraocosta, 2019-03-11:
Hello @pilievwm,

Have you always tested versions this way, with two Wemos boards?

pilievwm, 2019-03-13:
Hey @rjduraocosta,

I got no problems with v2.1.1. It is working perfectly fine. The problem with that version is that there are no some important features for me
1. No MQTT authentications 
2. No, retain flag which makes my alarm status Unknown once the Hassio is restarted. 

If I manage to implement those two features into v2.1.1 I will leave with that version forever :)

p.s. Every day when I get my home after work, I check here with the hope that there is a solution for my stupid case. I tried with 4 different wemos devices. None of them are working stable with the latest release, even if I use the precompiled .bin file which you and @maragelis provided.

I am totally stuck. I got 18 Wemos devices into my home working for almost 2 years. Most of them are Tasmota enabled and I've never experienced such thing :(

maragelis, 2019-03-13:
Is the v2.1.1 pre compiled or do you compile from source ?  If you compile add the features into the file. But I believe the authentication is giving you problems. I will check it today and get back.

maragelis, 2019-03-13:
If you can get me a trace file it would help me determine the problem.

maragelis, 2019-03-13:
Another thing you can try is playing around with older versions of pubsubclient and arduino core libraries.

maragelis, 2019-03-13:
Arduino core 2.4.1
WifiManager by tzapu 0.12.0
PubSubClient by Nick O`Leary 2.6.0

ciscozine, 2019-03-13:
hi @maragelis 
what about the MQTT_MAX_PACKET_SIZE? Still 256?
Thanks

rjduraocosta, 2019-03-13:
Hello @maragelis,

Just to give my contribute. I am using these configurations with the latest master and no issues about disconnections:

Arduino core 2.5.0
pubsubclient by Nick O`Leary 2.7.0
wifimanager by tzapu 0.14
arduino json 5.13.4
MQTT_MAX_PACKET_SIZE 256

I am using a Wemos D1 mini pro. I do not know if this has an influence on this subject.

Thanks again.

maragelis, 2019-03-13:
I cannot explain but when using 2.4.2 and 2.5.0 it disconnects  when using user name password on mqtt, I will try your settings @rjduraocosta

maragelis, 2019-03-13:
@ciscozine  yes 256 is ok

maragelis, 2019-03-13:
I will be doing some homeassistant enhancements tonight, then we can go to 128

rjduraocosta, 2019-03-13:
> I cannot explain but when using 2.4.2 and 2.5.0 it disconnects when using user name password on mqtt, I will try your settings @rjduraocosta

@maragelis I do not have suthetication activated at my MQTT Brooker that is why I am using in the Wemos void Username and password:
[image]
Maybe that's the issue that mine works.

maragelis, 2019-03-13:
@pilievwm 2.2.4 with username password in wifimanager config. 
SOS. Dont forget to wipe wemos
https://github.com/maragelis/ParadoxRs232toMqtt/tree/Dev

Egyras, 2019-03-17:
> @pilievwm 2.2.4 with username password in wifimanager config.
> SOS. Dont forget to wipe wemos
> https://github.com/maragelis/ParadoxRs232toMqtt/tree/Dev

I have tried, when sending ARM via MQQT.fx it disconnects:

{
  "password" : "xxxx",
  "Command" : "arm",
  "Subcommand" : "0"
}

{
  "status" : "Paradox Disconnected"
}

Iam using SP7000 with latest firmware and have PSC250 connected as well.

Egyras, 2019-03-17:
swapped RX and TX wires, get /OUT messages now :)

Egyras, 2019-03-22:
If anybody has issues:
1. check TX and RX connection to Paradox panel - related to ESP8266 reboot
2. Never use master password for arm/disarm as this doesn't work, create separate password with required permissions (Babyware can be used)

I solved all my problems, 2.2.3 works/