maragelis/ParadoxRs232toMqtt #46: Home Assistant

thanasisloi7, 2019-03-13:
Hello, 
i am trying to get in the Home assistant the values of MQTT topic paradoxdCTL/out when i send the topic paradoxdCTL/in for panel status. I am trying this

maragelis, 2019-03-13:
after reading this https://www.home-assistant.io/components/sensor.mqtt/
I dont see how it panel status can work without dowing some recoding. 
Every status needs its own mqtt/topic can someone  confirm this ?

maragelis, 2019-03-13:
Because HA has a big install base I will try implement this. I will need some one to test it as I myself dont use HA.

maragelis, 2019-03-13:
Created v2.2.4 in dev branch (FOR TESTING) below Hassio topics with there values for panelstatus 0
https://github.com/maragelis/ParadoxRs232toMqtt/tree/Dev

paradoxdCTL/hassio/Timer_Loss
paradoxdCTL/hassio/PowerTrouble
paradoxdCTL/hassio/ACFailureTrouble
paradoxdCTL/hassio/TelephoneLineTrouble
paradoxdCTL/hassio/PSUDCVoltage
paradoxdCTL/hassio/BatteryDCVoltage
paradoxdCTL/hassio/BatteryTrouble

use in HASSIO like this : 

platform: mqtt
name: 'Paradox Voltage'
state_topic: 'paradoxdCTL/hassio/BatteryDCVoltage'
unit_of_measurement: 'V'
qos: 0

georgelza, 2019-06-05:
... now if we could just get my D1 working and I'll happily give you remote access. problem, one pesky IP150 in the way, so after doing some digging, that IO1 C IO2 ports are actually to control devices, but right below them there is what looks like 2 small 4 pin ports, that seems to be where the cable plugs in that goes to the serial port of the sp6000 (top right 4 pin connector) Panel, there is a 2nd port marked "serial out" on some units, ... now to try and figure out how this is wired and if can work as a entry point.
G