maragelis/ParadoxRs232toMqtt #47: Pending Status in HA

thanasisloi7, 2019-03-13:
Hello,
how can i get the pending status for this component https://www.home-assistant.io/lovelace/alarm-panel/
Now i am using the https://github.com/alfredopironti/Paradox_IP150 so when i arm the panel in standby time until the alarm is armed i get the status pending.
Here is a picture https://www.home-assistant.io/images/lovelace/lovelace_alarm_panel_title_card.gif

maragelis, 2019-03-13:
https://www.home-assistant.io/lovelace/alarm-panel/
The above is a combination with this 
https://www.home-assistant.io/components/alarm_control_panel.manual_mqtt/

what I understand without knowing much about HA:

alarm_control_panel:
 platform: manual_mqtt
 state_topic: "paradoxdCTL/hassio/Arm"
 command_topic:"paradoxCTL/in"
 code:1234
 pending_time: 30
 delay_time: 20
 trigger_time: 4
 payload_disarm: '{"password":"0000","Command":"disarm","Subcommand":"0"}'
 payload_arm_night: '{"password":"0000","Command":"sleep","Subcommand":"0"}'
 payload_arm_home: '{"password":"0000","Command":"stay","Subcommand":"0"}'
 payload_arm_away: '{"password":"0000","Command":"arm","Subcommand":"0"}'

after adding this the component https://www.home-assistant.io/lovelace/alarm-panel/ should work

SiderisT, 2019-03-13:
Hallo to everyone, after the change to manual_mqtt pending status is working normally, but after restarting the system (ΗΑ),  the alarm **always** disarmed, **maybe it's bug.**

[(https://community.home-assistant.io/t/alarm-is-disarmed-after-restart/10879)](url)