maragelis/ParadoxRs232toMqtt #48: Arming 2 zones with 1 json playload

Egyras, 2019-03-22:
Hi, any options to make 1 json playload to arm/disarm both zones at once ?

maragelis, 2019-03-23:
partions ? no the paradox does not allow it. you can use node-red to do this

Egyras, 2019-03-23:
> partitions ? no the paradox does not allow it. you can use node-red to do this

with PSC250 you can do this via SMS, so I thing its possible

maragelis, 2019-03-23:
The psc250 is running two commands 
The serial api only allows 1 partition 
Action = 0x05 	Partition number (0 **or** 1).

maragelis, 2019-03-23:
You could create a new command in my program and send both commands

Angel0ffDeath, 2020-02-24:
Hi Panagiotis,
First of all "perfect job". I'm with MG5050 and everything works perfect!!!
I have also 2 partitions and as you wrote above it is not possible to arm them with 1 json
Just a simple test - I armed the system via keyboard and my PC subscribed for the topics says that a lot of things happens: Arm partition0; Bell squeck; Arm partition 1 and another Bell commands...
So to answer the question in this topic: It is  not necessary to recompile your code and create new commands. When using home automation system just be sure you issue the correct commands. I'm using FHEM (perl server but in general it doesn't matter). My script looks like Arm(partition0); sleep 2; Arm(partition1). The same is valid for disarm commands. 
So in general:
Arm partition 0
Wait 2 sec
Arm partition 1

works here :)

@Egyras : Can give you some support if the problem still exists. Contact me at angeldeath743atgmail.com

@maragelis : I do believe with this post you can close the issue

Egyras, 2020-02-24:
@Angel0ffDeath I just changed arm part in the code :)

closed 2020-02-24 by Egyras

Angel0ffDeath, 2020-02-24:
Also a solution. Good luck :)