maragelis/ParadoxRs232toMqtt #49: Arm, disarm payload with wrong password

chrisgsauto, 2019-03-29:
Hi Panagiotis,
i realized by mistake that sending in the topic "paradoxdCTL/in" payloads with wrong password, the paradox alarm system is arming and disarming. As example i'm having as acount password 1234 and i'm sending from my HA  payload_disarm: '{"password":"aaaaaaa","Command":"disarm","Subcommand":"0"}' and my system goes to disarmed. These happens when i send a command one time with the suitable code (1234), after that any commands with any wrong code is controlling my paradox system.

chrisgsauto, 2019-03-29:
I am running 2.2.3

maragelis, 2019-03-30:
That is correct after 30 seconds the program will logoff from the panel and it will be allow other requests