maragelis/ParadoxRs232toMqtt #50: Hardware question

farkinca1971, 2019-04-09:
Hello

Sorry to open an issue for this hw issue of mine, but I would like to ask you asap about this one.
I connected a wemos board to paradox sp6000. The wemos is supplied via usb , thus only 3 cable is used for the serial connection (gnd, tx and rx). I uploaded your scripts to wemos and apparently is working because I received messages in paradoxdCTL/status, but not in paradoxdCTL/in. Over there is nothing :(
My questions are the following.
The 3 cable are enough or i have to use 4, aux+as well ? Do I have to do something on the SP6000, eg to let the pc communcitaion start ? I am stuck right now, I would appreciate any help ! :)  

Thanks a lot and sorry for bothering you with this !

cybertza, 2019-04-09:
Hi, the in MQTT topic is for commands sent to the device, if you read the read me you will get a full message that you send to the panel, just replace 0000 with a valid code to arm and disarm and send the command to the panel to do the function arm,disarm,stay and so forth.

cybertza, 2019-04-09:
json Command payload template 
{ "password":"1234", "Command":"arm", "Subcommand":"0" }

password is user 4 digit password

Command can be one of the following

arm,
disarm,
sleep,
stay,
bypass,
armstate,
panelstatus 
setdate 
PGM_ON
PGM_OFF

farkinca1971, 2019-04-09:
Hi, Thanks
My question:  If i start up the wemos, I should not see any live events without any commands?
Do I have to send command like panelstatus ?

Thanks again

cybertza, 2019-04-09:
they should automatically land on the mqqt topic paradoxdCTL/out on your mqqt server when every zone opens or closes, also i believe that the new firmware has a topic for "paradoxdCTL/hassio/Arm/zoneX" gives values ON and OFF

cybertza, 2019-04-09:
if there is nothing on the topic when you open close zones, just confirm all your wiring with the picture in the readme.

farkinca1971, 2019-04-09:
Yep
That was my original question :)
Do I have to use 4 cable without the usb, or can I use 3 without Aux+ with usb power supply ?
Maybe the sp6000 can detect how many cable are connected on serial port and blocks the traffic when there are only 3 connected !?
That's what I do not know !

Thanks a lot

cybertza, 2019-04-09:
Look, Logic tells me that since its RS232 ir should be fine since you have a common, mostly people just use the step down since its very cheap and easy.
perhaps just research to get a pull up or something wemos to 232 basically

farkinca1971, 2019-04-09:
Ok thanks for your help ! :)

maragelis, 2019-04-10:
If powering from usb 3 wires is ok tx rx and ground.

farkinca1971, 2019-04-11:
Hello
I ordered a step-down converter, I will try with that.  Thanks

farkinca1971, 2019-04-11:
Hello
Another question : may i trigger alarm via serial somehow? I could not find the command ! :(

maragelis, 2019-04-11:
No you cannot, Not allowed by paradox

farkinca1971, 2019-04-11:
Any workaround to trigger ?

maragelis, 2019-04-11:
yes, I am using a zone on the panel which i trigger with a esp8266/relay and i configured it as a 24Hour zone.

farkinca1971, 2019-04-11:
Can u share the how-to, pls?! I am noob with actual electronic :)
Thanks a bunch

maragelis, 2019-04-11:
get one of these https://shelly.cloud/shelly1-open-source/
attach the I 0 to panel zone,

farkinca1971, 2019-04-11:
Thanks a lot mate. !  :)

farkinca1971, 2019-04-11:
Hi
With the converter it is working ! Thanks guys for the help ! Next step is to do the manual triggering !

closed 2019-04-11 by farkinca1971