maragelis/ParadoxRs232toMqtt #53: PGM

ammuller, 2019-05-25:
Hi, I would like to have a pgm pulse command that turns a PGM on and then off a second later. My alarm has a PGM connected to a garage door opener.

maragelis, 2019-06-09:
do this using node-red pulse command

sparkydave1981, 2020-06-30:
You can also (I think better in this use case scenario) program the PGM to be pulsed via the Paradox programming.  I have the same setup (PGM to garage control) and set it for the minimum pulse duration of 1s.  I haven't yet tried controlling this PGM from Home Assistant yet to confirm that the 1s duration is still observed.

sparkydave1981, 2020-06-30:
How would we code a Home Assistant 'cover' to control the PGM?

Something like this I guess?  (help would be great, thanks. Do we need to send a password?)

```
cover:
  - platform: mqtt
    command_topic: "paradoxdCTL/in"
    device_class: door
    name: Garage Door
    payload_open: "{\"password\":\"1111\",\"Command\":\"PGM_ON\",\"Subcommand\":\"5\"}"
    payload_close: "{\"password\":\"1111\",\"Command\":\"PGM_ON\",\"Subcommand\":\"5\"}"
```

... and then hope the timer does the pulse