maragelis/ParadoxRs232toMqtt #55: More a query - what version of ArduinoJSON is required

georgelza, 2019-06-02:
More a query - what version of ArduinoJSON is required.
It's not listed in the required libraries folder.
Firs time trying to get something compiled for Arduino.
since I have to specify the mqtt broker IP, username and password i guess making this available as a bin file is out of the hand. would have been nice if the build could have been available as a bin and if that could have been configured after first boot, after joining home network/SSID.
G

georgelza, 2019-06-02:
resolved, 5.13.5

zsolt320i, 2019-06-03:
is the remote arming working for you ?
i do have "dologin" (panel login) failure.
spectra4000, 5.23 firmware

georgelza, 2019-06-03:
All that was resolved is I was able to compile the sketch and burn it to the D1, it started up, registered with a test MQTT Broker... once i got all of this working I redid the sketch with the to be expected IP addresses, I'll only know this weekend coming if all goes well, if the communication with the SP6000 panel is working, from there I'll be working remotely to see if I can interface, read zone status's and issue commands.
G

georgelza, 2019-06-04:
ArduinoJSON 5.13.5
Maybe someone can add it to the main project page.
G