maragelis/ParadoxRs232toMqtt #56: This image of sp6000 differ from images in package

georgelza, 2019-06-03:
Hi all... 
so understandably Paradox is not so sharing about what ports is what on the board, or well more than what i can find online atm.

Attached is the best image i can get of a current SP6000, which pins am I to use ?
as far as I can see the image shown as part of the package is the sp4000

Definitely not keen on blowing my board... :)

G

[image]

maragelis, 2019-06-03:
The 4 pins on the top right

georgelza, 2019-06-03:
hmmm, think i might have a IP150 plugged in there already ?
Can i wire my Wemos in also ?
G

maragelis, 2019-06-03:
https://github.com/maragelis/ParadoxRs232toMqtt/issues/16

georgelza, 2019-06-03:
hmmm, the most i saw there was to wire the wemos into series with the IP150, sorry, but not sue how to do this. the images i'm seeing makes me think there is another device now required.

Could it not maybe be possible to wire the IP150 into wemos and then relay messages either direction from there... note I don't know C++ at all so won't be able to code this, hope someone understands my idea and can expand the project. and yes I also need the IP150 for my wife's interaction with the system, she prefers it that way, I primarily want to display the data on HA.
G

cybertza, 2019-06-03:
If all else fails don’t connect the RX pin on the panel from the wemos TX pin, just use RX and power cycle everything, when the 150 logs in the wemos should just understand the messages and process them with some luck.

cybertza, 2019-06-03:
Also just wanted to send a special shout out to paradox for their new firmware, it’s waisted hours of my life.
 
It’s not everyday that you can do a firmware upgrade and loose most functionality.

Sad thing is, this has actually made me not recommend the unit anymore, as before for home automation projects I recommended paradox alarm systems with IP150 modules. 

Now I don’t recommend them at all and if I come across one I build an wemos with power regulator. 

It’s a small matter of time before someone writes an interface for baby ware or the old software, and an web interface for the wemos to take over all the previous ip150 functionality and then no more ip150 sales at all and most likely not loose the connection when you log into the unit...

Honesty most D1 projects already do so much more than the current 150 firmware excluding baby ware.

Greed is feed.

georgelza, 2019-06-03:
... so we will basically just eves drop on messages flowing out of the panel:tx ping.
That I'm guessing will work for the display of data, but damm would have been nice to have been able to also control/arm/disarm/bypass zones in my 2 partitions form HA.
G
Ye... also been a Paradox customer at my previous house for 8-9 years, recently moved, ripped out old unit at the property and installed a SP6000, so regretting this now, as soon as I can will be looking at replacing this most probably with a IDS unit.
G

cybertza, 2019-06-03:
I also wanted to point out that there is code laying around to read data from the keypad bus or the keypad program port to MQTT, their protocol was clock based and clear text on the data bus. If I recall it was 64bytes after the clock was keypad to panel and the other 64 bytes before the clock was panel to keypad. Was fairly easy to use with interupt pin. And there was loads more you could read of the keypad bus like when you arm it even reports 10 seconds before arm; and so, kind of useless info but all there, I recall having to use a relay. I could post the last version of that code if someone wants to complete find the last few bugs. Just had a voltage decider and needed an NPN to write to the bus, I was never able to reliably write to the bus 100% as most of my work was guessing and I think there is an pull to ground before the keypad tx or something, there is a great amount of data on the bus so it’s hard to know what’s important and what not.

georgelza, 2019-06-03:
all way over my head... at the moment wife uses remote to arm/disarm when she goes out, otherwise uses InsiteGold on iPhone to manage system while home.
I'm wanting to integrate into HA, but atm this does not seem do-able.. the RS232 sounded the most promising but atm, looking like thats not a perfect choice, other than just listening to message coming out of the TX port using maragelis project. can't send commands as I have a IP150 installed, have  aided how to get around this, but need someone to climb into maragelis code and see if possible and if yes do the coding, I"m more Python, and my Golang skills are far to "beginner" level to consider playing in C++ in this situation
G

georgelza, 2019-06-03:
hmmm, question, If I configure my Wemos D1 mini to just listen, how will it now which zones are open/close and the state of the alarm, is all of this broadcasted at a regular interval, or at any zone change is all zones state send out ?
G

cybertza, 2019-06-03:
Subscribe to your MQTT server topic with a #  
Eg. paradox/# 
Then look at the messages the HA integration is documented somewhere

cybertza, 2019-06-03:
It’s all documented in the wiki

georgelza, 2019-06-04:
Cyber, think we missing each other, so as per maragelis posting above linking to #16 I can't have my D1 / D1Mini connected and the IP150... well if I want to do this I can only connect my Rx to the Panel Tx. other wise i'm just eves dropping.
Subscribing to MQTT is not a challenge and figuring out what which message means, ye thats all in the events.txt listed.
I was asking 
#1 if I just subscribe it implies I can send a status request of sort to get a full board status. so either the board needs to regularly publish all zone status's for me to paint a full picture, or I wait...
#2 my other idea was, would be nice if we could get to a point where we rather wire the IP150 onto the D1/D1Mini, and then have the D1* play relay, when ever a msg comes from the panel it pushes that down the MQTT topics, but also copies it onto the new pins where the IP150 is and similarly, if a msg comes from the IP150 it's send down the MQTT topics but also send to the panel, 
#3 and then of course if there is a msg coming down the MQTT stream, that is then send to the panel (I'm not sure if the IP150 needs to be told about this message, thinking it might do everything as a "whats the status" call on wakeup/login).
G

maragelis, 2019-06-04:
On the ip150 there is a second serial, it’s marked io2, this is used by paradox to drive second serial bus devices. You can connect the wemos on that connection. But I don’t know the pin out of this connector. 

There are quite a lot of messages coming in every second, so the wemos is quite busy. Writing a web interface takes up quite a lot of resource. When you have hassio or HomeKit there is no need for a web interface. 

Adding a second serial to the wemos is nothing hard. It’s the wiring I am afraid people won’t know how to do. If it’s something, useful I can add it as a module in the code. For better results though you will have to wait for the esp32 version I am working on as the esp32 has three hardware serial ports.

georgelza, 2019-06-04:
... On the ip150 there is a second serial, it’s marked io2, this is used by paradox to drive second serial bus devices. You can connect the wemos on that connection .... thanks, i will try and find out which is which on this out, which is TX and which is RX.
sorry, not planning on writing a web interface, if it came over like that, will be pulling data into Home Assistant, either directly of the MQTT Broker or via Node Red.

I'm personally def thinking wiring in the wemos between the panel and the IP150 could be a interesting fix/solution... as a 2nd option compared to going with the above io2 option.that way the wemos will see all comm between the IP150 and the panel and can then be configured to selectively send what we want down the MQTT topics

Can you tell me bout the esp32 ? depending on time line and capabilities might just make sense to put this build/now on ice.

My biggest logistic thing is... my house, with system and wife/kids are In Jeffreysbay, 1000km from me in Johannesburg.
So I need to compile/flash and install every 3rd weekend when I'm down there, from there I then have to do everything remotely... also ... ie why I've like the Raspberry Pi and sonoff platforms till now as I can do minimal conf while I'm there and then do everything else remotely via ssh.

G

mydakota, 2019-06-04:
Top right port serial

maragelis, 2019-06-04:
@georgelza  maybe this project would be more suited for you https://github.com/jpbarraca/pai 
it uses a raspberry pi and can connect to ip150. I was born and raised in SA moved to Greece when I was 20 right after High School.

georgelza, 2019-06-04:
From where in SA ? Till when... small world.
I've looked at the api project that web interacts with the IP150, experience so far, it kills the IP150 and the InsiteGold link.
G

maragelis, 2019-06-04:
Durban Natal. Left SA 1991

georgelza, 2019-06-04:
any type of time line you can give for the esp32 project, bit more detail, if need, in pvt.
georgelza@gmail.com
G