maragelis/ParadoxRs232toMqtt #57: Firmware download

au190, 2019-06-04:
Hi

Ian using 5.23 firmware version on sp4000.
Events are comming but I can not send arm commands. Arm commands rejected with 0x70 - login error.

I would like to downgrade the firmware. 
Have you seen older version of firmware ?
As I see on this topic pictures its using older firmware version on sp4000

zsolt320i, 2019-06-04:
Same problem here

maragelis, 2019-06-09:
try changing your arm code I have seen problems with some firmware versions dont use 0 in your code

maragelis, 2019-06-09:
if you need paradox firmware send me mail i will send you how to get

au190, 2019-06-09:
I do not have 0 in my arm code.
I will try to somehow monitor the output, if I find the issue I will update.
Any suggestion will be welcome. 
Thx

maragelis, 2019-06-09:
Are you using the Rx tx pins or the software pins ? Post problems with logins are pin not connected properly, other device already logged on.

au190, 2019-06-14:
Finally I realised the I have 0 in my code :) so I chaged it and now its working properly with 5.23 firmware version.

I want to thank you for your help ! Your sugession was good.

closed 2019-06-14 by au190