maragelis/ParadoxRs232toMqtt #59: unable to compile sketch

sparkydave1981, 2019-06-25:
Unable to compile sketch in Arduino IDE, version 1.8.9

esp8266 board installed in board manager, all libraries as per Git repo installed. 

error: 
ParadoxAlarmSystemOTA:1:19: error: FS.h: No such file or directory
compilation terminated.
exit status 1
FS.h: No such file or directory

File exists on PC in Arduino library, where is this looking for the file?3

Additional info here:
https://community.home-assistant.io/t/paradox-alarm-panel-wemos-d1-mini-communication/62003/12

sparkydave1981, 2019-07-08:
After seeking some help I managed to get the sketch to compile.  Setting required:
Arduino IDE version 1.6.13 (wouldn't work with 1.8.9)
ESP8266 core 2.4.1
WifiManager by tzapu 0.12.0
PubSubClient by Nick O`Leary 2.6.0
Arduino JSON lib version 5.13.5  (won't work with version 6.x)

john2oo2, 2020-03-15:
Using Arduino IDE v1.8.12
Install this from "Tools" -> "Board: "some boad" -> "Boards Manager":
 - ESP8266 Community v2.6.3
Install these from "Tools" -> "Manage Libraries":
 - WiFiManager by tzapu v0.15.0
 - PubSubClient by Nick O'Leary 2.6.0
 - Arduino JSON lib version 5.13.5 (**does not currently work with version 6.x**)

Files to be changed:
 - Open webpage.ino:
1) From "String page = FPSTR(**HTTP_HEAD**);" to "String page = FPSTR(**HTTP_HEADER**);"
2) From "page += FPSTR(**HTTP_HEAD_END**);" to "page += FPSTR(**HTTP_HEADER_END**);"

Hope I did not miss anything...