maragelis/ParadoxRs232toMqtt #60: Weird messages received in the OUT topic

Flopescu, 2019-07-09:
Hello!
First of all, thanks for your work, I really appreciate what you have done.

I have tried to get this working in the past week, could not make it work on the wemos d1 mini, tried with all versions available (including dev) and all esp8266 core that I found referenced in the issues.
During the weekend I flashed one wemos d1 pro ... and wow, out of nowhere I started receiving messages in the out topic - I used 2.2.4 firmware with esp8266 2.5.2 and pubsubclient.h having the mqtt max packet size set at 256 

The problem that I have now is that the messages that I received cannot be found in the events / sub-events list and from what I have followed they are changing for the same action and therefore the arm/disarm/sleep actions are not retrieved.

I do have some messages that are translated correctly, and these are the zones ON and OFF, and are sent to the hasio/<zoneX> topic.

But the all of the arm / disarm actions (via remote or panel) are not translated or sent to any hassio topic. Further more, if I send any json command to the in topic the wemos disconnects in 15 seconds or so. Note that I let the wemos connected for more than 2-3 hours, no problems ... as long as I don't send any commands.

The connection I have to the board is temporary - I used an external 5v adapter and connected to the serial pins only the TX and RX via small jumper wires (could this be the issue?)
Here's the list of the messages I get:

```
{"event":1,"sub_event":7,"sub_eventD":"","eventD":"Zone open ","data":"one 07    09#��"}
{"event":1,"sub_event":7,"sub_eventD":"","eventD":"Zone open ","data":"���\t�"}
{"event":7,"sub_event":0,"sub_eventD":"","eventD":"","data":"on_L�ߐ$"}
{"event":2,"sub_event":11,"sub_eventD":"Disarm partition","eventD":"Partition status","data":"��$$\nɕ��1    �"}
{"event":2,"sub_event":12,"sub_eventD":"Arm partition","eventD":"Partition status","data":"`�b\nɕ��1"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"Ar���"}
{"event":3,"sub_event":2,"sub_eventD":"Bell squawk arm","eventD":"Bell status","data":"���B"}
{"event":255,"sub_event":111,"sub_eventD":"","eventD":"","data":"Zone 07 # �"}
{"event":80,"sub_event":177,"sub_eventD":"","eventD":"","data":"Area 1 �/"}
{"event":2,"sub_event":9,"sub_eventD":"Squawk OFF","eventD":"Partition status","data":"�\t\nɕ��1      �"}
{"event":9,"sub_event":0,"sub_eventD":"","eventD":"","data":"A�����"}
{"event":7,"sub_event":0,"sub_eventD":"","eventD":"","data":"one �7;\t"}
{"event":9,"sub_event":0,"sub_eventD":"","eventD":"","data":"A��zHL"}
{"event":147,"sub_event":26,"sub_eventD":"","eventD":"","data":"�O@"}
{"event":160,"sub_event":8,"sub_eventD":"","eventD":"","data":"A���'H"}
{"event":15,"sub_event":141,"sub_eventD":"","eventD":"User code activated output","data":"��'"}
{"event":5,"sub_event":0,"sub_eventD":"","eventD":"","data":"one 050�'"}
{"event":16,"sub_event":194,"sub_eventD":"","eventD":"Wireless smoke maintenance signal","data":"one04 ���"}
{"event":66,"sub_event":77,"sub_eventD":"","eventD":"","data":"ne04���G@"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"stem Myer"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"p��"}
{"event":2,"sub_event":11,"sub_eventD":"Disarm partition","eventD":"Partition status","data":"\"\nɕ��1     8"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"�Ov��1"}
{"event":2,"sub_event":11,"sub_eventD":"Disarm partition","eventD":"Partition status","data":"�\nɕ��1    8��"}
{"event":0,"sub_event":2,"sub_eventD":"","eventD":"Zone OK","data":"��<1     ^�"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"���"}
{"event":194,"sub_event":15,"sub_eventD":"","eventD":"","data":"Area 1<Ҡ"}
{"event":141,"sub_event":254,"sub_eventD":"","eventD":"","data":"Area(7 �"}
{"event":2,"sub_event":14,"sub_eventD":"Exit delay started","eventD":"Partition status","data":"��  Area 1     `"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"��c 1"}
{"event":11,"sub_event":0,"sub_eventD":"","eventD":"","data":"�4��׮V�1"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"@�"}
{"event":35,"sub_event":2,"sub_eventD":"","eventD":"Zone bypassed","data":"x��"}
{"event":177,"sub_event":33,"sub_eventD":"","eventD":"","data":"Area�b"}
{"event":3,"sub_event":154,"sub_eventD":"","eventD":"Bell status","data":"p���"}
{"event":2,"sub_event":251,"sub_eventD":"","eventD":"Partition status","data":"Area 1��b"}
{"event":0,"sub_event":0,"sub_eventD":"","eventD":"Zone OK","data":"ne ��"}
{"event":1,"sub_event":7,"sub_eventD":"","eventD":"Zone open ","data":"����07"}
{"event":2,"sub_event":9,"sub_eventD":"Squawk OFF","eventD":"Partition status","data":"\nɕ��1   `�"}
{"event":3,"sub_event":2,"sub_eventD":"Bell squawk arm","eventD":"Bell status","data":"�"}
```

If you have any suggestion or if there is something that I can do to trace it further (the above list is from mqtt trace)

PS - in the wiki, is there a typo for the **in** topic, is there a missing "d" ???
 --> paradoxCTL/in 
instead of 
--> pradox**d**CTL/in 
Thanks!

cybertza, 2019-07-09:
> I do have some messages that are translated correctly, and these are the zones ON and OFF, and are sent to the hasio/ topic.

Yea that’s your issue, you need a ground 232 requires a common ground for it to work 

Regards

cybertza, 2019-07-09:
Also this may be under correction but I think the D is the dev build, the topic is defined in code, so can be changed close to the top. You should be able to use a lm regulator they super cheap, and that how I run mine.

I’m far from the current build, but why fix it if it’s not broken  :)

Regards

cybertza, 2019-07-09:
I also found the market flooded with a cheaper d1 clone, the chip is “ok” but the regulator seems to be unstable causing loads of unforeseen issues.

cybertza, 2019-07-09:
For now you should be able to just common the ground between the esp grnd pin and the panel grnd,  provided the PSU is semi decent  it shouldn’t cause harm.

Flopescu, 2019-07-09:
Thanks for your quick replay.

Please believe me that I have tried sooo many things until I believe it was the alarm installation at fault. 
The alarm was already installed and configured when I got it, so at first I believed it was the GPRS module installed or the phone line added... (took them out... of course it triggered the alarm for minutes ... )
So, last week I had hours and hours flashing in and out my d1 mini (ripoff) and tried with rx-tx, rx-tx, flash, another core, another pubsub, with that ground on /off, with the computer on micro-usb, with a lot more of serial messages  ... all that without a result ...
-- and now that I just remembered, when I navigate to the board IP shoudn't I get the content of what's in the "webpage.ino"? The thing is that I only have "padaroxdCTL" text on that page and nothing else..

Anyway, I will try and post here my results. The thing is that I have already bought 2 shields for the pro already, waiting for them to be delivered, as I thought it would be "easier" to just but the shield on top of the board and there I have it ... power and board.
In the meantime I've seen that Maragelis has added a few resistors (in his photos). Do I need to add any resistors or a simple TX->RX and RX->TX plus power should do it?

Thanks!

cybertza, 2019-07-10:
The resistors should not be the issue, 

 

The esp is slightly 5V tolerant, I don’t run resistors, but RS232 requires 3 pins to work, TX,RX,GRND otherwise you are in for the random chars you are getting.

Flopescu, 2019-07-16:
After about one week of intense testing, everything is working perfectly with a proper step-down converted and the wemos connected directly to the panel.

During the build and testing I managed to burn the wemos d1 pro and I had to use the wemos d1 mini which, without ground from the panel did not show any activity on the OUT mqtt topic.

Just a heads up for anyone who is trying to make it work: 

--> You can edit the firmware file () and add password to the board hotspot. 
Add the password on this line:  
`if (!wifiManager.autoConnect(Hostname, "")) {`
Like:  
`if (!wifiManager.autoConnect(Hostname, "password")) {`

Now your board settings are secure. And if your board cannot connect to your WiFi it will create a hotspot with the password you've added.

--> After flashing and configuring wifi and mqtt, the only test you can make is to check if the board has connected to your WiFI and mqtt topic. 
You need to connect the board to the panel in order to send/receive info on the mqtt topics. The ground from the panel is a must. Everything else will just fail.

--> For Home Assistant use platform: mqtt, intead of manual_mqtt like: 
```
alarm_control_panel:
  - platform: mqtt
    name: Home Alarm
    state_topic: "paradoxdCTL/hassio/Arm"
    command_topic: "paradoxdCTL/in"
    code: 1234     # <- this is optional, it's the code use to arm/disarm from Home Assistant
    payload_disarm: '{"password":"0000","Command":"disarm","Subcommand":"0"}'
    payload_arm_night: '{"password":"0000","Command":"sleep","Subcommand":"0"}'
    payload_arm_home: '{"password":"0000","Command":"stay","Subcommand":"0"}'
    payload_arm_away: '{"password":"0000","Command":"arm","Subcommand":"0"}
```
This way it will also be sent to homekit, if you have it enabled.

Again, big thanks for everything!

sparkydave1981, 2019-07-16:
Hi Flopescu, thank you so much for posting this code for Home Assistant.  I am in the process of trying to get this working myself so this should be a huge help.  One question though, I currently have an alarm_control_panel setup in HA and really just want to get the Paradox alarm to coordinate with HA such that if either one is armed/disarmed then the other alarm follows.  Do you know how I can integrate the mqtt from this Paradox interface to my existing HA alarm panel?

Flopescu, 2019-07-16:
If you leave the topics as default in the firmware then all you have to do is to add the state_topic and command_topic to your alarm_control_panel in HA as above.
The HA will listen to the state_topic and post the payloads in command topic. 
That's all.

If you want, you can add a few sensors to monitor the zones and perhaps trigger additional notifications if you like.
The alarm zones mqtt topics are :  "paradoxdCTL/hassio/zone_<X>_"  with values of ON and OFF

Something like (including the status):

```
sensor:
- platform: mqtt
  name: alarm_status
  state_topic: 'paradoxdCTL/hassio/Arm'
- platform: mqtt
  name: alarm_zone_1
  state_topic: 'paradoxdCTL/hassio/zone1'
- platform: mqtt
  name: alarm_zone_2
  state_topic: 'paradoxdCTL/hassio/zone2'
- platform: mqtt
  name: alarm_zone_3
  state_topic: 'paradoxdCTL/hassio/zone3'
- platform: mqtt
  name: alarm_zone_4
  state_topic: 'paradoxdCTL/hassio/zone4'
- platform: mqtt
... ...
```
Enjoy

sparkydave1981, 2019-07-16:
Thank you. Just so I can understand this a bit better, those alarm_zones (`/hassio/zone1`) relate to the physical zones connected to my Paradox system?  I know I have been able to see (using MQTT Lens) the data coming from my Paradox alarm when I trigger a zone, but that data was coming from `paradoxdCTL/out` rather than `paradoxdCTL/hassio/` and I never saw any data come from the `/hassio` topic