maragelis/ParadoxRs232toMqtt #61: Hi

favarinbarracano, 2019-08-07:
I have a Paradox EVO 192 alarm system.
The board don't reveal the alarm through serial port.
Any advice?
Thanks,

Manuel

maragelis, 2019-09-23:
Sorry cannot help. I havent seen or worked with an EVO panel

y1ann1s, 2019-10-02:
I am working on EVO192, @maragelis  can you please some info on the protocol. I find some differences on the location of data within the 37 bytes. I have decoded the received data and adjusted your code. But I have a long way to go on the send data, any help here is welcome
@favarinbarracano  EVO192 has a serial port on the board ,If you have an ip100/150  it is exposed through there also
Thanks

mydakota, 2019-10-25:
Had mine working well on my EVO48 along time ago had to make small changes but was 90% working I would like to try it on the EVOHD out of interest

pilievwm, 2019-10-25:
@mydakota you mean your EVO is working with @maragelis code without any changes?