maragelis/ParadoxRs232toMqtt #63: Paradox Magelan MG-5050 alarm system

erelis77, 2019-10-16:
Hi,
you have created a great product. I replicate it.

When I connect to the Paradox Magelan MG-5050 alarm system.
Through MQTT I get only one status from 
„**paradoxdCTL/status**“ - disarmed 
and 
"**paradoxdCTL/hassio/Arm**"- disarmed

I created binary sensors for Home Assistant but the statuses do not change :
  - platform: mqtt
    name: Sensor_Tamburas_Durys
    state_topic: "paradoxdCTL/hassio/zone1"
    qos: 0
    device_class: door

  - platform: mqtt
    name: Sensor_Virtuves_PIR
    state_topic: "paradoxdCTL/hassio/zone2"
    qos: 0
    device_class: motion 

[image]
[image]
[image]
[image]

Please help find where the problem is and how to fix it.

maragelis, 2019-10-18:
Two things to check is if the RX/TX are the right way.
Second read the wiki for Hassio https://github.com/maragelis/ParadoxRs232toMqtt/wiki/Home-Assistant-in-V2

maragelis, 2019-10-18:
One more thing check mqtt channel paradoxdCTL/# with a mqtt client to see if you are receiving updates.

erelis77, 2019-11-04:
Hi,
thank you for the quick reply. 
After changing the MQTT request to paradoxdCTL/# , I get the following statuses
„paradoxdCTL/status“ - disarmed
"paradoxdCTL/hassio/Arm"- disarmed
"paradoxdCTL/hassio/zone1"- OFF
"paradoxdCTL/hassio/zone2"- OFF

I checked the security system and found that the Pir sensor and door sensor is connected to zone1. Is it possible to get the status of both different sensors when both sensors are connected to the zone1.

mydakota, 2019-11-04:
That's called zone doubling Paradox can do so e doubling in series or parelil my spelling OMG anyway depending on how it's wire will effect the EOL resistance used check manual for config and programing if required

erelis77, 2019-11-04:
I know it is possible to combine two sensors into one zone, but how to get status about each sensor with MQTT

maragelis, 2019-11-05:
If both sensors are in the same zone number. No system can distinguish between them. If they are using different zone numbers then you will get status of both separately.

mydakota, 2019-11-05:
Assuming you have 8 zones on control board Zone doubling using zone 1 will use zone 1 and 9 zone 2 will be zone 2 and zone 10 and so on .
It is possible to configure zone 1 to be zone 2 or zone 22 but there's no reason you would do this in programing . Just use winload or babyware and check your zone programing and zone doubling options. If you have a ip100 or ip150 you can used to program with either winload or babyware otherwise you'll need a 307adapter USB to serial .
If you don't know the programing and have zones higher than inputs on your main pcb you would assume zone doubling

closed 2020-01-29 by maragelis