maragelis/ParadoxRs232toMqtt #65: partition topic issue

zsoltkazda, 2020-01-21:
Hi Maragelis,

My paradox have two partitions. I would like to use these partition separate. Now the state and command topics is the same on the different partitions. This is the source of my problem.
Now when I disarm one of the partitions (onyl one), after that all partitions disarmed, and vice versa too.

Regards
Zsolt

maragelis, 2020-01-21:
Just to be clear, disarming and arming works ? Just the hassio topic does not send partion number, correct ?

zsoltkazda, 2020-01-21:
Yes, the disarming and arming works, and yes, the hassio topic does not send partition number.

maragelis, 2020-01-21:
Test branch updated. 
the topic on which arm status is sent has partition info. 
eg. paradoxdCTL/hassio/Arm0 paradoxdCTL/hassio/Arm1
https://github.com/maragelis/ParadoxRs232toMqtt/tree/test

john2oo2, 2020-03-10:
> Yes, the disarming and arming works, and yes, the hassio topic does not send partition number.
@zsoltkazda could you please provide your panel info because this does not work for me?
I am using the MG5050 with "MG5050_4.73.72.PUF" firmware from 2011.
I can connect to the panel and receive the following topics:
 - paradoxdCTL/out
 - paradoxdCTL/status
 - paradoxdCTL/hassio/zoneX

maragelis, 2020-03-10:
Change your code on the alarm and dont use 0 in the code

john2oo2, 2020-03-10:
@maragelis my alarm panel user code is 4 non-zero digits.
Trace indicates that "sending command 0x5f to panel" is executed but nothing after that. 
When I compare the login sequence by sniffing the serial port between the MG5050 and the Paradox BabyWare application I see there seems to be a different initialisation sequence. 
When I first bought my panel around 2010, it worked with the WinLoad software. Then after upgrading the firmware the WinLoad software would no longer work and I needed to use the BabyWare application.
Could it be that there is a difference?
[filtered_response.txt](https://github.com/maragelis/ParadoxRs232toMqtt/files/4314648/filtered_response.txt)

maragelis, 2020-03-11:
Looking at the trace I see a perfect login after 5f there is the 00 handshake and a 10 login. Then the is status command 50 52 issued. Do you get login failed from the status mqtt topic when trying to arm? If not it’s surly the code try the master code or change your code.

john2oo2, 2020-03-11:
The trace provided is the communication between BabyWare and the MG5050, so the perfect login is expecetd. Let me first modify the code so that I initialise the communications with the "72 00 00 .... 00 00 72" packet and then see if the panel responds at all.

zsoltkazda, 2020-03-11:
Dear john2oo2!

I don’t know the firmware version. But ont he panel, I see the following nformation:
SP6000 NR Q018
27aug19

Zsolt

maragelis, 2020-03-11:
Do you get login failed from the status mqtt topic when trying to arm?

john2oo2, 2020-03-15:
**Feedback regarding the "login" issue:**
 - Turns out that my issue with the "login" was not a programatic error but rather a hardware issue!
 - I first used a knock-off D1Mini version. The serial ports of this device did not handle the 5V TTL levels and popped receive port.
 - I then got hold of the WEMOS D1 Mini and installed a 5V <-> 3V TTL level converter which also gave problems.
 - Then I threw out the level converter and connected the WEMOS D1 Mini directly as indicated on the original WIKI page.

**Feedback regarding the reporting of the Arm Status:**
 - The **arm** and **disarm** command is always reported correctly for both Partion 1 (0) and 2 (1).
 - There is still an **issue** where the **sleep** and **stay** command is actioned, but the reporting only happens for the first partition that  is put into the sleep or stay armed mode.

bigjuanpa, 2021-05-17:
how can you get the panelstatus information for the panel 2 (Arm1)?
When I run 
{
  "Command": "panelstatus",
  "Subcommand": "1",
}

I only get the Arm0 status

thanks

maragelis, 2021-05-17:
Its not implemented in this project I have it implemented  on my other project

sparkydave1981, 2021-05-18:
So to get partition support we should swap to using an ESP32 with your other project?