maragelis/ParadoxRs232toMqtt #69: Nothing from paradoxdCTL/# after receiving Paradox Connected

ppan21, 2020-04-18:
Hello,

I have a Paradox 1759EX alarm and I flashed a Nodemcu ESP8266 with ParadoxAlarmSystemOTA.ino
I connected RX/TX and Ground between the Paradox board and the ESP8266 relevant pins.
The Nodemcu ESP8266 is powered by USB.
On mqtt.fx I have subscribed to topic paradoxdCTL/# and when i boot the ESP8266 I get:
"status":"Paradox connected"
and
"firmware":"SYSTEM PARADOX_2.2.3"

after that I don't get any other message from mqtt from any zone when it opens or closes.

Any ideas why and how to get the status from my zones?

Also when I try to send a command to paradox (e.g arm) I publish on topic paradoxdCTL/in the payload { "password":"1234", "Command":"arm", "Subcommand":"0" } nothing happens and after a few seconds I receive on paradoxdCTL/status the message "Paradox Disconnected" and after around 2 minutes I receive on the same topic the message "Paradox connected".

Any idea how to proceed?
Thank you in advance.

Flopescu, 2020-04-18:
I had the same problem.
Your issue is that you have the esp powerd via an external source.

Connect the esp to the paradox serial +/- . USE A PROPER  DC stepdown converter to 5v as the board delivers 12v

ppan21, 2020-04-18:
Thank you @Flopescu, I will buy one and I'll try.

maragelis, 2020-04-23:
Connect the ground pin from the Panel to the esp. Common ground connection.

ppan21, 2020-04-23:
Καλημέρα Παναγιώτη,

I have already connected the ground of Paradox serial connector to the ESP ground pin and also the TX/RX pins respectively and as I mentioned in my initial post I only get on mqtt.fx "status":"Paradox connected" and "firmware":"SYSTEM PARADOX_2.2.3", and after that I don't get any other message from mqtt from any zone when it opens or closes.

I will try to connect the ESP8266 to a DC stepdown converter, as soon as I will get it, as advised by  @Flopescu.

ppan21, 2020-05-05:
Hello,

I received today the DC stepdown converter and connected it to 5v and GND of the WEMOS D1 as suggested by @Flopescu and also the respective RX/TX pins.
Unfortunatelly, the situation is the same with the previous one (when I have connected only the GND pin between Serial and WEMOS D1 and powered the WEMOS D1 through usb 5v), i.e. when I reset the WEMOS D1, I receive on mqtt.fx for the topic paradoxdCTL/# ==> "status":"Paradox connected" and "firmware":"SYSTEM PARADOX_2.2.3" and after that nothing else if open one of the Zones of the alarm system.

I also noticed that, when I publish on topic paradoxdCTL/in the payload { "password":"1234", "Command":"stay", "Subcommand":"0" } nothing happens and after a few seconds (30-40 sec) I receive on topic paradoxdCTL/status the message "Paradox Disconnected"

Do you any idea why I cannot receive nothing except the initial connection message?

ppan21, 2020-05-11:
I have connected a serial FTDI to the TX pin of my Paradox 1759EX and unfortunately, I didn't get the 37 bytes. Instead for each event I got two chunks of 4 bytes each (8 bytes in total).
So, I suppose that my paradox serial protocol is different from MG4000, MG5000 and MG7000.

Angel0ffDeath, 2020-06-04:
Yes - protocol is different

closed 2020-06-06 by ppan21