maragelis/ParadoxRs232toMqtt #72: Wemos D1 Mini Pro - frequent re-connects

qntris, 2020-04-26:
Hi,

I am using a Wemos D1 Mini Pro and a buck converter connected to the Gnd and + serial pins of the MG5050 alarm system. I have properly soldered the pins and checked the connections several times, confirmed that the system is working (I could read/arm/disarm using MQTT). However after ~20-30 minutes the Wemos D1 Mini Pro is disconnected (for no obvious reason) and does not come back online. I have resoldered the connections precisely, tested and connected back to the alarm system. Same thing happened again after a few minutes.

Now I have connected the power to the 3v pin (stepped down the voltage with the buck converter) and testing it for a few minutes but I still see disconnects - Socket error on client 84:F3:EB:DB:5B:06, disconnecting. (at least with quick reconnects this time).

Should I be targeting a faulty Wemos module? I have a NodeMCU board that I have already programmed and is waiting in the shelf. If those disconnects continue, I will be trying it instead of the Wemos board.

Angel0ffDeath, 2020-06-04:
You can upgrade esp8266 core to 2.7.1 (last release) - tested. However try to compile with WiFiManager by tzapu v0.12.0 as it is stated in readme file and then report again the results

qntris, 2020-06-04:
Hi @Angel0ffDeath,
I have tried with:
Arduino 1.8.12

- ESP8266 core 2.7.1
- WifiManager by tzapu 0.12.0
- PubSubClient by Nick O`Leary 2.6.0
- ArduinoJSON 5.13.4

The sketch won't compile for D1 Mini Pro:
`
In file included from /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,
                 from /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/Arduino.h:32,
                 from sketch/ParadoxAlarmSystemOTA.ino.cpp:1:
/Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))
                                                                                                                                  ^
/Users/User/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM'
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                        ^
In file included from /Users/User/Downloads/ParadoxRs232toMqtt-20190218/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino:6:0:
/Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src/ESP8266WebServer.h:34:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
exit status 1
Error compiling for board Generic ESP8266 Module.
`

Angel0ffDeath, 2020-06-04:
Hi, Sorry but it really won’t compile. Sorry my fault… forgot to give some instructions… 😊You need to install WiFiManager by tzipu 0.12.0After that please extract the attached file and replace both files in Arduino folder/libraries If you have some problems or you don’t have RAR, please download actual version of WINRAR – it is for free.  Then try to compile again. It should compile 😊If not report any troubles.After that do not forget to report troubles with your alarm also. Good luck Sent from Mail for Windows 10 From: qntrisSent: 04 юни 2020 г. 21:10To: maragelis/ParadoxRs232toMqttCc: Angel0ffDeath; MentionSubject: Re: [maragelis/ParadoxRs232toMqtt] Wemos D1 Mini Pro - frequent reconnects (#72) Hi @Angel0ffDeath,I have tried with:Arduino 1.8.12ESP8266 core 2.7.1WifiManager by tzapu 0.12.0PubSubClient by Nick O`Leary 2.6.0ArduinoJSON 5.13.4The sketch won't compile for D1 Mini Pro:In file included from /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0, from /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63, from /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/Arduino.h:32, from sketch/ParadoxAlarmSystemOTA.ino.cpp:1: /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "." __STRINGIZE(__COUNTER__) "\""))) ^ /Users/User/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM' const char HTTP_HEAD[] PROGMEM = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>"; ^ In file included from /Users/User/Downloads/ParadoxRs232toMqtt-20190218/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino:6:0: /Users/User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src/ESP8266WebServer.h:34:39: error: previous declaration of 'HTTPMethod HTTP_HEAD' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ exit status 1 Error compiling for board Generic ESP8266 Module.

qntris, 2020-06-04:
Hi @Angel0ffDeath , I don't see any attachment and the post seems a bit off.

Angel0ffDeath, 2020-06-04:
If you don’t see it now (although it was attache in previous mail), this means gmail somehow remove it. Sent from Mail for Windows 10 From: qntrisSent: 04 юни 2020 г. 23:50To: maragelis/ParadoxRs232toMqttCc: Angel0ffDeath; MentionSubject: Re: [maragelis/ParadoxRs232toMqtt] Wemos D1 Mini Pro - frequent reconnects (#72) Hi @Angel0ffDeath , I don't see any attachment and the post seems a bit off.

Angel0ffDeath, 2020-06-04:
If you don’t have the attached file send it via some file share service Sent from Mail for Windows 10 From: qntrisSent: 04 юни 2020 г. 23:50To: maragelis/ParadoxRs232toMqttCc: Angel0ffDeath; MentionSubject: Re: [maragelis/ParadoxRs232toMqtt] Wemos D1 Mini Pro - frequent reconnects (#72) Hi @Angel0ffDeath , I don't see any attachment and the post seems a bit off.

qntris, 2020-06-04:
Can you zip it or upload somewhere and provide a link?

As for the reconnects, I am actually quite puzzled, the device works flawlessly:
- with another AP that I have on the same network (different room)
- with my main router but if I am pinging the device from the other AP (if I try to ping it from a device that is hooked up via LAN to my main router, it starts reconnecting every few minutes again). 

My main router is TP-Link Archer C7 and the secondary one is TP-Link RE305 range extender, set up as Access Point, DHCP turned off, hooked up to the main router via cable. The main router and the RE305 are far away from each other, the beacon interval and the DITM interval is the lowest possible. The SSIDs and the passwords of the 2.4ghz and 5ghz networks are the same. Tried changing them as I suspected that the Wemos might be switching them, but that did not help either. I checked all the settings on my main router and on the RE305 and I honestly don't have any idea, why it would not be reconnecting when hooked up to the RE305 or when hooked up to the main router and being pinged from the RE305.... The signal strength is excellent in both rooms, the interference is very small. Other ESP8266 based devices (smart sockets) work without any issues in both rooms...

Angel0ffDeath, 2020-06-04:
Here is the file: https://gofile.io/d/GLQf8d Will read more carefully now your mail below and will write you additionally if necessary  Sent from Mail for Windows 10 From: qntrisSent: 04 юни 2020 г. 23:59To: maragelis/ParadoxRs232toMqttCc: Angel0ffDeath; MentionSubject: Re: [maragelis/ParadoxRs232toMqtt] Wemos D1 Mini Pro - frequent reconnects (#72) Can you zip it or upload somewhere and provide a link?As for the reconnects, I am actually quite puzzled, the device works flawlessly:with another AP that I have on the same network (different room)with my main router but if I am pinging the device from the other AP (if I try to ping it from a device that is hooked up via LAN to my main router, it starts reconnecting every few minutes again).My main router is TP-Link Archer C7 and the secondary one is TP-Link RE305 range extender, set up as Access Point, DHCP turned off, hooked up to the main router via cable. The main router and the RE305 are far away from each other, the beacon interval and the DITM interval is the lowest possible. The SSIDs and the passwords of the 2.4ghz and 5ghz networks are the same. Tried changing them as I suspected that the Wemos might be switching them, but that did not help either. I checked all the settings on my main router and on the RE305 and I honestly don't have any idea, why it would not be reconnecting when hooked up to the RE305 or when hooked up to the main router and being pinged from the RE305.... The signal strength is excellent in both rooms, the interference is very small. Other ESP8266 based devices (smart sockets) work without any issues in both rooms...

Angel0ffDeath, 2020-06-04:
Ok. To answer on your main e-mail. First of all Wemos devices do not know what is 5GHz, so there is no problem with same name. Wemos only sees 2.4GHz network. So this should not be you problem although both networks (2.4 & 5 GHz have the same SSID). Try to recompile the source as I suggested and report again what is going on… Regards Angel Sent from Mail for Windows 10 From: qntrisSent: 04 юни 2020 г. 23:59To: maragelis/ParadoxRs232toMqttCc: Angel0ffDeath; MentionSubject: Re: [maragelis/ParadoxRs232toMqtt] Wemos D1 Mini Pro - frequent reconnects (#72) Can you zip it or upload somewhere and provide a link?As for the reconnects, I am actually quite puzzled, the device works flawlessly:with another AP that I have on the same network (different room)with my main router but if I am pinging the device from the other AP (if I try to ping it from a device that is hooked up via LAN to my main router, it starts reconnecting every few minutes again).My main router is TP-Link Archer C7 and the secondary one is TP-Link RE305 range extender, set up as Access Point, DHCP turned off, hooked up to the main router via cable. The main router and the RE305 are far away from each other, the beacon interval and the DITM interval is the lowest possible. The SSIDs and the passwords of the 2.4ghz and 5ghz networks are the same. Tried changing them as I suspected that the Wemos might be switching them, but that did not help either. I checked all the settings on my main router and on the RE305 and I honestly don't have any idea, why it would not be reconnecting when hooked up to the RE305 or when hooked up to the main router and being pinged from the RE305.... The signal strength is excellent in both rooms, the interference is very small. Other ESP8266 based devices (smart sockets) work without any issues in both rooms...

qntris, 2020-06-04:
@Angel0ffDeath , can you delete your posts as they are spamming the topic - GitHub changes the content as it seems

Angel0ffDeath, 2020-06-04:
Will do. Contact me at angofdeath743@gmail.com I didn’t imagine when Im sending you an e-mail via gmail it is posted in GitHub… Will delete!!! Sent from Mail for Windows 10 From: qntrisSent: 05 юни 2020 г. 0:15To: maragelis/ParadoxRs232toMqttCc: Angel0ffDeath; MentionSubject: Re: [maragelis/ParadoxRs232toMqtt] Wemos D1 Mini Pro - frequent reconnects (#72) @Angel0ffDeath , can you delete your posts as they are spamming the topic - GitHub changes the content as it seems

Angel0ffDeath, 2020-06-04:
Done... Sorry

Angel0ffDeath, 2020-06-04:
@qntris - did you receive the file?

qntris, 2020-06-04:
Yeap, will test it first thing tomorrow (it’s late on my end) and report back. I however doubt that this is the problem, as the board is quite steady on my AP and seems to be disturbed when on the main router...

Angel0ffDeath, 2020-06-04:
just try, then disturb...

Angel0ffDeath, 2020-06-04:
here is also late. Will wait your feedback

qntris, 2020-06-05:
@Angel0ffDeath it compiled, but the result is the same - disconnects every couple minutes
[image]
Any ideas?

maragelis, 2020-06-05:
What serial connections are you using. Software or hardware? . Rx/TX pins or the 5,6 pins (can’t remember exactly). Try using Hardware RX TX pins without enabling debug this the most stable way.

qntris, 2020-06-05:
Hi @maragelis 
I use TX and RX, debug mode is per default settings - haven’t touched it. I moved the board to the other room, connected to the AP - very few disconnects (2 missed pings for 2 hours. However I see the following error in HomeAssistant’s Mosquitto’s log:

1591378630: Client 84:F3:EB:DB:5B:06 has exceeded timeout, disconnecting. 

No other connection attempts after that. When pinging the board from a laptop connected to the WiFi, it is much more stable. My main board (NodeMCU) is connected to the main router and I am constantly pinging it from an old laptop that is connected to the AP in the other room (via WiFi) - no disconnects for 20-30 straight hours. 
I would run a trace but I am not sure how to do it - don’t have serial adapter (if needed).

Angel0ffDeath, 2020-06-05:
@qntris - In addition to what @maragelis says I would like to add that in GENERAL WiFi works perfect. Your problems (as you describe them) are from the WiFi and not from Paradox communication. 
I am also with MG5050 and no problems here.

Nevertheless, first try what maragelis says and then additionally you can try the following: Check the signal strength. Check your WiFi router configuration - probably you would like to increase leasing time or to bound your alarm panel with certain IP. Also please adjust your DC-DC (Buck converter) to 5V.

Try to move your Wemos device outside metallic box of the alarm (or at least leave the door open) and see what will happen.
Finally (if the above does't work) - you are using Wemos Pro (you state that) - it is possible to connect external antenna to amplify the signals - you must resolder a resistance near SMA connector in order this to work (google that if you don't know how) - after that put the external antenna outside the metallic box.
Or try to install WiFi router closer to your alarm box....

I don't have more ideas... Just try... If it still doesn't work... I don't know :(

qntris, 2020-06-05:
Hi @Angel0ffDeath 
I am using several boards - two NodeMCUs anD one Wemos D1 Mini Pro. 
I am testing them while not connected to the Paradox alarm, hooked up via mini USB to the AC (stable power supply). Wifi signal is excellent, Wemos has external antenna (resoldered the resistor).

Boards are tied with reserved IPs, lease interval is 120 minutes (disconnects happen quite often so not related to this), beacon interval is 40ms (the lowest possible).
Absolute M Y S T E R Y

Angel0ffDeath, 2020-06-05:
@qntris - I don't know how you can test the devices without they are connected to the alarm, but ok... you know. I hope you are using 1 device at time - the connected one and the others are disconnected.  According to me it is also mystery... 
Do you recieve actually any messages from the alarm panel until it is connected?

qntris, 2020-06-05:
I test their connection to the MQTT broker and if they drip it, also test the ping availability. I am testing two at a time almost, but I thin in one of the posts @maragelis mentioned it’s not a problem - they don’t have similar ID’s or so. When connected to the Paradox alarm I get messages and I can send commands as well. The wiring is exactly as per the picture that Maragelis has included to the project here in github, bit I am also testing them connected to a mini USB and into the AC adapter. It’s something else. I have 4 other ESP8266 base devices (smart sockets) and they are hooked to the main router without any disconnects (from the broker).

I am not exactly sure how to do tracing - besides setting 1 and precompiling what else I need to do? Where do I look for the actual traces?

Angel0ffDeath, 2020-06-05:
@qntris - You don't need to recompile - just post "trace=1" in "in topic" (probably paradoxdCTL/in if you didn't change it) then connect your laptop via USB cable to the micro USB port of Wemos/NodeMCU. Now you can open PuTTY sesion (serial) and record the log (trace). Mean while you can issue some commands. If you have some home automation system this is already integrated - you can send the log file also

qntris, 2020-06-06:
@Angel0ffDeath , I have posted "trace=1" to the "in" topic and the Wemos' light flashes after sending it, so the connection is good. However hooking up to the serial port in Putty does not show anything - the connection to the serial port is established, but no traces are displayed on the terminal screen. I am using Home Assistant for home automation, can you give me some more hints for the traces?

maragelis, 2020-06-06:
Some Chinese wemos modules have been known to have problems with the my program. I have some suggestions. 
1.) keep it on 2.4.1 core. 
2.) use a good quality module. 
3.) Only enable the features you really need. 
4.) use the master branch of the code

To use trace you have to use swap serial=1 and connect the panel to D8 D7, or use a serial ftdi module when connected to D8 D7 when Using hardware serial.  

Know bug: there seems to be a problem with heavy systems (more then 16 zones) with high zone traffic. I believe the wemos can’t keep up with the rate of messages. 
Disable hassio/HomeKit/send all events flags. And check mqtt for steady flow of messages.

qntris, 2020-06-06:
Hi @maragelis, it’s not a faulty module or busy system. Everything’s fine when I keep pinging the device from the AP in the other room or if I connect it to the AP in the other room. It’s something with the router settings but I have checked e v e r y option - no reason for that behavior. Hoped that tracing will help. As I don’t have FTDI adapter - is there any other way - via miniusb cable or commands OTA?

Angel0ffDeath, 2020-06-06:
This is trace function from the code:
void trc(String msg){
  if (TRACE) {
  Serial1.println(msg);
 // sendMQTT(root_topicStatus,msg);
  }
}
uncomment sendMQTT... and comment Serial1.print

Then you should receive trace messages in status topic

qntris, 2020-06-06:
Hi @Angel0ffDeath , I made the change that you have proposed, adding "true" (without quotes) as a third argument as it wouldn't compile otherwise. Below is the output in the "status" topic (changed the _in_ topic to _tracein_ to not interfere with my main board that is connected to the Paradox). I am testing this one without it being connected to the alarm module, just want to make sure that the connection to the MQTT broker is stable.

`Message 50 received on paradoxdCTL/status at 1:08 PM:
 try again in 5 seconds
QoS: 0 - Retain: false
Message 49 received on paradoxdCTL/status at 1:08 PM:
0
QoS: 0 - Retain: false
Message 48 received on paradoxdCTL/status at 1:08 PM:
failed, rc=
QoS: 0 - Retain: false
Message 47 received on paradoxdCTL/status at 1:08 PM:
 try again in 5 seconds
QoS: 0 - Retain: false
Message 46 received on paradoxdCTL/status at 1:08 PM:
0
QoS: 0 - Retain: false
Message 45 received on paradoxdCTL/status at 1:08 PM:
failed, rc=
QoS: 0 - Retain: false
Message 44 received on paradoxdCTL/status at 1:08 PM:
paradoxdCTL/tracein
QoS: 0 - Retain: false
Message 43 received on paradoxdCTL/status at 1:08 PM:
subscription OK to
QoS: 0 - Retain: false
Message 42 received on paradoxdCTL/status at 1:08 PM:
{
    "status": "Paradox connected"
}
QoS: 0 - Retain: false
Message 41 received on paradoxdCTL/status at 1:08 PM:
MQTT connected
QoS: 0 - Retain: false
Message 40 received on paradoxdCTL/status at 1:07 PM:
{
    "status": "Paradox Disconnected"
}
QoS: 0 - Retain: false
Message 39 received on paradoxdCTL/status at 1:07 PM:
failed, rc=
QoS: 0 - Retain: false
Message 38 received on paradoxdCTL/status at 1:07 PM:
 try again in 5 seconds
QoS: 0 - Retain: false
Message 37 received on paradoxdCTL/status at 1:07 PM:
0
QoS: 0 - Retain: false
Message 36 received on paradoxdCTL/status at 1:07 PM:
failed, rc=
QoS: 0 - Retain: false
Message 35 received on paradoxdCTL/status at 1:07 PM:
paradoxdCTL/tracein
QoS: 0 - Retain: false
Message 34 received on paradoxdCTL/status at 1:07 PM:
subscription OK to
QoS: 0 - Retain: false
Message 33 received on paradoxdCTL/status at 1:07 PM:
{
    "status": "Paradox connected"
}
QoS: 0 - Retain: false
Message 32 received on paradoxdCTL/status at 1:07 PM:
MQTT connected
QoS: 0 - Retain: false
Message 31 received on paradoxdCTL/status at 1:07 PM:
{
    "status": "Paradox Disconnected"
}
QoS: 0 - Retain: false
Message 30 received on paradoxdCTL/status at 1:04 PM:
 try again in 5 seconds
QoS: 0 - Retain: false
Message 29 received on paradoxdCTL/status at 1:04 PM:
0
QoS: 0 - Retain: false
Message 28 received on paradoxdCTL/status at 1:04 PM:
failed, rc=
QoS: 0 - Retain: false
Message 27 received on paradoxdCTL/status at 1:03 PM:
 try again in 5 seconds
QoS: 0 - Retain: false
Message 26 received on paradoxdCTL/status at 1:03 PM:
0
QoS: 0 - Retain: false
Message 25 received on paradoxdCTL/status at 1:03 PM:
failed, rc=
QoS: 0 - Retain: false
Message 24 received on paradoxdCTL/status at 1:03 PM:
paradoxdCTL/tracein
QoS: 0 - Retain: false
Message 23 received on paradoxdCTL/status at 1:03 PM:
subscription OK to
QoS: 0 - Retain: false
Message 22 received on paradoxdCTL/status at 1:03 PM:
{
    "status": "Paradox connected"
}
QoS: 0 - Retain: false
Message 21 received on paradoxdCTL/status at 1:03 PM:
MQTT connected
QoS: 0 - Retain: false
Message 20 received on paradoxdCTL/status at 1:03 PM:
{
    "status": "Paradox Disconnected"
}
QoS: 0 - Retain: false`

maragelis, 2020-06-06:
could you try this sketch if you have a second module. 
https://github.com/PascalSI/ParadoxRs232toMqtt/blob/master/ParadoxAlarmSystem/ParadoxAlarmSystemOTA/ParadoxAlarmSystemOTA.ino

its a first version many years back which I am still running on my system. just give it a go.

qntris, 2020-06-06:
Hi @maragelis, I will give it a try, but I have two questions:
1. Should I use the library versions from the Readme of the current master, or others?
2. Where should I input my mqtt-user and mqtt-password - I have created a separate user in HomeAssistant that I use in all my mqtt clients for authentication.

maragelis, 2020-06-06:
add them in the sketch ino file. just sniff mqtt messages.

Angel0ffDeath, 2020-06-07:
@qntri - According to what I see from the trace, I think your Wemos D1 Mini Pro is going to sleep mode to save energy.
Probably on your board this is the default mode. In order to prevent this you should switch to non-sleep mode I don't have Wemos Pro, but probably here @maragelis can help

qntris, 2020-06-11:
> add them in the sketch ino file. just sniff mqtt messages.

Hi @maragelis, I managed to compile and upload it, but the wifi network that I usually connect to in order to enter my wifi credentials is not showing up.

@Angel0ffDeath 
I also think it is something with the sleep mode and the beacon interval, that is probably lower on the access point (no option to see it in the config page) versus the router (40ms is the lowest there). @maragelis  is there some way we can try to add no sleep functionality so that I can test it that way?

Angel0ffDeath, 2020-06-11:
@qntris  - In setup function just add:
wifi_set_sleep_type(NONE_SLEEP_T);

This will disable sleep mode

Angel0ffDeath, 2020-06-27:
@qntris Any success??? I'm interested...

qntris, 2020-06-27:
When I unplug my TP-Link RE305 range extender it is stable (using the latest master). If I have the RE305 plugged and constantly pinging the Wemos itnis also stable. Why this happens, to this day, I have not found out. All other ESP8266 devices at home work well regardless of the setup.

Angel0ffDeath, 2020-06-30:
Well then it seems the problem is between Wemos D1 mini Pro which you are using for the alarm and range extender. What type are the others ESP8266? Try to use the same type...
I am not familiar with the extender you are using. Try to remove everything which extender could change or overwrite during establishing of the connection, i.e. give permanent IP address by reserving with MAC address (if not done), check again lease times..... I think you have some access to the extender via lan cable to set some options - check extender options and configuration as well - not only the router.
I don't know what else... If without extender it works good then as I said the problem is in extender or communication between both devices.

marianomd, 2020-08-29:
> @qntris - In addition to what @maragelis says I would like to add that in GENERAL WiFi works perfect. Your problems (as you describe them) are from the WiFi and not from Paradox communication.
> I am [...]

I read that Paradox serial port is TTL 5v. Shouldn't a ttl level shifter be needed?

Mine is a MG5000.

Angel0ffDeath, 2020-08-30:
@marianomd Correct it is 5V TTL, however Wemos D1 Mini tolerates 5V on input so you don't need TTL logic level shifter unless you are using some other board which is not 5V tolerable

marianomd, 2020-09-01:
Great, thanks.