maragelis/ParadoxRs232toMqtt #74: Suggestions

Angel0ffDeath, 2020-06-02:
Hi maragelis,

First of all perfect job you have done. However I would like to suggest you some improvements:
1. If someone has 2 partitions - code could be easily modified with parameter #define partitions = 1 (or 2) and then it is only necessary to make a cycle when issuing a command to paradox. It is not necessary to define new command.... I don't have problem with that, just suggestion
2. Due to unknown (for now) reasons it seems that after some time (random) serial port is not answering (hanging), but Wemos device is still accessible. I don't know why this happens but it is good idea if it is possible to make software reset (via mqtt command).
3. Is it possible to send me what you have discovered as communication protocol. Probably can assisst you in the above problems.
I believe toghetjer we can make this soft perfect.

Thank you.

maragelis, 2020-06-04:
Would like some help as I am very busy with day job. 
Paradox api document
https://1drv.ms/w/s!Ajw0z_yW1iqFk4JdYq19LWy1kuD2OQ?e=Bo5gfL

Angel0ffDeath, 2020-06-04:
@maragelis - Thanks for the link. Issue 2 above solved. A friend of mine discovered a perfect way for software "hard reset" of Wemos:
before setup function I added:
void reboot() {
  wdt_disable();
  wdt_enable(WDTO_15MS);
  while (1) {}
} 

In callback function - I defined new command (before last else):
    else if (progEvent == "reset")
    {
      Serial1.println("Performing software reset ...");
      reboot();
    }

Now when Paradox serial do not respond you can just publish in paradoxdCTL/in just reset and it will reboot. Both - I and my friend - we are using Wemos D1 mini, but I don't see any reasons not to work with Pro version.
For home automation it is also possible periodically to send command armstatus and if there are no answer automatically to publish reset command.

Issue 1 - most of the people have only 1 partition, and those who have 2 partitions probably want to arm them separately. My suggestion is good only if you want to arm both partitions. If you want to arm them separately then 2 jsons (or definition of new command) is necessary. Actually probably definition of new command is better and let the user decide how to make it...

Will close this issue since  everything is solved. You can add the code above...

Regards

maragelis, 2020-06-04:
There seems to be some sort of memory leak in one of the libraries, this I think is causing the esp to hang. One thing I dont like, is waiting for the 37 bytes in a while loop on the main thread.
Thats the reason I started the esp32 version, the esp32 has 2 cores so I can do different work on each core.  I would like this project to be stable without having to reset it, its an alarm system and it has to be alive when we need it that one time.  
Check out the the esp32 version I think its worth our time.
https://github.com/maragelis/ParadoxMqtt32 
This version is a complete port from the current project. Its working on my bench system but yet to try it on my main alarm system.

Thanks for any help.

Angel0ffDeath, 2020-06-04:
Actually I discovered the same – memory leak. Currently I don’t know is it from your code or some library, but you code look perfect.You are correct about ESP32, however the soft should work! All of the libraries were updated and probably this problem do not exist if you use the last libraries. Will try to compile your code with the last libraries and will test it, but in order to obtain correct results (no leak, no hanging) will take some time to test.I have only 1 alarm system, but it is not a problem… Will manage it somehow. Will update you with info, but I really want this soft to be better – continue to work. See my post about software “hard reset” 😊 Btw – your name sounds like Greek 😊I’m from Bulguaria. Do you live in Greece? Best reagards Angel   Sent from Mail for Windows 10 From: PanagiotisSent: 04 юни 2020 г. 22:05To: maragelis/ParadoxRs232toMqttCc: Angel0ffDeath; State changeSubject: Re: [maragelis/ParadoxRs232toMqtt] Suggestions (#74) There seems to be some sort of memory leak in one of the libraries, this I think is causing the esp to hang. One thing I dont like, is waiting for the 37 bytes in a while loop on the main thread.Thats the reason I started the esp32 version, the esp32 has 2 cores so I can do different work on each core. I would like this project to be stable without having to reset it, its an alarm system and it has to be alive when we need it that one time.Check out the the esp32 version I think its worth our time.https://github.com/maragelis/ParadoxMqtt32This version is a complete port from the current project. Its working on my bench system but yet to try it on my main alarm system.Thanks for any help.

Angel0ffDeath, 2020-06-05:
@maragelis - can you please contact me on my e-mail: angeldeath743 at gmail dot com

Your code works perfect (for now) from about 3-4 hours. Compiled with the latest versions of all libraries. However, according to me, it seem the serial communication is not according to protocol you sent me. I would like to discuss mine observations in private...