maragelis/ParadoxRs232toMqtt #75: Commands always result in "Problem connecting to panel"

kareem613, 2020-06-05:
I have almost everything working great and all events are showing up in the MQTT broker verified with mqtt explorer. It's been solid for months now. 

My problem is when trying to send a command. No matter what command I try I get 
`{"status": "Problem connecting to panel"}`

Ex command.  xxxx substituted with the same user code that works in the display.
topic
`paradoxdCTL/in`

body
`{"password":"xxxx","Command":"stay","Subcommand":"0"}`

Any suggestions on how to troubleshoot this?

maragelis, 2020-06-06:
dont have 0 in your code, check cables are connected correctly

kareem613, 2020-06-06:
Zero in the code! That was it. Wow. I wasted a lot of time on that one. 
That's worth a wiki or readme.md entry. 
All working perfectly now. Thanks!

closed 2020-06-06 by kareem613