maragelis/ParadoxRs232toMqtt #80: Problem Connecting to Panel

mrsnob, 2020-08-15:
Hi firstly thanks for an excellent module, I am having a problem  like Issue 75 that I get all stats zones etc but when I try to issue a command I get the Problem connecting to Panel. I have checked cable and 0 in command as with issue 75 but the problem still persists. Did the rx/tx swap also but to no avail. Note I am using a sp6000. Have the same setup at parents home and it is working perfectly.

maragelis, 2020-08-24:
Check cables and power supply

mrsnob, 2020-08-24:
[image]
Hi Sir I have tried putting in new cables and a new wemos as well, but still getting the error message note I am getting all status messages perfectly it is just the controlling of the panel. Thanks for your help so far