maragelis/ParadoxRs232toMqtt #81: BabyWare or set Zone dara name

bigjuanpa, 2020-08-17:
Hi, thanks for all your work.
I start playing today and this works really good.
My question is if I can use this to connect to babyware to set some values like zone data name.
Since now my zone have the default name.

Regards.

maragelis, 2020-08-24:
No you can not

Angel0ffDeath, 2020-08-30:
@maragelis and @bigjuanpa - I have implemented memory read and memory write functions. I was able to dump whole memory of the panel, then I wrote separate software (for PC) for offline programing, and this soft uploads changes to Wemos D1 Mini via ftp and starts write command to write changes to panel.

So the short answer is - with this software it is not possible - not supported, but in general it is possible :)

Flopescu, 2024-01-12:
hi @Angel0ffDeath , 
would you be so kind in sharing the approach you made to dump and upload the memory?
long story short a few years ago, when I added the wemos to my mg5050 I changed all the codes, did some zone labels etc ... and now I in need of the installer code..