maragelis/ParadoxRs232toMqtt #82: Two Panels - How can I see what panel is arm

bigjuanpa, 2020-08-27:
Hi, I have an alarm with two panels, so I can arm or disarm the two panels perfect with the sub command.
the problem is that when I arm one panel the second panel show's arm since the hassio send only the arm command with no panel info.

How can I do to see what panel is arm to don't show on hassio the 2 panels arm when is not .

Thanks

sparkydave1981, 2020-08-27:
By 'panels', do you mean the two Partitions?

What code are you using to arm/disarm?

bigjuanpa, 2020-08-27:
Yes two partitions.
Im using
alarm_control_panel:
 platform: mqtt
 state_topic: "paradoxdCTL/hassio/Arm"
 command_topic:"paradoxdCTL/in"
 payload_disarm: '{"password":"0000","Command":"disarm","Subcommand":"0"}'
 payload_arm_night: '{"password":"0000","Command":"sleep","Subcommand":"0"}'
 payload_arm_home: '{"password":"0000","Command":"stay","Subcommand":"0"}'
 payload_arm_away: '{"password":"0000","Command":"arm","Subcommand":"0"}'

And the same for the second partition but with sub command 1.
This works fine but when I run for partition 1 on the alarm control panel 2 shows activated. But is not real just the partition one is activated.
The think is that it check on arm status and is arm but just one partition don't send the Matt showing what partition is arm or maybe I'm doing something wrong.

Thanks

Angel0ffDeath, 2020-08-30:
@bigjuanpa Hi, first of all I'm curious you are able to issue any commands with password '0000' because conversion of the password in the code is incorrect, but probably this is some issue of panel....
Now to your question - since the arm status messages are one and the same for both partitions you probably would like to change them depending on what partition you are issuing commend. For instance, if you arm partition 1 to return armstatus armed1 or armed2 if you arm partition 2. For 2 panels (I assume you have 2 Wemos devices - 1 for each panel, you can further divide them by making different root topics - paradoxdCTL1 - for the first panel and paradoxdCTL2 for the second.
Anyway you will need some program skills :) ....

bigjuanpa, 2020-09-23:
The 0000 os just for demostration I use the alarm code and it works.
I will try to use the test code that is on the test branch that sends arm1 and arm2 let's see if that works.
Thanks