maragelis/ParadoxRs232toMqtt #84: Showing Disarm when is arm

bigjuanpa, 2020-09-23:
Hi for some reason sometimes if the status is change and is not registered on mqtt after check when the alarm is arm and I check the panel status it show's disarm.
I test that on node red and report disarm when is arm.
The only fix is disarm and arm again from the panel and after that on the next arm it fix.

Any idea why the serial report disarm when is arm?
Thanks

userMak, 2020-09-23:
Same think happening to me some times

bigjuanpa, 2020-10-06:
at this moment im checking every 15 minutes the state.

bigjuanpa, 2020-10-19:
Same issue even asking a few times return disarm when is arm.
any help?

maragelis, 2020-11-24:
You will have to send a status request message.

userMak, 2022-05-27:
> Same issue even asking a few times return disarm when is arm.
> any help?

Hi have you found a solution? 
I am not sure how I could the status request message as @maragelis  suggest

croftsquirrel, 2022-07-08:
I also have the same problem.  Can't figure out what is causing it.  In my particular case it seems to more likely happen when the system is armed and we have a power outage.  When the power is restored I can't send disarm message through.  Or sometimes even when it says disarmed it still actually armed and then is triggered.  Also then need to manually disarm on panel.
@maragelis is there any updates or fixes that can resolve this?  I have the latest version available 2.2.3

maragelis, 2022-07-09:
There is a way to get over this, just set the sendmqtt method to send an mqtt retain message, that will keep the last message sent on the mqtt topic. That way when HA reboots it will always read the last status sent.

userMak, 2022-07-09:
> There is a way to get over this, just set the sendmqtt method to send an mqtt retain message, that will keep the last message sent on the mqtt topic. That way when HA reboots it will always read the last status sent.

Hi, can you give an example please?

croftsquirrel, 2022-07-09:
It's seems to be a problem with connecting to WiFi or MQTT even though WiFi is back and up and running.  Also the zones opening also doesn't get sent through from the ESP8266.
Is there a way to program the "void handleMqttKeepAlive()" or similar section in the ParadoxAlarmSystemOTA.ino to restart the device if there is no connection.  I need the ESP8266 to restart when it can't connect, not just to retry the connection.
Sorry I'm not a developer so not too sure what to do but could one put this in the code somewhere if it doesn't connect. "ESP.reset();" ?

maragelis, 2022-07-10:
I am currently testing a version using micropython on the esp32. Seems to be very stable and doesn't suffer from random disconnects. It will also be able to run on a raspberry pico w. So far I have got the hassio code working, and working on the web interface. So be patient code will be uploaded to GitHub soon.

maragelis, 2022-07-13:
https://github.com/maragelis/micropython_paradox

bigjuanpa, 2022-07-16:
> https://github.com/maragelis/micropython_paradox
Looks really good, I never use micropython on esp32.
Do you have any tutorial or something?
Do you think this is better tan your paradox for esp32 code?
Thanks

maragelis, 2022-07-18:
yes much better and runs quite well.

maragelis, 2022-07-18:
need testers so your help will be gratefully appreciated

bigjuanpa, 2022-07-18:
I will try to install and test it in a few days.
The connection are the same than before?
Regards.