maragelis/ParadoxRs232toMqtt #86: Possible to use this FW with ESP-01S?

vampywiz17, 2020-11-19:
Hello there,

I install this FW on ESP-01S. It seems to work well, i able to set the wifi and MQTT. 

I'm not able to test it, because i have no paradox alarm system. 

So my qestion is that ESP-01S TX/RX pins is working? it is same pin address all ESP8266 board?

My friend will test it (temporary powering with USB-UART 3.3V) I hope it will work. If not, of course we will buy a D1 mini, but the ESP-01 is "stock" it is the reason that we try it.

maragelis, 2020-11-24:
Yes you can. It will work just fine

Angel0ffDeath, 2021-03-07:
A little late response, but...
ESP-01 is working ok - TESTED. Use ESP-01 with black PCB (1M flash) in order to be able to use OTA.  The blue PCB is with 512k and no way to use OTA. Still can work, but new flashing only through PC.